letscontrolit/ESPEasy #3158: Document P093 Mitsubishi Heat Pump plugin

TD-er, 2020-07-06:
See [forum post](https://www.letscontrolit.com/forum/viewtopic.php?p=45694#p45694)

In short:
To send a command, you should publish the ESPEasy command into a `cmd` sub topic right under the topic the plugin publishes in.

crnjan, 2020-07-15:
Hey!

In the [docs](https://espeasy.readthedocs.io/en/latest/Plugin/P093.html#p093-page) there is a link to fully working [example](https://community.openhab.org/t/mitsubishi-heat-pump/91765) where one can see the exact syntax:

```
Type number:acTemperature [ 
  stateTopic="FF_Nik_Room/ac/settings", transformationPattern="JSONPATH:$.temperature", 
  commandTopic="FF_Nik_Room/ac/settings/cmd", formatBeforePublish="MitsubishiHP,temperature,%d" 
]
```

TD-er, 2022-05-31:
Sure!
One tip, if you put something like this in the first message of a pull request, it will close the issue when it is merged.

```
Fixes: #3158
```