letscontrolit/ESPEasy #3161: LCD2004 i2c address in plugin does not match the HW address but LCD is working

ghtester, 2020-07-08:
I have a strange 'hidden' issue. It probably appeared during several ESP Easy firmware upgrades. on this node. 
Now the LCD2004 i2c address visible in plugin configuration is different than the physical PCF8574 device's address, detected by i2cscanner (see the pictures). The display is working despite the LCD2004 i2c address visible in plugin configuration is wrong... I did not try to put the correct value there yet for the case if some of developers is going to analyze it and needs a debug info.

 
Build:⋄ | 20108  - Mega
-- | --
System Libraries:⋄ | ESP82xx Core a5432625, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ | My Build: Jun 26 2020 20:47:09
Plugin Count:⋄ | 32 [Normal]
Build Origin: | Self built
Build Time:⋄ | Jun 26 2020 20:49:40
Binary Filename:⋄ | ESP_Easy_mega_20200626_custom_ESP8266_4M1M
Build Platform:⋄ | Linux-4.15.0-51-generic-x86_64-with-glibc2.27
Git HEAD:⋄ | HEAD

TD-er, 2020-07-08:
That's a bug that has been fixed in the last build and was only present during this build you link here.

I optimized the memory usage of generating the combobox for a lot of things, among the I2C selector.
But the bug is that the selector value was not the actual I2C address, but ranging from 0... nr of options.
So if you save this page, your LCD will probably no longer work until you change the firmware and save the settings again.

ghtester, 2020-07-09:
OK, thanks for the explanation, I'll wait for some recent PRs to be merged, then recompile a fresh custom build and I'll close this issue when it fixes the problem on the specific node.

TD-er, 2020-07-09:
Which PRs specifically?

ghtester, 2020-07-09:
At least #3145 and #3148.

TD-er, 2020-07-09:
Well the last one, with the WiFi, is not yet finished as I have seen some situations where it would not reconnect so I'm looking into that first before merging it.

ghtester, 2020-07-15:
I have just compiled the custom firmware using Vagrant and I can confirm that topic issue is fixed, also the "Submitted" box appearing without submit is fixed. But the issue which should be already fixed and which is already closed, still persists: #3138

closed 2020-07-15 by ghtester