letscontrolit/ESPEasy #3167: Boot state not working combined with switch input configured on same GPIO

steljwagh, 2020-07-12:
When the GPIO boot state is configured for, for example GIO-O to "Output Low", the behaviour is expected and there the GPIO-0 is "off" after boot.
But when you have configured that boot state and also configured a "Switch input - Switch" on that same GPIO-0, the output of GPIO-0 is in a half state, looks like some PWM is active after boot.
When you give a ,0 or ,1 command to that switch, the behaviour is expected.

So the boot state is doing something strange when there is a switch input - switch is configured.

The workaround can be easily achieved by making a On System boot rule set. But configuring boot state should aquire the same result right?

I have a LED+resistor connected directly to GPIO-0 to check the result.

tonhuisman, 2020-07-12:
I wouldn't combine the `Output Low` state with a gpio to be used as an input switch, either `Default` or `Input` would be more appropriate. Your settings contradict each other. The switch input should be properly terminated by a pull-up resistor, as GPIO-0 must be high during boot to avoid entering flash-mode. Your switch should pull it to 0 when activated.
This setup is bound to cause issues, because you can guess what happens when power is applied while the switch is in 'activated' (low) mode...

steljwagh, 2020-07-12:
The problem only occurs after a boot and can be 'fixed' by giving a on/off command to that switch, after that it's working. The same behaviour can be seen on GPIO-2, so it's not GPIO-0 specific related.

The switch input should be probperly terminated by a pull-up resistor, but... I'm only using 9 milliampere from GPIO-0 (and 2 for that matter, same issue) to test this behaviour.

This dimmed / half state has however, has something to do with drawing power from the GPIO for the LED.

I just changed my LED position to 3v3 <-> GPIO-0, the boot-states are now working as expected. I will close this after your comment. 👍

tonhuisman, 2020-07-12:
My point is that you shouldn't combine the `Output Low` gpio state and eventually _using_ it as an input. That puts the gpio in the wrong mode for sensing the input state.

steljwagh, 2020-07-12:
I use a switch input device, but don't use it as a (data) input, all I want to do is control pin state high / low. In my understanding, The function of boot- state "output low" it to get the pin state in low. And it does work that way, so I know have a ESP-01 which can control a LED on/off by pin state low/high.

I do my best to understand what you mean, but I don't use it as a input, although the device name is input in ESP_Easy, it's a common way to make a remote switch (from what I learned so far using documentation)

closed 2020-07-12 by steljwagh

steljwagh, 2020-07-13:
Thanks again for your feedback. It took me way longer then needed to realize what you are saying. To control the state of a GPIO a input switch device is not needed to be configured in ESPEasy (Unless you have multiway-switching and you want to also sense the state, but that's a new problem to solve for later.)