letscontrolit/ESPEasy #3168: SendTo executed in Event System#Boot

stwa2020, 2020-07-13:
Hello,

I have two ESPs connected via P2P Network, the communication works fine, the Events are executed via trigger.
When the Event System#Boot is used to execute the Event will not reach the remote ESP.

_Example:_
When a reboot of ESP1 is done it will not set the GPIO 14,0

**ESP1**
on System#Boot do
 OLEDFRAMEDCMD,1,REMOTE:GPIO14=OFF
 sendTo,2,"event,SET_OFF"
endon

on SET_ON
 OLEDFRAMEDCMD,1,REMOTE:GPIO14=ON
endon

on SET_OFF
 OLEDFRAMEDCMD,1,REMOTE:GPIO14=OFF
endon

**ESP2**
on System#Boot do
 OLEDFRAMEDCMD,1,REMOTE:GPIO14=OFF
 sendTo,1,"event,SET_OFF"
endon

on SET_OFF do
 GPIO,14,0
 OLEDFRAMEDCMD,1,GPIO14=OFF
 sendTo,1,"event,SET_OFF
endon

on SET_ON do
 GPIO,14,1
 OLEDFRAMEDCMD,1,GPIO14=ON
 sendTo,1,"event,SET_ON
endon

Problem:
In my example I have used also a display to make it easier to understand what I mean.
The command http://IP_ADDRESS_ESP2/control?cmd=event,**SET_ON**  or **SET_OFF** _will set the display on ESP1 and ESP2 and set the GPIO14_.
When ESP1 is rebooted the display on ESP1 will show **REMOTE:GPIO14=OFF** but it will not set it to **GPIO14,0** and show on the ESP2 display **GPIO14=OFF**
When ESP2 is rebooted the display on ESP2 will show **GPIO14=OFF** but it will not set the display on ESP1 to **REMOTE:GPIO14=OFF**

Any Idea how to execute sendto when after the Event System#boot is done.

Description for Event System#Boot
occurs once after Tasmota is **fully intialized** (after the INFO1, INFO2 and INFO3 console messages). System#Boot triggers after Wi-Fi and MQTT (if enabled) are connected. If you need a trigger prior to every service being initialized, use Power1#Boot

Does this mean the P2P network is not part of **fully initialized** or is this a bug ?

Thank you for your time and support,

Stefan

TD-er, 2020-07-13:
Act on the event `wifi#connected`.
At boot the Wifi is not yet connected, so you cannot send anything.
If you need to send it only once after boot, you can set a variable, which will be cleared at boot.

So on the boot event, you set a variable to `1` to signal it has to send out the stuff you need to send when you got connected to WiFi and on the `wifi#connected` event you check for this variable to be `1` and then send out the message plus set the variable to `0`.

stwa2020, 2020-07-13:
I've created the example configuration now in reality 
I did the test with fixed IP and with DHCP
Used Version: mega-20200608

Following events have been used.

**##### CONTROLLER #####**
on System#Boot do
 OLEDFRAMEDCMD,1,R:GPIO14=OFF
 sendTo,2,"event,SET1_OFF"
endon

on SET_OFF do
 GPIO,14,1
 OLEDFRAMEDCMD,1,GPIO14=OFF
 sendTo,2,"event,SET1_OFF
endon

on SET_ON do
 GPIO,14,0
 OLEDFRAMEDCMD,1,GPIO14=ON
 sendTo,2,"event,SET1_ON
endon

**#### REMOTE1 ####**
on System#Boot do
 OLEDFRAMEDCMD,1,R:GPIO14=ON
 sendTo,1,"event,SET_ON"
endon

on wifi#connect do
 OLEDFRAMEDCMD,1,R:GPIO14=ON
 sendTo,1,"event,SET_ON"
endon

on Http#Initialized do
 OLEDFRAMEDCMD,1,R:GPIO14=ON
 sendTo,1,"event,SET_ON"
endon

on Time#Initialized>10 do
 OLEDFRAMEDCMD,1,R:GPIO14=ON
 sendTo,1,"event,SET_ON"
endon

on SET1_ON DO
 OLEDFRAMEDCMD,1,R:GPIO14=ON
endon

on SET1_OFF DO
 OLEDFRAMEDCMD,1,R:GPIO14=OFF
endon

When I reboot **REMOTE1** the GPIO 14 is not set on **CONTROLER** when I execute on **REMOTE1** the command _sendTo,1,"event,SET_ON"_ it will set the right status for GPIO 14

Note: in the documentation is written "...the _System#Boot_ triggers **after** **Wi-Fi** and **MQTT** (if enabled) are **connected**"
Correct me if I am wrong but this would mean the _wifi#connect_ would be similar to _System#Boot_

I guess the event trigger are not working correct with P2P Network

stwa2020, 2020-07-13:
I did some tests with a timer and found the following,

I use, 
 **SendToHTTP 192.168.1.60,80,"/control?cmd=event,SET_ON"**
instead of, 
 **sendTo,1,"event,SET_ON"** 
and it works.

#### REMOTE1 ####
on System#Boot do
  timerSet,1,10
endon

On Rules#Timer=1 do 
 OLEDFRAMEDCMD,1,R:GPIO14=ON
 OLEDFRAMEDCMD,2,TIMER
 **SendToHTTP 192.168.1.60,80,"/control?cmd=event,SET_ON"**
endon

TD-er, 2020-07-13:
Can you check in the system variables page what the value of `%iswifi%` is ? (or use it to display it on the OLED)

stwa2020, 2020-07-14:
%iswifi% is 4

I've created now a clean config and can confirm the **sendto** command doesn't work when %iswifi% is 4 while the **SendtoHTTP** works using the event **wifi#connected**
It works when I execute the **sendto** command manually via tools -> command while %iswifi% is 4 

PS: in the previous test I had a typo I've used **wifi#connect** and not **wifi#connected**

TD-er, 2020-07-14:
That makes sense.
The sendto command does have a look at the `WiFi.isConnected()` and the `%iswifi%` does look at the internal state I keep of handled events.

Working on that part and it is related to https://github.com/esp8266/Arduino/issues/7432
Thanks for testing.

tonhuisman, 2023-09-01:
This seems to be solved, so can be closed.

closed 2023-09-01 by TD-er