letscontrolit/ESPEasy #3171: Retain flag not appearing on tools->advanced menu

fnees, 2020-07-17:
Hi,
I have all my devices running on mega-20190315 (ESP_Easy_mega-20190315_normal_ESP8266_4M.bin).

I decided to start to upgrade to Build: 120, Core Version: 2_3_0 (ESPEasy_R120_4096.bin) which is stated as the Current Stable on the getting started page (https://www.letscontrolit.com/wiki/index.php?title=ESPEasy#Get_started).
The docs say there should be the retain flag on menu Tools -> Advanced, as it has always been (https://www.letscontrolit.com/wiki/index.php?title=ESP_Easy_web_interface#Tools_Advanced_page).

But my web interface has it not. So, is it a documentation problem and we wont have even this global retain anymore?

I have the OpenHAB MQTT controller configured, and I don't have the controllers page either, by the way. 

[image]

tonhuisman, 2020-07-17:
> I decided to start to upgrade to Build: 120, Core Version: 2_3_0 (ESPEasy_R120_4096.bin) which is stated as the Current Stable on the getting started page (https://www.letscontrolit.com/wiki/index.php?title=ESPEasy#Get_started).

That documentations is really outdated, get a more current release from here: https://github.com/letscontrolit/ESPEasy/releases

TD-er, 2020-07-17:
I have updated the Wiki page a bit to indicate it is best to download the versions from the Github releases page.

fnees, 2020-07-17:
Great guys, thank you. I will close as it was a documentation mistake and has been fixed.
Sorry about it but I thought that through the webpage I would get to the latest release more quickly!

TD-er, 2020-07-17:
No worries, the documentation is a bit behind, and indeed we need to make it more focused towards what people want to know when they start learning about the project and not written from the programmer's perspective assuming people already should know it all.

tonhuisman, 2022-04-18:
This can probably be closed, as updating documentation is an ongoing effort.

closed 2022-04-18 by TD-er