letscontrolit/ESPEasy #3172: ESP flasher 0.04.007 does not find COM port of my connected device

inzuno, 2020-07-17:
Hello.

I was previously on ESP flasher 0.02.002 (provided with ESPEasy_mega-20181105) and I had issues with uploading Post flash actions.
So I decided to update my version to the latest (ESPEasy_ESP82xx_mega-20200703) and it find 14 ports but not mines.

I obviously unzipped the entire folder, tried to reboot computer, ereasing all and download again, run as admin and now run out of options.
Is it possible to set port manually or change a config file?
I have a D1 mini pro with CP210x USB to UART under win 7
[image]

This is my first time opening an issue so please be gentle :)

Thanks!

TD-er, 2020-07-17:
The flasher does read the available ports from the registry.
Problem with this approach is that it may also find COM port numbers which are no longer connected as Windows does assign a new port number if there is a new combination used of USB chip + USB port in a hub.

What is obviously a bug here (in the flasher?) is that the outdated COM ports are listed as being "online"

inzuno, 2020-07-17:
Tanks for your answer.
In my other software (ie Arduino) everything is fine...

Grovkillen, 2020-07-17:
Yep, I really need to spend some time fixing this.

haldi4803, 2020-07-25:
How Comes the COM12 does not show up in ESP Easy Flasher 0.4.007 but works fine when using the FlashESP8266.exe directly ?

[image]

Grovkillen, 2020-07-25:
The problem is that I look into the registry for specific naming
conventions and list found COM ports based on that. I think I need to do
this in another way for future versions.

BaGRoS, 2022-09-13:
Some progress on this? Same bug ESP.Easy.Flasher 0.04.007

B4STI1, 2022-11-06:
UP i'm using esp easy flasher 0.04.007 on windows 11 22621.755
the COM port is in the periph manager (COM1) but in the software there is only a COM3 that doesn't work

tonhuisman, 2022-11-06:
Please use the, also included, Espressif flash download tool to flash your ESP. And when working with ESP32, this is even a requirement, as the ESP Easy flasher won't work with ESP32/ESP32s2 at all.

B4STI1, 2022-11-06:
thx to answer this quickly, sorry i forgot to precise i'm triing to flash an esp8266EX

B4STI1, 2022-11-06:
[image]

B4STI1, 2022-11-06:
i don't know how to use this tool but if it even strugle to open the COM port i think it won't go anywere

B4STI1, 2022-11-06:
i tried to change the  COM port but it don't seem to go anywere neither
[image]

TD-er, 2022-11-06:
Here is some basic introduction to flash ESP32 using the Espressif download tool:
https://espeasy.readthedocs.io/en/latest/Reference/Flashing.html?highlight=espressif#flash-esp32-with-espressif-download-tool
The procesure is the same for ESP8266, only there is no "factory" bin file you need to pick.
Make sure not to use a .bin.gz file, but just the .bin file.

When you can't select the COM port, or access it, it often helps to unplug the USB cable and replug it.
This way you know there isn't any program or process trying to still access it.
Sometimes plugging it into another USB port may force Windows to assign a new COM port nr to it.
This may also "fix" some strange issues when there are funky settings in the registry for a specific COM port.

N.B. you may also need to set the "offset" to "0" in the download tool. (right colomn next to the field where you have the filename)

closed 2022-11-09 by TD-er