letscontrolit/ESPEasy #3174: 2-step uploader core 2.7.1 needed

Wiki591, 2020-07-19:
In the latest releases the minimal_core_242_ESP8266_1M_OTA comes without Domoticz Controller. The release contains a special build named ESP_Easy_mega_20200703_minimal_core_271_ESP8266_1M_OTA_Domoticz.bin.gz. Bt the 2-step uploader is still compiled against 2.4.2 and is not able to update the device using a compressed file.

Please include a 2-step-uploader in the nightly builds to enable the update of the small devices further on, otherwize the compressed files for the different controllers make no sence.

TD-er, 2020-07-19:
First initial tests showed we do not (yet) gain as much space as expected.
- Core 2.7.x builds are quite large
- You need either one of the active or 2-step bin in uncompressed form on the flash

It may be usable when we strip out the SPIFFS library and settings struct and only read the required info from somewhere, set by the last running build.
For example:
- hostname
- SSID/key
- IP
- portnr

Those must then be stored somewhere on the flash, outside the SPIFFS.
So we may need to use the EEPROM library for this, but that also implies we need to store the active settings in there before starting the OTA update.

Also the build script needs to be updated to be aware of the uncompressed and compressed size of both binaries (2-step OTA and the new to be installed bin)

Wiki591, 2020-07-19:
Eeeehm, thats kind of ridiculous: how do I get my Domoticz Controller back by using a minimal OTA bin.gz? In this case it makes no sence to include the core 2.7.1 minimal OTA builds, devided in releases with/without controller, you will never get them uploaded without a 2-step-uploader core 2.7.1? The uncompressed .bin core 2.4.2 is without controller.

TD-er, 2020-07-19:
I'm not stating it won't be done, I merely put down some kind of to-do list for me of everything popping in my mind when reading your issue :)

Wiki591, 2020-07-19:
To avoid any irritations the compressed minimal OTA builds should be removed from the nightly builds (practically - if I understood right - at this time they are useless) and the minimum of controllers should be added again to the uncompressed - core 2.4.2 - minimal OTA builds.

Or, may be, did I miss something and they are useful for whatever?

TD-er, 2020-07-20:
The 2.4.2 builds make no sense to be compressed (unless you flash them from a core 2.7.x build)
For that the build script should be adapted.

Removing the 1M .gz builds from the build script can be done... or we make a new 2-step OTA binary.

Wiki591, 2020-07-20:
Seems as if I didn't explain correctly. If the actual compressed minimal OTA bin are flashable ( I uncompressed the one containig Domoticz Controller, size: over 620k) there is the need of a new 2-step OTA binary compiled against core 2.7.1. With the existing 2-step OTA binary flashing a compressed one is not possible.

TD-er, 2020-07-20:
Yep, that's correct (and I understood it like that)
For flashing .bin.gz you need to have a version installed running core 2.7.0 (or very late 2.6.3)

So it makes sense to have those for 4M and 2M flash sizes.

For 1M flash size, things get complicated:
- 2-step-OTA must be core 2.7.x or newer to flash .bin.gz
- compressed size of 2-step-OTA image + decompressed new image must fit (or else no future upgrades possible)
- decompressed size of 2-step-OTA image + compressed new image must fit

So unless we have a 2-step OTA image that supports .bin.gz, there is no use for generating 1M .bin.gz files.
However, it will depend on the (future) 2-step OTA image whether the 1M .bin.gz files of current builds may be usable if we have this new 2-step OTA image.

It all depends on whether the new 2-step OTA image can read the settings or the last running build should prepare for an upcoming 2-step-OTA update.

Wiki591, 2020-07-20:
Yep, thats what I understood. But flashing the uncompressed actual minimal OTA .bin, recognizing that Domoticz Controller is missing and having a look at the bin-collection, seeing there is a compressed minimal OTA image including Domoticz and not to be able to flash it - thats like the carrot in front of donkeys nose.....

So might it be possible to readd the domoticz controller to the uncompressed minimal OTA.bin?

TD-er, 2020-07-20:
If a .bin file is missing, while the same .bin.gz is present, then the build script removed it from the ZIP file (or did not include it) due to size restrictions.
The build script is aware of the max. free space and the currently known smallest 2-step OTA bin.

But I guess it needs some updating to get rid of these annoyances.