letscontrolit/ESPEasy #3175: No device called BH1750 but can find it with I2C Scan

hme0354, 2020-07-19:
I'm using espeasy with an ESP32. With the I2C Scan i can find my BH1750 (0x23), but if i want to add it as a device, there is no device BH1750

tonhuisman, 2020-07-19:
What bin file have you uploaded?
The only bin files where it _isn't_ included are the ones with minimal in the name, and the specific hard* configurations.

hme0354, 2020-07-19:
BIN-File: ESP_Easy_mega_20200608_test_ESP32_4M316k.bin 
and also tested: ESPEasy32_R20100

Hardware: LOLIN32 (ESP32-WROOM-32)

tonhuisman, 2020-07-19:
Aha, well, I've found this reference in the code:
```
#ifdef PLUGIN_SET_TEST_ESP32
  #undef USES_P010   // BH1750          (doesn't work yet on ESP32)
```

So that isn't included in any ESP32 build because it isn't compatible with that chip (yet?)

hme0354, 2020-07-19:
how?

tonhuisman, 2020-07-19:
The bin files for that are included in the ESP8266 download (and may have WROOM02 instead of ESP32 in their name).

Disclaimer: I haven't tried this for myself on an ESP32 controller.

tonhuisman, 2020-07-19:
If the BH1750 doesn't work on your ESP32, you can connect it to an ESP8266 board, that combination should work as expected.

hme0354, 2020-07-19:
ok. I will try

TD-er, 2020-07-19:
Well no matter if the plugin is included in the build or not.
If it isn't found on the node by an I2C scan there may be some other issue at hand.

`(doesn't work yet on ESP32)` may be some kind of left-over part from when I started to make it all compile for the ESP32.
In the beginning there were lots of I2C devices not working well (or not working at all) on the ESP32 as the code for I2C was far from stable.
But right now, I don't see why it should not be included (apart from the fact I have not tested the sensor myself on ESP32)

At least it should show up in the I2C scan, if it is connected correctly.

Just make sure to double check the assigned pins for I2C (Hardware tab) and maybe also test with some 4k7 (or 10k) pull-up resistors on both the SDA and SCL lines.

hme0354, 2020-07-19:
as wrote, it shows up in the scan. but I cant't add it as an device, because there is no lib for it.

TD-er, 2020-07-20:
@hme0354 Sorry, missed that (quite essential) part of your sentence...

ventilator, 2020-07-31:
Same issue here. So there is the possibilty to flash these WROOM files to enable that device? What can I (or you) do to enable it in the testing ESP32 builds?
It is even recognized by the I2C scan by its name but not selectable as a device (as stated above)
[image]
[image]

hme0354, 2020-07-31:
I flashed the same test build and now it works. but now I decided to take the TLS2591 instead of the bh1750

ventilator, 2020-08-01:
> I flashed the same test build and now it works. but now I decided to take the TLS2591 instead of the bh1750

Okay, but do you use BH1750 sensor hardware or an TLS2591? Because the latter plugin uses a different I2C address and cannot work with an BH1750.

[image]

hme0354, 2020-08-01:
no, I bought a new sensor.

tonhuisman, 2020-08-06:
Today I received a BH1750 sensor I ordered (locally), and changed the source a little (some complaint about a pointer size and removed the explicit excludes) to be able to compile it into an ESP32 release. It seems to be working just fine here on an ESP32s boards (no 'funny' outcomes, even in debug mode), so I'll cook up a PR to include this sensor for ESP32 use.

PR #3200 (2020-08-06): [P010] Changes to enable use on ESP32 (#3175)
- Fixed compile errors to allow inclusion in ESP32 release
- Adjust build configuration to not explicitly exclude the needed library
- Don't disable plugin for ESP32 builds

@TD-er :
Not sure if the `#typedef` should also be used for ESP8266?

Resolves #3175
closed 2020-08-06 by TD-er