letscontrolit/ESPEasy #3176: Feature Request: HC-12

Jarppuli, 2020-07-22:
Please add support for communicating with HC-12 transceiver.

TD-er, 2020-07-22:
Can you be a bit more specific?
What should be sent or received via that module?

Jarppuli, 2020-07-23:
Ok let me give some thoughts how I would like to use it:

There could be espeasy device in my home network, that has a HC-12 transceiver connected to it.
Then there could be a second wemos in a distant location that also is paired with HC-12.

Now, the communication could be just one way, by assigning a role of RX or TX to each wemos, but two way communication would allow for more possible uses of course.

Example 1:
In the remote location, a wemos could have a set of relays connected to it, and it would then be listening for commands to open or close relays.
On the transmitter at home, there could be a simple website running on the wemos that has buttons that transmit suitable commands that will then control those remote relays.

Example 2:
There could be a motion sensor in the remote location, and if motion is detected the wemos uses HC-12 for sending a predefined command forward to home location.
At home, the wemos with HC-12 listens to a predefined command and if it is heard, actions are taken.
(there could be a buzzer or light that activates to let me know there is someone moving on the remote property)

There propably needs to be a HC-12 compatible "controller" for sending commands, and then a HC-12 "device" that is used for listening to commands?
Also, there should be possibility to read and send serial commands using rules.

To my understanding HC-12 works by a similar way as Reyax transceivers do, so it should be compatible automatically?
I personally have achieved over 7 kilometers even in poor weather conditions.
(I used arduino nano, but as Im a poor with code, using espeasy and its UI would make things easier)

[image]

TD-er, 2020-07-23:
Why not use a reliable way of communication, like LoRa?
The current implementation of TTN/LoRa is not suitable for this specific use case, so there for sure must be something implemented, but then I think it would be better to chose something that does have some pairing involved and radios on board that will work.
These don't have anything present to specifically address a single node and also some safe guards to prevent a neighbor from eavesdropping and sending messages (or replay messages)

Jarppuli, 2020-07-26:
I dont have LoRa wan network in my area, if that is what you mean?
There are no neighbours either in one location, and this would come handy.

If you could just make it possible for wemos + espeasy to communicate with these radio modules, it would open possibilities to use Reyax LoRa, HC-12 or HC-05 because they all communicate via serial communication.
So far I´m in understanding that this is not possible with espeasy.

I wish could use Reyax LoRa modules too, those have encryption and much longer range.
I have succesfully tested two arduino nanos talking to each other using Reyax modules, but that coding part is so difficult for me.
Also, I like the possibility to access espeasy via wifi on my phone, for configuration changes rather than having a computer with me and use arduino IDE as I have to do with arduino boards.

https://www.espruino.com/RYLR