letscontrolit/ESPEasy #3183: sendSysInfoUDP in Networking.ino sends 80 bytes instead of 41 (43) in present mega branch

pm-cz, 2020-07-29:
Hello,
according to the [documentation](https://espeasy.readthedocs.io/en/latest/Controller/C013.html), the sysinfo structure sent via sendSysInfoUDP in Networking.ino should be 13 or 41 bytes. There is an additional extra field of HTTP server port now, so it should be 43 bytes, but it is set to 80 bytes. 

This may be intentional because of some future modification, but the rest of the memory is not initialised, so there is some information leakage inside this reused memory, and the packet is sent via broadcast. So zeroing the memory of the rest of the packet may be a good idea.

TD-er, 2020-07-29:
I've noticed the same when using the rest of the data for a future extention I'm working on.
So yep, it should for sure be zeroed before working on it.

tonhuisman, 2023-09-02:
This should be fixed already, so can be closed.

closed 2023-09-02 by TD-er