letscontrolit/ESPEasy #3186: Security issue - CVE-2020-12638

qwertangel, 2020-08-01:
### Summarize of the problem

A security issue has been discovered in esspressif sdk which allows an attacker to gain man-in-the-middle position.

https://lbsfilm.at/blog/wpa2-authenticationmode-downgrade-in-espressif-microprocessors 

"This vulnerability allows forcing the ESP8622 and ESP32 chip families into downgrade their WiFi authentication mode, effectively disabling their encryption entirely. Using a channel switch attack an adversary can easily gain a man-in-the-middle position and read, replay and manipulate any unprotected traffic of the device. It works by sending a beacon frame with the same data as the WiFi network that the ESP is currently connected to, but switching the Privacy bit in the authentication header to 0. This will cause the Espressif device to switch to the OPEN authentication mode and send out unencrypted traffic until it receives a beacon frame from the original access point again. To stabilize the attack a so called Channel Switch Announcement can be sent to force the ESP chip to switch to a different wireless channel. This way it will not receive the original access point beacons anymore and keep sending unencrypted communication to the rouge access point."

Please check if this issue is applicable to espeasy and require new build.

TD-er, 2020-08-01:
The latest core 2.7.3 should already have fixed it. :)
See: https://github.com/esp8266/Arduino/pull/7486

There has not yet been a  nightly build for it, but it already has been merged.
So the current mega branch has this already taken care of :)

TD-er, 2020-08-01:
I just started the 'nightly' build, so this can already be closed :)

closed 2020-08-01 by TD-er

qwertangel, 2020-08-01:
looks like the binaries didn't make it in 8266 zip file (compile issue ??). can you please check ?

TD-er, 2020-08-01:
> looks like the binaries didn't make it in 8266 zip file (compile issue ??). can you please check ?

Done