letscontrolit/ESPEasy #3188: no bin inside the bin of ESPEasy_ESP82xx_mega-20200801

fly74, 2020-08-01:
only:

\bin

01.08.2020  16:32    <DIR>          .
01.08.2020  16:32    <DIR>          ..
01.08.2020  16:32        16.777.216 blank_16MB.bin
01.08.2020  16:32         1.048.576 blank_1MB.bin
01.08.2020  16:32         2.097.152 blank_2MB.bin
01.08.2020  16:32         4.194.304 blank_4MB.bin
01.08.2020  16:32           276.848 ESPEasy_2step_UploaderMega_1024.bin
               5 Datei(en),     24.394.096 Bytes

tonhuisman, 2020-08-01:
A new set of zip files has been created. Please re-download.

TD-er, 2020-08-01:
Fixed for now, as I generated a build on my PC and added those ZIPs to the release: https://github.com/letscontrolit/ESPEasy/releases/tag/mega-20200801

I will have a look at the Travis build script to see why it doesn't generate the ZIPs.
The recent change to the travis.yml file was to create build binaries in parallel to speed up the build.
But it may be related to the deployment of the ZIP files.

fly74, 2020-08-01:
thumbs up!

closed 2020-08-01 by fly74

fly74, 2020-08-01:
but no Build ist shown for ESP_Easy_mega_20200801_test_beta_ESP8266_4M1M.bin:

System Info |  
-- | --
Unit Number: | 1
Git Build: |  
Local Time: | 2020-08-01 22:02:21
Uptime: | 0 days 0 hours 3 minutes
Load: | 5.84% (LC=5653)
Free RAM: | 14568 (32936 - )
Free Stack: | 3728 (3728 - )
IP Address: | 172.25.99.27
RSSI: | -61 dB (Wlannotforall24)

TD-er, 2020-08-01:
The build script filters out bin files that are too large to flash.
When built on Windows, I have this size:
```
RAM:   [======    ]  56.4% (used 46188 bytes from 81920 bytes)
Flash: [==========]  98.3% (used 1027200 bytes from 1044464 bytes)
```

But on Linux I build it in a subdir which is a bit more characters, so maybe that makes a difference?