letscontrolit/ESPEasy #3192: Travis builds do not generate complete ZIPs

TD-er, 2020-08-02:
To improve build speed and make it way more practical to see what build is failing, I did change the Travis build to use separate jobs.
One job per PlatformIO env to build.
This reduces build times by roughly a factor of 2 and give a clear overview of each build step, so win-win.

Sadly the life time of a job is quite limited, so the generated file is lost as soon as the job has finished.
This does result in rather small .zip files, without any built binary in it.

I can revert back to how it was, running in 1 job, but then we will run into build time exceeded errors.

Another option is to split it into a number of jobs, which each result in a ZIP file to upload. (e.g. ESP32/ESP8266/ESP8285)
But then I loose the easy overview of what build fails and what doesn't.

I can also keep the jobs per bin file, but then it generates a download link per object which makes it harder to use for the end user.

Or I keep it like this and manually create a set of ZIP files to upload manually.

Or maybe someone else has a good suggestion how to overcome this?

tonhuisman, 2020-08-02:
Well, I was going to suggest to use an external storage, and then you added the link to the Travis docs 😄 
One option not mentioned there, and I don't know if that's (still) possible, is to use Dropbox (create a special account for this purpose) as an intermediate storage, and as a last buildstep get a count of the files there (should be a known number, could be determined upfront), and if matched, generate the zip for release.

TD-er, 2020-08-02:
Well if external storage is needed, I do have some servers out there connected to the internet, so I can store data there.

I'm a bit more concerned about a public service needing access to such an account, which is a bit of a security nightmare.
Then it would make sense to create the ZIP files myself.

Other approach could be to make a new Travis stage (stages can have multiple jobs, but a new stage is only entered if all jobs of a previous stage are finished) which does try to download all separate .bin files using wget from Github and then create a zip file.
But I'm not sure yet if the files are already publicly accessible if the build process has not completed yet.

tonhuisman, 2020-08-02:
Yes, the default 'solutions' provided by Travis/Github are quite poor. (I did some reading in the Travis docs as I've not used Travis before, I'm more of a Jenkins' guy, and love the maven/Artifactory combo 😃)

Packing the separately published .bin files using an extra stage does seem like a feasible solution, if it actually works of course.
That's more of a Github thing than a Travis thing, I guess, but as they are separate 'entities', I'd expect the files to be available immediately, why would _that_ depend on the result of the build? If the build failed, then the deploy part wouldn't complete, I'd expect. (But the docs are IMHO rather vague here).

TD-er, 2020-08-02:
The reason it may fail is when Github uses an "eventually consistent" approach of caching.
Meaning a fetch to a random host of the Github servers may be to a different host since the previous request.
Since the Travis jobs may run each on a separate instance, it may also be getting a different IP from a DNS request.
I can imagine the Github servers serve files via a distribution system that will be eventually consistent, which could mean the same file is not (yet) available via all hosts.

tonhuisman, 2022-04-18:
Is this still a valid issue? Can probably be closed now we have a reasonably reliable Github Actions build configuration in place.

closed 2022-04-18 by TD-er