letscontrolit/ESPEasy #3211: SendToHTTP to Domoricz using &param generates weired log entry

Wiki591, 2020-08-14:
Using rules to send

`SendToHTTP 192.168.xxx.yyy,8090,'/json.htm?type=command&param=udevice&idx=46&nvalue=0&svalue=[WZTemp#C];[WZTemp#H];1;[WZTemp#hPa];0'
`

generates log entry 

`611880877: ACT : SendToHTTP 192.168.101.92,8090,'/json.htm?type=command¶m=udevice&idx=46&nvalue=0&svalue=27.0;50;1;1012.54;0'`

Its a misinterpretation of what is really sent

tonhuisman, 2020-08-16:
@Wiki591 Where did you get the log-line from, Serial, Syslog server or Web log?

TD-er, 2020-08-16:
Weblog
(was mentioned on the forum)

tonhuisman, 2020-08-16:
In that case tha cause (bug?) may be in the js code of the web-log viewer

TD-er, 2020-08-16:
Yep it sure is, as it should not allow the browser to interpret the `&para` as is done here.

tonhuisman, 2020-08-16:
I expect it is interpreted in the js code, as the browser may interpret `&para;` but shouldn't interpret `&para`, that would be a huge bug, discovered long time ago

TD-er, 2020-08-16:
Well the Javascipt code does fetch the JSON from the ESP and update it on the screen.
So I'm not sure what you mean by a "huge" bug. As far as I can see, it just is a visualisation bug.

tonhuisman, 2020-08-16:
Well, there is no way that `&para` should be replaced by a paragraph symbol if it's not terminated by a semicolon, `;`, and if a browser would do that, *that* would be a huge bug. That is why I'm suggesting it is probably in the js code for the web-log itself.

TD-er, 2020-08-16:
Well I guess we both share the same allergy then.
Maybe @Grovkillen can have a look?
After all, he is the one that wrote it.