letscontrolit/ESPEasy #3215: Ser2Net for different Serial ports (using ESPeasySerial)

thomasheiser85, 2020-08-16:
### Summarize of the feature request

By using ESPeasySerial in Ser2Net it shall be possible to choose from the different available Serial Ports from ESP32 and also Software Serial for ESP266.

By configuring two devices from Ser2Net ist shall also be possible to gateway diffferent Serials ports from one ESP.

ozett, 2020-09-26:
can anybody show (links?) of some practical use-cases for the serial on esp? 
thx...

thomasheiser85, 2020-11-01:
> can anybody show (links?) of some practical use-cases for the serial on esp?
> thx...

I use the esp to read the power consumption with a NPN phototransistor. And i have two power meters which i want to record with one esp.

Like documented here: https://mwinkler.jimdo.com/smarthome/aktoren-sensoren/stromz%C3%A4hler-digital/

Nikolay-Ch, 2020-11-15:
Hi! I try to use serial port on M5 AtomLite board, but can't...
This board uses ESP32-Pico board, and have only one serial port (serial0) used for update firmware...
I can use other PGIO, like 32 or 26 to software emulation Serial protocol.
But Ser2Net not have any setup option to change default serial port.

So, it would be nice to have the option to configure the Ser2Net to use any pin...

immiimmi, 2021-02-08:
look at the proposed change; it fixed ser2net; it enables multipe ser2net ports at the same time and different pins;
tested only on esp32, but should work also for the older esp8266
https://github.com/letscontrolit/ESPEasy/pull/3483#issuecomment-774238886
immi

tonhuisman, 2022-04-18:
This seems to have been resolved, so issue can be closed.

closed 2022-04-18 by TD-er