letscontrolit/ESPEasy #3219: Sonoff TH16 + Sonoff Si7021 does not works properly

pil2776, 2020-08-17:
I have received Sonoff TH16 and Temp & Hum sensor si7021 from Sonoff also. I cannot get these component working properly with Last firmware of ESPEasy (ESP_Easy_mega_20200812_normal_ESP8266_1M)
here is my configuration on ESp Easy web interface : 

[image]

[image]

Does any one have try this ?

TD-er, 2020-08-17:
On the Hardware tab in your screenshot you can see you have to set pins for the I2C bus.
So make sure to set the SDA and SCL pins.

If you experience instabilities with this plugin (e.g. sensor no longer reports values after a while), can you test the test builds [linked here](https://github.com/letscontrolit/ESPEasy/pull/3209#issuecomment-672906826) ?

TD-er, 2020-08-17:
Hmm I looked at the Sonoff TH + AM2301 I have here.
That one is connected to GPIO-14.
[image]

The remark about I2C is only when you have a true Si7021 connected to an I2C bus.
So please check what sensor you're using. Is it the Sonoff module and if so, what sensor does it use?

thomastech, 2020-08-19:
@pil2776, I don't see a Name assigned to your device plugin (the entry is blank). If for some reason the device name remains blank then perhaps there is data corruption that is causing your problem. If you cannot assign a name then clear memory and reflash the bin. Beyond that, I don't see any other mistakes.

For reference, I'm using a TH10 with AM2301 that is running mega-20200305.  So if you cannot get your TH16 and S7021 to work on the latest release, then try mega-20200305.

#TD-er: The Sonoff S7021 is a "one-wire" device (Vcc, Gnd, Data). It has a 3.5mm TRRS plug that is compatible with the mating receptacle on the TH10 and TH16 modules, which communicate via GPIO-14. 

- Thomas

birosz, 2021-03-19:
hello, did you guys manage to resolve this issue? just flashed a sonoff th16 (v2.1) + Si7021 with this, ESP_Easy_mega_20210223_normal_ESP8266_1M.
the Si7021 wont show any value when added as a I2C device (0.00 / 0.00 values are shown), but if it's added as DHT11/12/22 SONOFF2301/7021 from gpio14 most of the time "nan / nan" shows up and very rarely the measured values so it must be this config but there is definitely an issue somewhere. 

[image]

[image]

thalesmaoa, 2021-11-26:
Any updates? I'm facing the same problem.

tonhuisman, 2023-09-02:
This sensor is supported by both P005 (1-wire mode) and P014 (I2C mode), so this issue can be closed.

closed 2023-09-02 by TD-er