letscontrolit/ESPEasy #3221: Word "OK" at the end of the JSON status GPIO

Gamepigeek, 2020-08-18:
Hello, since the ESPEASY MEGA version, when you check the status of a GPIO, a JSON appears like this : /control?cmd=Status,GPIO,12
`
{
"log": "",
"plugin": 1,
"pin": 12,
"mode": "output",
"state": 0
}

Ok
`

on previous versions of ESPEASY, for example ESPEasy_v2.0.0-dev11, there was no "OK" word at the end of the JSON. This word bothers me because in php JSON_DECODE can no longer function correctly because of the word.

can you tell me why "OK" is here? How do I get my PHP code to work ? My code is

`
$json = file_get_contents("http://mysonoff/control?cmd=Status,GPIO,12");

$array = json_decode($json, true);

if ($array["state"] === 0) {
    echo "OMG my sonoff is OFF";
} else {
    echo 'OH YES my sonoff is ON';
fclose;
}

Thanks x)

ironataerial, 2020-08-18:
#2892

Gamepigeek, 2020-08-18:
> #2892

thanks, my best solution is : 

$json = file_get_contents("http://192.168.1.7/control?cmd=Status,GPIO,12");

$json = str_replace( "Ok", '', $json );

$array = json_decode($json, true);

if ($array["state"] === 0) {
    echo '[image]';
} else {
    echo '[image]';
fclose;
}

closed 2020-08-18 by Gamepigeek

giig1967g, 2020-08-18:
#2778 will fix it.
The PR is in the queue to be merged.