letscontrolit/ESPEasy #3224: Allow multiple instances of I2C plugins to support multiplexer

TD-er, 2020-08-20:
To support the I2C multiplexer, it is even more important to allow multiple instances of a plugin.
So every I2C plugin should be adapted to use a data struct like this:
`struct P028_data_struct : public PluginTaskData_base`

See P028_BME280 as an example.

Todo:
[ ] P006 - BMP085/180 I2C Barometric Pressure Sensor
[ ] P007 - Analog input - PCF8591
[x] P009 - MCP23017 input  (maybe not do this one, as it will also be part of the core GPIO structure)
[x] P010 - Light/Lux - BH1750   (not needed, as it does not keep run time data)
[x] P011 - Extra IO - ProMini Extender  (not needed, as it does not keep run time data)
[ ] P012 - Display - LCD2004
[x] P014 - Environment - SI7021/HTU21D
[ ] P015 - Light/Lux - TSL2561
[x] P017 - RFID - PN532  (But may need some changes as it looks like blocking code instead of a state machine
[x] P019 - PCF8574 input  (maybe not do this one, as it will also be part of the core GPIO structure)
[ ] P022 - Extra IO - PCA9685
[ ] P023 - Display - OLED SSD1306
[ ] P024 - MLX90614 IR temperature I2C 0x5A)
[ ] P025 - ADS1115 I2C 0x48)
[ ] P027 - Energy (DC) - INA219
[x] P028 - Environment - BMx280
[ ] P030 - Environment - BMP280  (is this one still needed?
[ ] P032 - MS5611 (GY-63) I2C Temp/Barometric Pressure Sensor
[ ] P034 - Temperature and Humidity sensor DHT 12 (I2C)
[x] P037 - Display - OLED SSD1306/SH1106 Framed
[ ] P045 - Gyro - MPU 6050
[ ] P047 - Environment - Soil moisture sensor  (very simple change as only the I2C address is kept as a member)
[ ] P048 - Motor - Adafruit Motorshield v2
[x] P050 - Plugin 050 I2C TCS34725 RGB Color Sensor with IR filter and White LED
[x] P051 - Plugin 051 Temperature and Humidity Sensor AM2320 
[ ] P057 - Display - HT16K33 LED
[ ] P058 - Keypad - HT16K33
[ ] P060 - Analog input - MCP3221
[x] P061 - PCF8574/MCP23017 KeyPad
[ ] P062 - Keypad - MPR121 Touch
[ ] P064 - APDS9960 Gesture
[x] P066 - VEML6040 RGBW
[x] P068 - Environment - SHT30/31/35
[ ] P069 - LM75A Temperature Sensor (I2C)
[ ] P072 - Temperature and Humidity sensor HDC1080 (I2C)
[x] P074 - TSL2591 I2C Lux/IR Sensor
[ ] P079 - Wemos Motorshield  (very simple change as only the I2C address is kept as a member)
[ ] P083 - SGP30 - Gas (TVOC - Total Volatile Organic Compounds) and Air Quality (eCO2)
[x] P084 - VEML6070 UV
[ ] P090 - CCS811 Air Quality TVOC/eCO2 Sensor
[ ] P095 - ILI9341 TFT 2.4inches display

tonhuisman, 2020-08-20:
You can remove P095 from the list, as that's actually a SPI-connected device.

TD-er, 2020-09-01:
Added a test build, containing the changes upto P062
See: https://github.com/letscontrolit/ESPEasy/pull/3226#issuecomment-684707482

TD-er, 2020-09-11:
As it appeared the latest test builds may not have been using the actual changes, I made a new test build for this PR:

- [ESPEasy_ESP32_mega-20200829-82-PR_3226.zip](https://www.dropbox.com/s/rfxcdg3p8tobmf6/ESPEasy_ESP32_mega-20200829-82-PR_3226.zip?dl=0)
- [ESPEasy_ESP82xx_mega-20200829-82-PR_3226.zip](https://www.dropbox.com/s/hfmurxc3s457s72/ESPEasy_ESP82xx_mega-20200829-82-PR_3226.zip?dl=0)

closed 2020-09-29 by TD-er