letscontrolit/ESPEasy #3233: Feature Request: Native support for Wemos IR Controller Shield

Wookbert, 2020-08-26:
See https://docs.wemos.cc/en/latest/d1_mini_shiled/ir.html

NOOB question: Any chance to get built-in, interactive support for the neat IR transceiver shield from Wemos?

[image]
[image]

TD-er, 2020-08-26:
And what exactly do you need what is not yet included in the separate IR builds?

Wookbert, 2020-08-26:
Separate IR builds? Where? You are officially allowed to slap me in the face ... NOOB, as I wrote.

Able to control a Wemos Relays Shield (easy peasy), but still looking for idiot proof instructions on how to implement the IR transceiver without coding (both learning IR cmds from existing remotes, as well as sending e.g. via URL. Didn't come across the right manual yet using Google.

TIA for any hint/link!

Wookbert, 2020-08-26:
Did you mean this one?
https://www.letscontrolit.com/wiki/index.php/User:Shardan/IR-Transceiver

TD-er, 2020-08-26:
Have you seen the ["nightly builds"](https://github.com/letscontrolit/ESPEasy/releases) ?
In there are some IR builds, meaning the builds which include IRRX and IRTX plugins.
There are even builds for the 4M flash boards out there and IRext builds which have support for the more elaborate appliances like air conditioners etc.

Wookbert, 2020-08-26:
I'm running Release mega-20200812, which, correct me if I'm wrong, is the latest nightly build.  I found the [Wiki on the IRTX](https://www.letscontrolit.com/wiki/index.php/IRTX), but where can I find a manual of the IRRX?

TD-er, 2020-08-26:
Not sure yet where the "manual" is most up-to date, as we're (slowly) moving/updating documentation from the wiki into the readthedocs format.
I don't use IR plugins myself, so I really don't have a good idea of what the most recent documentation is.

The IR plugins are using [this library](https://github.com/crankyoldgit/IRremoteESP8266). 
Both the library and keeping it in sync with the ESPEasy plugins are updated frequently. Updates are done by @jimmys01 so maybe he knows best where to find the best documentation on these?  (This is also a hit to him, to at least provide some links in our documentation :) )

jimmys01, 2020-08-27:
As for documentation for the Ir library of the ESPEasy, I have done my best to write something (small) for the user to start with.
You will find that at the start of the [TX file](https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P035_IRTX.ino) and [RX file](https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P016_IR.ino)
As for the details of the library are all very well [documented here](https://github.com/crankyoldgit/IRremoteESP8266)

>  built-in, interactive support

If I understand what you mean, no there is no interactive mode and I do not know how someone would go about doing it.
The closest to what you want [is this](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRMQTTServer) 
The potential of this library is enormous but it seems that IR stuff is not easy for the average tinkerer. (RF even more so)

Wookbert, 2020-08-27:
Actually, what I have in mind would be an easy-to-use, interactive integration directly in ESPEasy’s Web UI. Something like a "Learn"-Button to learn/record IR-signals from existing IR remote controls, name them and then assign those, so they can be triggered/send e.g. through a URL string/HTML request.