letscontrolit/ESPEasy #3234: Support fo the HMC5883

Ognik69, 2020-08-28:
Is the addition of support for the HMC5883L magnetometer planned in the next versions? I care about access to clean X, Y, Z data, with the possibility of setting the sampling frequency and the amount of averaged samples. The data would be downloaded by Domoticz.

TD-er, 2020-08-28:
Not yet, but I just added the sensor to my Ali Express shopping basket.
There is still quite a large pile of sensors waiting to be implemented though.

Ognik69, 2020-08-31:
Thank you very much for the info, then I'm waiting for this version.

svn2208, 2021-10-06:
Is there any development or can use this sensor somehow?

svn2208, 2021-11-12:
I am still on it. The dev version is already in my repository; I just need to find some time for a final test.

TD-er, 2021-11-12:
You should leave something left for us to comment on ;)

svn2208, 2021-11-18:
I have some question regarding "AddFormFloatNumberBox". Can I contact you somewhere?

TD-er, 2021-11-18:
> I have some question regarding "AddFormFloatNumberBox". Can I contact you somewhere?

Sure, see my profile page :) https://github.com/TD-er

PR #3852 (2022-01-09): Development/Test Version for HMC5883L
Resolves #3234
closed 2022-01-09 by TD-er