letscontrolit/ESPEasy #3235: htu21d not sending to controller automatically anymore

KimCarlsen, 2020-08-28:
Hi
After upgrading my wemos d1 mini device, my HTU21d sensor does not automatically send data to controllers anymore.
Previous version, that worked: 20191108
New version that fails: 20200812
All other sensors (switches and DS18b20) on that ESP works fine.

If I manually click submit on the HTU21d device, data is read and published to controllers.
I have tried to change the interval from 300 sec down to 5 and back again without any luck.

Thanks for at great product

TD-er, 2020-08-28:
Hmm that sensor has not changed for a while so it seems.
But it is an I2C device.
Can you test with a lower I2C clock frequency?

KimCarlsen, 2020-08-30:
There is also 9 months between the versions :-)
Anyway. Lowering the I2C frequency from 400kHz to 200, did the trick. But don't know why it started to fail. No hardware changes. Only firmware update.

If you need any log/debug for investigation, let me know. Otherwise issue can be closed.
Thanks for fast reply.

TD-er, 2020-08-30:
Well the core library has been updated in between and it seems the default I2C clock frequency may have changed as there are more reports on badly behaving I2C devices lately.

tonhuisman, 2023-09-02:
This seems to be resolved, so can be closed.

closed 2023-09-02 by TD-er