letscontrolit/ESPEasy #3236: Sending data to Domoticz interrupted

rolfie23, 2020-08-29:
Went from the stable release (R120) to the latest build of the mega (12-08) in the hope the 4digit display now shows negative value good.

But have an problem with this latest build, is suddenly stops sending data to Domoticz. After reboot it resumes for a minimal of 1 day. In the web GUI the good value and the 4 digit display is shown, but Domoticz doesn't receive it. The log only mention:
`494312408: WD   : Uptime 8239 ConnectFailures 0 FreeMem 19408 WiFiStatus WL_CONNECTED
494322867: EVENT: Clock#Time=Sat,05:15
494333412: DS   : Temperature: 21.19 (28-70-52-79-97-9-3-a9)
494333415:  Domoticz: Sensortype: 1 idx: 71 values: 22.9
494333460: EVENT: Aquarium#Temperature=22.89
494333478: ACT  : 7dt,22.9
494333480: 7DGT : Show Temperature=22.90
`

How to solve this?

TD-er, 2020-08-29:
Did you use the old settings, or started over from scratch?
Can you check the Domoticz controller settings and save them again?

What other symptoms to you see when the ESP stops sending data to Domoticz?
Which Domoticz controller do you use? HTTP or MQTT?

rolfie23, 2020-08-29:
The old setting, just made an update trough the web GUI.

Wil save the controller setting again when I'm at home.

I see no other symptoms, only that Domoticz receive nothing any more. That's trough http.

TD-er, 2020-08-29:
Are you using a hostname or an IP address?
When using a host name, is its IP changing every now and then?

Can you also test with sending Gratuitous ARP enabled (Tools => Advanced, bottom of the page) ?

rolfie23, 2020-08-29:
IP adres, and it's static.

Periodical send Gratuitous ARP was already checked.

Did save the hardware controller now for 1 esp.

TD-er, 2020-08-29:
OK, so let me know if it happens again.
Absolutely no idea if it even may work, but maybe worth a try, to check the "SendToHTTP wait for ack" on the Tools -> Advanced page. (at the top of the page)

When enabling it, please check also the first few minutes to see if it even works with this checked.

But rather test one thing at a time otherwise we have no idea what may have fixed it in the end.

rolfie23, 2020-08-29:
For now, first solution didn't work. Will look now for the sendtohttp. My log now shows:
`34442443: WD   : Uptime 574 ConnectFailures 1 FreeMem 16096 WiFiStatus WL_CONNECTED
34472104: EVENT: Clock#Time=Sat,14:50
34472440: WD   : Uptime 575 ConnectFailures 1 FreeMem 19184 WiFiStatus WL_CONNECTED
34493133: DS   : Temperature: 22.31 (28-70-52-79-97-9-3-a9)
34493136:  Domoticz: Sensortype: 1 idx: 71 values: 24.0
34493184: EVENT: Aquarium#Temperature=24.01
34493203: ACT  : 7dt,24.0
34493205: 7DGT : Show Temperature=24.00
34502419: WD   : Uptime 575 ConnectFailures 1 FreeMem 19152 WiFiStatus WL_CONNECTED
`

TD-er, 2020-08-29:
Can you check the timeout setting in the controller settings?
Maybe this is a bit too low?
You can also try to remove the controller and add it again, this way we know for sure the used default values.

rolfie23, 2020-08-29:
Client timeout is 100ms, did remove and add it but it remebers all the credentials. Also the GUI is now (since 1 hour) really slow for the both ESP's

TD-er, 2020-08-29:
Can you check with the timeout set to a larger value, like 500 msec?

rolfie23, 2020-08-29:
I changed first an option of no sleep of wifi, if that not works, will increase the timeout!

rolfie23, 2020-08-30:
All off it doesn't work, can I get back to the r120 or other stable release? It's used for my aquarium temperature for example so really necessary to work properly.

TD-er, 2020-08-30:
You can try a build based on core 2.3.0, which was used until February or March 2018. 

So I guess it is wise to go back to the one you actually know is running well on this hardware and maybe use another node to try to find out what is causing the issues here.

tonhuisman, 2020-08-30:
What is your controller setting for `Check Reply` ? and did you perhaps enable `Use Extended Credentials`? (Though you shouldn't have/need that, as the older version didn't allow for extended credentials) (and it shouldn't really make a difference, but it *could*)
Could you switch to the Domoticz MQTT controller?, though that implies you'll also need a MQTT broker next to Domoticz...

rolfie23, 2020-09-01:
Hello, sorry for late reply... Had troubles going back to oler firmware and work and stuf....

The main esp is now van to v2.0-20180322. that one is now online for 1 day and 16 hours without an problem.

The second one is still on version ESP_Easy_mega_20200812_dev_ESP8266_4M1M but that one is now already also for 1 day and 17 hours without an problem.

Check reply is ignore, already tried acknowledge. Use extended credentials is off. I have an mqtt broker, but with the mqtt plugin on Domoticz it feels very unstable so that one is off. The http solution worked already for 3 years fine.

But the strange thing is, that it works now with one older firmware so is it possible that they are bullying eachother?

Sorry for my worse English, hope you understand.

TD-er, 2020-09-01:
It is very easy to understand what you're telling, so no worries 👍 

Sometimes, on some specific nodes, the firmware isn't running very stable regarding WiFi.
No idea why and I've spent many many hours trying to find a fix for it.
In the recent nightly builds, there are some builds with "alt_wifi" in the name.
Those do use slightly different settings and some users report success with them on such specific nodes that would otherwise run not stable at all.
You could try those, if you like.
Just make sure to backup your settings in case you want to revert to what's running now.
Also it is best to do a full erase (write blank image) before flashing one which uses a different core version, just to be sure.

tonhuisman, 2023-09-02:
No response, can most likely be closed.

closed 2023-09-02 by TD-er