letscontrolit/ESPEasy #3244: sketch size too big for esp32

augisbud, 2020-09-03:
Hello, I'm trying to build the latest source code for olimex esp32-poe, but the sketch size appears too big on Arduino Ide, therefore I cannot upload it.

TD-er, 2020-09-03:
What flash layout did you select?
In PlatformIO I have made a partition layout where the sketch and OTA space are the same and SPIFFS roughly 400k
The default layout only has 1.4 M for the sketch, which cannot be used as almost any build for ESP32 is too large to fit, regardless the number of plugins selected.

So please try building using PlatformIO, as building with Arduino IDE has not been tested for ESP32 (at least not by me)

tonhuisman, 2022-04-18:
This issue is no longer relevant, IMHO, as currently only the 1.8MB sketch sizes are supported for ESP32 with 4MB flash, and up to 4MB sketches (MAX builds) can be used on the 16MB flash ESP32 units.

closed 2022-04-18 by TD-er