letscontrolit/ESPEasy #3247: Problem RDM6300 RFID Token not deleted in Release mega-20200829

fraeggle, 2020-09-07:
Enviroment
ESP Wroom02  (Latest Release 20200829, went back to "stable" mega-20200721 
RDM6300

As already in RDM6300 - Reading Interval of rfid token #2982, unfortunately in release mega-20200829 the value is not reset to "0" after reading an RFID card, as e.g. release mega-20200721 still does. So if the card is read several times in a row, the value always remains the same.
As I noticed, the error occurs now since release mega-20200812. 

Releasw mega-20200721:
663860: WD   : Uptime 11 ConnectFailures 0 FreeMem 17536 WiFiStatus WL_CONNECTED
669162: Command: Publish   --> ID of Card send to Broker
669166: ACT  : GPIO,12,1
669167: Command: GPIO
669168: SW   : GPIO 12 Set to 1
669170: ACT  : TaskValueSet,12,2,1
669172: Command: TaskValueSet
669176: ACT  : Publish,ESP_RFID/RFID_OK/status,1
669178: Command: Publish
669195: ACT  : TaskValueSet,10,1,1+1
669197: Command: TaskValueSet
669209: ACT  : Publish,ESP_RFID/RFID_Zaehler/ok,2
669211: Command: Publish
669215: ACT  : timerSet,2,2
669216: Command: timerSet
669646: RFID : Removed Tag
669739: EVENT: RFID#Tag=0		--> ID set to "0"
669819: ACT  : TaskValueSet,12,4,0
669820: Command: TaskValueSet
669825: ACT  : Publish,ESP_RFID/RFID/lesen,0
669826: Command: Publish
671547: SW   : GPIO 12 Set to 0
671549: ACT  : TaskValueSet,12,2,0
671550: Command: TaskValueSet
671555: ACT  : Publish,ESP_RFID/RFID_OK/status,0
671556: Command: Publish
671560: ACT  : GPIO,13,0
671561: Command: GPIO
671562: SW   : GPIO 13 Set to 0
671564: ACT  : TaskValueSet,12,3,0
671565: Command: TaskValueSet
671570: ACT  : Publish,ESP_RFID/Fehler/status,0
671571: Command: Publish
671575: ACT  : TaskValueSet,10,3,0
671576: Command: TaskValueSet

Newest Release mega-20200829:
Reading "new" Card. After that Reading same Card again:
226883: WD   : Uptime 4 ConnectFailures 0 FreeMem 19112 WiFiStatus WL_CONNECTED
230185: ACT  : GPIO,13,0
230187: SW   : GPIO 13 Set to 0
230189: ACT  : TaskValueSet,12,3,0
230192: ACT  : GPIO,14,1
230194: SW   : GPIO 14 Set to 1
230195: ACT  : TaskValueSet,12,1,1
230200: ACT  : Publish,ESP_RFID/RLS/status,1
230204: ACT  : GPIO,12,1
230206: SW   : GPIO 12 Set to 1
230207: ACT  : TaskValueSet,12,2,1
230212: ACT  : Publish,ESP_RFID/RFID_OK/status,1
230229: ACT  : TaskValueSet,10,1,2+1
230243: ACT  : Publish,ESP_RFID/RFID_Zaehler/ok,3
230249: ACT  : timerSet,2,2
230628: EVENT: Clock#Time=Mon,11:51
232365: EVENT: rlsaus
232433: ACT  : GPIO,14,0
232435: SW   : GPIO 14 Set to 0
232437: ACT  : TaskValueSet,12,1,0
232442: ACT  : Publish,ESP_RFID/RLS/status,0
232446: ACT  : GPIO,12,0
232448: SW   : GPIO 12 Set to 0
232449: ACT  : TaskValueSet,12,2,0
232454: ACT  : Publish,ESP_RFID/RFID_OK/status,0
232459: ACT  : GPIO,13,0
232460: SW   : GPIO 13 Set to 0
232462: ACT  : TaskValueSet,12,3,0
232467: ACT  : Publish,ESP_RFID/Fehler/status,0
232471: ACT  : TaskValueSet,10,3,0
236031: RFID : Old Tag: 1542526
236096: RFID : Old Tag: 1542526
236162: RFID : Old Tag: 1542526
236227: RFID : Old Tag: 1542526
236292: RFID : Old Tag: 1542526
236357: RFID : Old Tag: 1542526
236422: RFID : Old Tag: 1542526
236487: RFID : Old Tag: 1542526

But since now the ID is no longer set to 0 after reading, it is not recognized that the card is read again.

TD-er, 2020-09-07:
As far as I can remember, there has been a change in the RFid readers plugins to make sure the plugin would not trigger lots of events when the card was held over the reader.
But those changes were already present in earlier builds, as can be seen here: https://github.com/letscontrolit/ESPEasy/commit/03aad0e44e75fdd42a2d14eeaf572a9f996ff623#diff-f71bfc7a504e67f3284732cdcbd15674

fraeggle, 2020-09-07:
Maybe this ist the issue:
Gijs Noorlander (19):
      [Timers] Cleanup of timer code.
      [Timers] Change interval timer defines into enum class

Because this was changed:
@TD-er have set timer 500ms at end of assign cardid, when timer triggers reset cardid to 0
@uzi18
Contributor
uzi18 commented on 4 Apr

@@ -56,6 +56,17 @@ boolean Plugin_040(byte function, struct EventStruct *event, String& string)
         break;
       }
 
+    case PLUGIN_TIMER_IN:
+      {
+        if (Plugin_040_init) {
+            // Reset card id on timeout
+            UserVar[event->BaseVarIndex] = 0;
+            UserVar[event->BaseVarIndex + 1] = 0;
+            success = true;
+        }
+        break;
+      }
+
 
@@ -133,6 +160,7 @@ boolean Plugin_040(byte function, struct EventStruct *event, String& string)
               log += key;
               addLog(LOG_LEVEL_INFO, log);
               sendData(event);
+              setPluginTaskTimer(500, event->TaskIndex, event->Par1);
             }
           }
           success = true;

Found it on the "old" issue

TD-er, 2020-09-07:
OK, so it was forgotten on the P040 code, and only done on the other RFid plugins?

uzi18, 2020-09-07:
@TD-er maybe it is already done in my other PR

TD-er, 2020-09-07:
@uzi18 What PR?

And @fraeggle I don't really understand what you meant with your last post.
Is that a diff of another merged PR?
The code as you show in your diff is present in the current code.

TD-er, 2020-09-07:
Hmm, just looked through the code and it looks like `PLUGIN_TIMER_IN` is no longer called.

TD-er, 2020-09-07:
It looks like I have made some error in the scheduler where I accidentally changed  `PLUGIN_TASK_TIMER` into `RULES_TIMER`

TD-er, 2020-09-07:
Would be nice if you could try this PR: https://github.com/letscontrolit/ESPEasy/pull/3249

fraeggle, 2020-09-07:
> 
> 
> @uzi18 What PR?
> 
> And @fraeggle I don't really understand what you meant with your last post.
> Is that a diff of another merged PR?
> The code as you show in your diff is present in the current code.

sorry for the misunderstanding.
The code I posted was probably the solution of Bartłomiej Zimoń uzi18 and since I read about timer changes in the changelog of 20200812 I thought it might be related. i did not want to cause confusion...

fraeggle, 2020-09-07:
> 
> 
> Would be nice if you could try this PR: #3249

Ähm How should i try?

TD-er, 2020-09-07:
> > Would be nice if you could try this PR: #3249
> 
> Ähm How should i try?

You could try this test build:

- [ESPEasy_ESP32_mega-20200829-27-PR_3249.zip](https://www.dropbox.com/s/3uy42g0jobkgd53/ESPEasy_ESP32_mega-20200829-27-PR_3249.zip?dl=0)
- [ESPEasy_ESP82xx_mega-20200829-27-PR_3249.zip](https://www.dropbox.com/s/i95pgxeypw0iub8/ESPEasy_ESP82xx_mega-20200829-27-PR_3249.zip?dl=0)

fraeggle, 2020-09-08:
Still same issue. ID not resetet to 0 after read.

Build Time:	Sep 7 2020 21:30:57  -->> from ESPEasy_ESP82xx_mega-20200829-27-PR_3249.zip
Binary Filename:	ESP_Easy_mega_20200907_normal_WROOM02_2M256

77161: WD   : Uptime 1 ConnectFailures 0 FreeMem 19472 WiFiStatus WL_CONNECTED
81966: SW   : GPIO 14 Set to 1
81967: ACT  : TaskValueSet,12,1,1
81972: ACT  : Publish,ESP_RFID/RLS/status,1
81979: ACT  : GPIO,12,1
81981: SW   : GPIO 12 Set to 1
81982: ACT  : TaskValueSet,12,2,1
81987: ACT  : Publish,ESP_RFID/RFID_OK/status,1
82005: ACT  : TaskValueSet,10,1,0+1
82018: ACT  : Publish,ESP_RFID/RFID_Zaehler/ok,1
82022: RFID : Old Tag: 1542526
82025: ACT  : timerSet,2,2
82085: RFID : Old Tag: 1542526
82150: RFID : Old Tag: 1542526
82215: RFID : Old Tag: 1542526
84149: EVENT: rlsaus
84219: ACT  : GPIO,14,0
84221: SW   : GPIO 14 Set to 0
84222: ACT  : TaskValueSet,12,1,0
84230: ACT  : Publish,ESP_RFID/RLS/status,0
84235: ACT  : GPIO,12,0
84236: SW   : GPIO 12 Set to 0
84238: ACT  : TaskValueSet,12,2,0
84243: ACT  : Publish,ESP_RFID/RFID_OK/status,0
84247: ACT  : GPIO,13,0
84249: SW   : GPIO 13 Set to 0
84251: ACT  : TaskValueSet,12,3,0
84255: ACT  : Publish,ESP_RFID/Fehler/status,0
84260: ACT  : TaskValueSet,10,3,0
92380: RFID : Old Tag: 1542526
92445: RFID : Old Tag: 1542526
92511: RFID : Old Tag: 1542526
92576: RFID : Old Tag: 1542526

TD-er, 2020-09-08:
@uzi18 Can you maybe have a look at this one?

uzi18, 2020-09-08:
you catched this 👍

TD-er, 2020-09-08:
Well apparently the one that did test the test build claim it isn't solved (and I thought I removed the remark my PR would fix this issue)...
So it was closed by accident and still an issue.

fraeggle, 2020-09-08:
> 
> 
> Well apparently the one that did test the test build claim it isn't solved (and I thought I removed the remark my PR would fix this issue)...
> So it was closed by accident and still an issue.
No Problem.... 

BTW, quick test.... i used mega-20200829 made your changes from Merge pull request #3249 from TD-er/bugfix/plugin_timer_in to  src/src/Helpers/Scheduler.cpp compiled new and it looks like its working again. Setting to 0 seems to be very fast, faster then before, but looks good. Will test tomorrow again and will give you feedback.

Peter

TD-er, 2020-09-08:
OK, glad it does seem to work.

fraeggle, 2020-09-09:
took a look again :-) and it still works....... Fix in src/Helpers/Scheduler.cpp works.
ID is set to 0 right after you take away the card :-D

only weird thing i had during compile with arduino..   i had to made changes to \src\Globals\NetworkState.h
from:
#include "ESPEasy-Globals.h"
#include "ESPEasy_plugindefs.h"
to
#include "../../ESPEasy-Globals.h"
#include "../../ESPEasy_plugindefs.h"
because Arduino told me error in  \src\Globals\NetworkState.h ... ESPEasy-Globals.h not found.
Seems to be correct, because these two files are not in \src\Globals ....

Thanks for your help and work.....
Peter

tonhuisman, 2020-09-09:
> only weird thing i had during compile with arduino.. i had to made changes to \src\Globals\NetworkState.h

That's already fixed a couple of days ago in the main `mega` branch.

Many ppl switched to VSCode or Atom with Platformio for development. Personally I think the Arduino IDE is quite outdated/clunky and library management is a bit 'messy', to give it a nice name.

fraeggle, 2020-09-09:
> Many ppl switched to VSCode or Atom with Platformio for development. Personally I think the Arduino IDE is quite outdated/clunky and library management is a bit 'messy', to give it a nice name.

I know, but on this PC i had arduino installed and i know that this one works... :-D  I normally use Atom (specially for HM devices)

TD-er, 2020-09-09:
Glad to know the issue in the scheduler was causing this, as it may also have caused other -yet not reported- issues.
So that's at least one issue that can be closed.
I will probably make a new build today to make sure no other issues will be reported related to this :)

commit 7cbe105e (2020-09-08): [Scheduler] Fix PLUGIN_TIMER_IN not being called (#3247)

Fixes: #3247

The wrong timer ID was scheduled, so it was never dealt with.
closed 2020-09-09 by TD-er