letscontrolit/ESPEasy #3251: Contributing a new controller

grodansparadis, 2020-09-08:
Hi,
if I want to contribute a controller (actually three (variants of the same protocol) do I just take the id's (currently 19/20/21) that are in turn in my fork, or is there some assignment process for this?

Grovkillen, 2020-09-08:
I'd prefer them to be one of possible. Select your type using a drop-down.

grodansparadis, 2020-09-08:
OK. No problem. So I can just setup C019 ad hope that no one else uses
it before I do a pull request?

/Ake

Grovkillen, 2020-09-08:
We will approve it before anything being merged so go ahead and use 019

grodansparadis, 2020-09-08:
Sorry that I open this again, just want to make this clear before I go ahead.  In my case  I make a controller for VSCP which can use tcp/ip, MQTT, UDP or multicast. Isn't it a terrible waste of precious space to include all of them in one controller instead of separating them?

TD-er, 2020-09-08:
Please do not use C019, as that's one I'm currently working on.

grodansparadis, 2020-09-08:
Is 20 OK

Grovkillen, 2020-09-08:
I thought you said that they use the same protocol as in only small
differences. If they are completely different then yes, use multiple
controller files.

grodansparadis, 2020-09-08:
I will do the tcp/ip version first anyway so I do than one and then we see. Takes some time to dig into your excellent work before I have something to share anyway. But I just want to try to do it as close to correct the first time as I can. Therefore my stupid questions.

Grovkillen, 2020-09-08:
Not stupid at all, welcome aboard.

TD-er, 2020-09-08:
Please note that I didn't yet add the wrappers for controller queues past 20, so make sure to also add those.
And if the controllers are variant of the same protocol, please make sure to share the code that can be shared (or make it selectable in 1 controller) as it sounds like it would otherwise be a lot of code duplication.
And code duplication not only is a likely cause for bugs and really hard to maintain the code.
Also it does add to the build size and used resources.

So maybe it is a good idea to show a work-in-progress pull request so we can help in making decisions on what code to share.

TD-er, 2020-09-08:
Just couldn't withstand browsing through the links in your mail signature and I guess it would be about [VSCP](https://docs.vscp.org/spec/latest/#/./introduction) ?
If so, then I think it can be a great addition to ESPEasy.

Looking forward to your PRs.

grodansparadis, 2020-09-08:
Yes it is. VSCP turned 20 years some time ago (not a teenager anymore
;-) ) so it is probably on time to walk this path. I think this project
is really great work.  Well organized and useful.  Not to say I
understand everything in the setup yet but at least managed to do a
clean compile with a dummy C020 at least. Will hopefully come up with a
PR eventually.

grodansparadis, 2020-09-08:
Yes I added some missing defines which made the control file compile. 

I go for the tcp/ip version first. It needs the arduino-vscptcpclient
library <https://github.com/grodansparadis/arduino-vscptcpclient>. The
MQTT version uses the pubsubclient lib and as I understand it is
compiled in already. Need some JSON parser maybe (ArduinoJson? Also
always compile din?).  For udp/multicast the code is not that huge
either but there is an option there for AES encryption which may add
some code (or one can skip that part all together).

The tcp/ip and the mqtt variants could be in the same control. Mostly
the lib is what's added.  But I go for the tcp/ip version first and then
we see what road is best. Intuitively it feels like separate versions
would be best. Nut I really can't say at this point. Can change that in
either direction.

I think VSCP is just of minor interest for most people so including it
in a build may need some extra steps for a user needing it. No need for
it to bloat the popular main stream stuff.

Yes I think a work in progress is a must. Maybe I even should ask you
guys to check the fork before I do a PR.

btw for the library, is there a way to bind it's use to the USE_C020
define? Or should it be installed in the lib folder?

TD-er, 2020-09-08:
Please have a look at the files `define_plugin_sets.h` and if you build it using PlatformIO (which I hightly recommend), you can also see the Custom-sample.h (rename to Custom.h) and/or `tools/pio/pre_custom_esp82xx.py`

uzi18, 2020-09-08:
We will definitely check your eventual PR :)

grodansparadis, 2020-09-08:
Thanks. A little overwhelmed at this point.

grodansparadis, 2020-09-08:
:) I there ever is one. I still hope so.

grodansparadis, 2020-09-10:
If I want to handle SENSOR_TYPE_TEMP_BARO, SENSOR_TYPE_TEMP_EMPTY_BARO, SENSOR_TYPE_TEMP_HUM_BARO, SENSOR_TYPE_WIND etc is there a way to determine the unit for the measurements?  [VSCP defined measurements](https://docs.vscp.org/spec/latest/#/./class1.measurement) allow for a default unit (SI or SI derived) so it would be possible to specify that this unit should be used but better of course if the user could use any unit.

Also for SENSOR_TYPE_LONG can I tell type and/or unit?

Hope it is OK that I use this thread as I move forward.

TD-er, 2020-09-10:
I will explain it from a more historic perspective, which hopefully will help you understand why it is named like that.
Sending sensor data to Domoticz was one of the first integrations added to ESPEasy and as such a number of decisions made in the past were based on how Domoticz handles things.
For example, Domoticz does have a "sensor type" temp_hum  (lots of temperature/humidity sensors) and temp_hum_baro (typically the Bosch BME280 and a few more) and temp_baro (typically Bosch BMP280)

If you look at the code for those plugins, you will see those types mentioned there and also in the Domoticz helper functions, you will see some filtering based on those to decide in what order the parameters are output to Domoticz.

For later plugins, sometimes there simply isn't such a type, so it may output something like a type outputting 2 values. (not really descriptive)
I always try to use the most SI-like unit of measure, so you won't see Fahrenheit, but strictly speaking the temperature in Celcius is also not a SI unit of measure.

I do like to add (not right now as it does need a lot of code changes) a unit of measure and also some expected range per output value as it can be very usefult to have some basic idea of a value being within a useful range or not, regardless of the sensor you're using.

SENSOR_TYPE_LONG is almost exclusive used for RFid readers, so it is a value without a unit.

grodansparadis, 2020-09-10:
I think it is OK with the conversions provided. We are all bound by our historical decisions. 

I started with a MQTT version  and have the basics working for that.  It probably is enough to do this version when I think of it. Why confuse people with other transport protocols. Conversions can be done at the receiving end. 

It's some work to go through all plugins and find suitable VSCP events for all data they send out but take it bit by bit. 

Not fully clear over the receiving yet but  hopefully I will get it

All and all I like the work you guys done a lot.

TD-er, 2020-09-10:
You can also have a look at the implementation I made for LoRaWAN, for which I try to compress the data to send over to the TTN network.
Just look at where `PLUGIN_GET_PACKED_RAW_DATA` is called.
See also the misc/TTN directory where the .js files are located for decoding it.

Maybe something like that can also be used to transfer more than the 4 standard values, if it is useful.

grodansparadis, 2020-09-10:
OK thanks. Will have a look.

TD-er, 2020-09-10:
This "compression" entry is a step-up for future expansion where I also would like to do something similar to exchange values between nodes.
Right now we do have the ESPEasy p2p protocol, which is a bit too simple as it only supports sending over 4 floats, while you actually would like to send over all internals to the other side.
And it can be done as the other side also should have the same plugin code present, so they should understand their plugin internals.
For the "compression" you must know what number of bits are needed to apply an offset and/or factor to make sure you don't loose any information, which again comes to the point of having to know the proper range of the unit of measure.
So either this node-to-node exchange could be very plugin specific, or it could become rather generic as you would like to have in your protocol.
This means sending over some structured data containing the unit of measure, the value and maybe some encoding flags. (to interpret the data)
I guess you already have something like that, so that's why I mention it, as it probably already has quite some overlap in our ideas.

grodansparadis, 2020-09-21:
Sorry I missed your comment here. Was deep into other things probably or just out in the blue as always...  

Yes the compressed type is more similar to the approach taken by VSCP. Except for missing unit coding which I would prefer being present. And as always with binary data ending gets important to consider (at least specify) as well. The VSCP way would be to send several events (all small) instead of packing a bunch of data together. But the receiving side always has to know what is coming on the wire and that is the same for all coding so one way is as good as another.

I have implemented something now that works for the more specified types. As you state VSCP is more generic and concentrate on a way to abstract measurements and other things. Data comes from a "device" and have a "well defined" format. A temperature for example is still a temperature if it comes from a ds18b20 or a am2302. And you are probably more interested in the actual temperature then what kind of device that sent it.  And also if it is a temperature a unit must be specified for it to be more than a magic number. The GUID of VSCP events identify the nodes/devices and you must map it to on knowledge of the device on the receiving side (there are means to discover such info also in VSCP).  

VSCP events hold one thing. So for a bmp680 as an example three events will be sent as SENSOR_TYPE_TEMP_HUM_BARO here [](https://github.com/grodansparadis/ESPEasy/blob/vscp-mqtt/src/_C020.ino) One can argue that this is a waste of bandwidth but the gain is higher IMHO and if one find it to much one can go down to VSCP level I and get a VERY low bandwidth utilization but with the same information transferred.

But I am not here to sell in VSCP. I just want this plugin for my own use (and maybe a few others). 

The problem I currently face is mostly that I get an ESPEasy event from "something" and I should encode it in VSCP's stricter form. It would help some if the controller code can get the DeviceStruct for the device that sent the ESPEasy event. Is that possible?

TD-er, 2020-09-21:
The controller code is free to do it, but I guess you should ask yourself if you should want to as it is something you need to keep in sync.
So I guess we should simply add a new function to a plugin, just like the one returning the names of values, but now returning the unit of measure per value. (maybe integrate it with fetching the names)
The event struct does have a few integer fields unused when fetching the names, so I don't see any problem to then store an enum value in it.
And since those calls are per task instead of per plugin, you don't get those strange issues as we now are seeing with the plugins like Dummy and Sysinfo which may change the number of outputs which is a plugin definition and not a task definition. (as it should be)

So please don't refer to the plugin definition but rather think of things to ask the task via its taskindex.

TD-er, 2020-09-21:
By the way, there are lots of controllers sending out messages per value instead of combining them.
Just have a look at the delayhandler code to see how that's being handled.

grodansparadis, 2020-09-21:
I agree that that would be a better solution.

TD-er, 2020-09-21:
Maybe it is also good to realize that a unit of measure also indirectly implies some extra settings.
For example for a GPS coordinate it makes sense to use almost the complete float resolution of 6 decimals, but for some other units of measure, like free memory in Bytes, it is not really useful to have any decimals.

The GPS plugin already does set some preset of number of decimals when an output value is set to be either longitude or latitude.

grodansparadis, 2020-09-22:
Yes that is definitely so. I came to the same conclusion. VSCP (level I) uses a [coding model](https://docs.vscp.org/spec/latest/#/./vscp_measurements?id=vscp-data-coding) that is very effective and works also where payload are as small as eight bytes. integers are for example not limited to the standard types but rather to the number of bytes they are built of. Four units can be coded and index for eight sensors.

TD-er, 2020-09-22:
That's exactly what I've done for the LoRaWAN packed part of sending data.
For the GPS for example I created a 24-bit type which decodes into a float value with a range of -180 ... +180
And for other values too, the required resolution is leading in the number of bits used and the encoder/decoder use offsets and factors to "compress" the data.

grodansparadis, 2020-09-22:
OK.
I have reached the goal of what I wanted to get done so I close this thread now.

closed 2020-09-22 by grodansparadis