letscontrolit/ESPEasy #3256: open HAB MQTT disregards message retain flag

TommoT1, 2020-09-16:
Hi there,

Could it be that the release mega-20200608 is the last one where the retain flag for the MQTT messages is really set?
There after you can check the "Publish Retain Flag:" box but it doesn't produce a retained message at the broker, just a normal one.
Or what I'm doing wrong since 20200608?

Regards Tom

TD-er, 2020-09-16:
Well if it is, then it is for sure a bug.
Around that time I did change some things to how the controller settings are loaded and saved to save on memory resources, so it is possible I messed up there.

TD-er, 2020-09-18:
What controller do you use?

TommoT1, 2020-09-18:
Hi TD-er!
Like the topic says I'm using the Home Assistant (open HAB) MQTT Controller, espeasy ist
running on an esp8266 (esp-01 with 4MB RAM) and talks to a mosquitto broker.
The messages are being sent via rules and it is always, also after reboot.
 I didn't find a PUBLISH command wich allows to choose between retain or normal....

TD-er, 2020-09-18:
So you are sending the MQTT messages via the `publish` command in the rules and not by letting a task send data to the MQTT controllers?

TommoT1, 2020-09-18:
Yes!

TommoT1, 2020-09-18:
to be honest, I don't know  how to do it via a task....

TD-er, 2020-09-18:
> to be honest, I don't know how to do it via a task....

A bit outside the scope of the issue (apparently ;) ), but I guess it could be useful to know it at least.

If you have a controller enabled, at position 1, 2, 3 (multiple options possible), you will see those also appear as checkboxes somewhere at the bottom of a task config page.
N.B. this is only available for tasks that output 1 or more values, like almost all sensors do.
The only plugins that don't are typically display plugins.

When you bind a task to one or more controllers, the task will send the new data to the controller(s).
This is done every time the PLUGIN_READ call returns `success = true;`
And the interval this PLUGIN_READ is called is the "interval" parameter at the bottom of a task config page.

But I will have a look at how the controller does the publish, apparently without it being "enabled" as I think that's where the bug originates.

TommoT1, 2020-09-18:
Thank's for the explanation. I'll try this one too, but for the moment I need it being sent via rules 'cause I  collect several sensor values an publish it in one data string. A raspberry pie is logging such strings and drawing graphs via gplot.

TD-er, 2020-09-18:
That's OK and I'm glad you use it like this, or else the bug would remain unnoticed for quite some time.

Will have a look at it tomorrow.

TD-er, 2020-09-19:
Can you post a screenshot of your controller settings and part of the rules where you call the publish?
I've looked through the code quite a few times, but I can't find where it would go wrong.

When calling publish from the rules, the controller settings are loaded just for the single bool flag MQTT retain.

Also please show a screenshot of the list of all the controllers on the Controllers tab of the web interface. (incl. the enabled/disabled flag)

TD-er, 2020-09-19:
Ah, never mind, found the bug :)

closed 2020-09-19 by TD-er

TommoT1, 2020-09-20:
Cool! Thank you for fixing! Waiting for a new release....

TD-er, 2020-09-20:
Just made a test build of the current mega branch.

- [ESPEasy_ESP32_mega-20200913-72-g023f38a5.zip](https://www.dropbox.com/s/k183w03846szcrc/ESPEasy_ESP32_mega-20200913-72-g023f38a5.zip?dl=0)
- [ESPEasy_ESP82xx_mega-20200913-72-g023f38a5.zip](https://www.dropbox.com/s/iaea3k4zxe98964/ESPEasy_ESP82xx_mega-20200913-72-g023f38a5.zip?dl=0)

Would be nice if you could test it, as a lot has been touched and changed. (all I2C plugins)
It also includes some changes made to long term timers, mainly used for WiFi.

TommoT1, 2020-09-20:
Just uploaded it to two units and it  works again as expected!
Thanks!

PR #3260 (2020-09-19): [MQTT] Fix disregard retain flag using publish from rules (#3256)
The wrong controller settings were loaded.  (Commands/MQTT.cpp)

Fixes: #3256