letscontrolit/ESPEasy #3259: Bug in Plugin 008: RFID - Wiegand

GitCybers, 2020-09-19:
Hello,

there is a bug in the 008 RFID Wiegand plugin:
After entering a code the right value will be send but then the wiegand plugin sets the value back to "0", sends the "0" and in my device the value is "0".
In the old version 20200812 there wasn't this problem.

tonhuisman, 2020-09-19:
Do you have some (info) log from the ESP, when entering a code and a few seconds after that to see what happens next?
What exact version of ESPEasy are you using?

The only change to this plugin since release 20200812 is a fix to the setting to return the (hex) input from the device as a decimal value, to be used when having a 10 key decimal keypad. Assuming you haven't enabled that option, nothing should have changed.
One other thing that did change, but that should already be in your release, is improvements on timers/schedulers. It's not impossible that the timers now working as expected, may have changed the responsiveness of this plugin. 500 msec after a value is read, a new read action is triggered, so possibly a value of 0 will be read. This behavior already existed before 20200812 (AFAICS).
Could it be your previous version, working as expected, was older? As there have been other changes ~April/May.

Possible solution: Add a setting to not send 0 value. (Though I'm not sure this is desirable).

TD-er, 2020-09-19:
There has been a fix in the scheduler as the `PLUGIN_TIMER_IN` was never called. See #3247

ozett, 2020-09-24:
problem still persists today ... 
(always sending a 0 after the key-id)...somekind of annoying...

i will provide some details about espeasy version with next post...

i made a rule to overcome my additonal mqtt message of "0" for the moment...

```
On wiegand#Tag do
  if [wiegand#Tag]!=0
    Publish ESP/004/wiegand/tag,[wiegand#Tag]
  endif

endon
```

ozett, 2020-09-24:
[image]
```

Firmware
--
Build:⋄ | 20110 - Mega
System Libraries:⋄ | ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ |  
Plugin Count:⋄ | 46 [Normal]
Build Origin: | Travis
Build Time:⋄ | Sep 13 2020 00:22:08
Binary Filename:⋄ | ESP_Easy_mega_20200913_normal_ESP8266_4M1M
Build Platform:⋄ | Linux-4.19.104-microsoft-standard-x86_64-with-glibc2.29
Git HEAD:⋄ | mega_fe6e911

```

ozett, 2020-09-24:
i observed also some kind of "drift" ? after 1 hr running the reading is delayed for 2 minutes...

dont know why the rfid resets are happening, but i cannot understand why this happens with the wiegand-protocol option..

[image]

ozett, 2020-09-24:
i have a old firmware which does not show this drift over time.
it runs now over a year or so...

[image]

configuration of my new wemos-d1 👍 
[image]

ozett, 2020-09-24:
to check if it the pin-selection caused the "drift" i changed the pins the new build-version like the old-one
now i will wait 1 hr to check for RFID-reset or "drift"...

[image]

ozett, 2020-09-24:
i tested two wemos-devices over 2 hrs, same gpio pins with wiegand readers...

device1: with build 20110  everthings seems no ok.  (no "drift " to see.. transmits key-ID immediately)
device2: with build 20109 double reading of the same rfid-key seems not to send. only if a different key is inbetween..
i will check this observation on the specific device with  exchange only firmware build 20110 tomorrow..

TD-er, 2020-09-24:
Please don't refer to the "20110" as a build, as that's actually the internal file format version.
The build should be refered to as the date it was built on (when using nightly builds)

ozett, 2020-09-25:
> Please don't refer to the "20110" as a build, as that's actually the internal file format version.
> The build should be refered to as the date it was built on (when using nightly builds)

yeah...i have exactöy this trouble not to see "build" differences clearly on the node-list. 
Maybe the date should be used as "build", or at least additional shown?

[image]

TD-er, 2020-09-25:
Well I was thinking of adding some encoder/decoder function as we only have 16 bits there.
Maybe something like 21000 as base and then #days since <some date> adding to it.

ozett, 2020-09-25:
i look like there is all   you need already in the sysinfo...
why not show the filename with build-day- and also the spiffs-size (helpful info for debugging)? 
or substring of this?
`ESP_Easy_mega_20200913_normal_ESP8266_4M1M`

i am not familiar with git, but maybe use "git-build" somehow for "build"?

[image]

TD-er, 2020-09-25:
Well the data shared via p2p doesn't have a long enough field to include full build info.
But I do agree with your observation about the shown "build" in the node list.
So to keep compatibility with older builds, there should be some kind of encoding using the available 16 bits.

But this is a bit offtopic here.

ozett, 2020-09-26:
a last point off-topic: 
git build-nr is already on first page. maybe easiest to use this...

[image]

## ON-Topic again:
in  `ESP_Easy_mega_20200913_normal_ESP8266_4M1M`  to my observation a **un**changed key is recognized correct.
and also send correct.

ozett, 2020-09-26:
## on-topic
hard-coded send of "0" for key-removal. would be nice to have it configurable: maybe like this:?
[image]

SamNitro, 2021-01-02:
I am of the same opinion as ozett.
Someone who starts new with RFID and does not see an ID tag may be looking for the error elsewhere.

I updated from a 2017 version to the newest and wondered why the tags weren't displayed.

tonhuisman, 2021-01-02:
I've added the options to achieve all this, resulting in this updated Device settings page:

[image]

NB: No easy way to add options to the separate `Send to Controller` groups, as the settings are device specific, but controller stuff is generic.

I'll create a PR for this asap (documentation has to be updated too).

PR #3449 (2021-01-08): [P008] Wiegand add options for Tag removal
- Added options for Enable/Disable automatic tag removal, the value for the Tag when removed, the delay after scanning the last tag before it is removed, and Enable/Disable sending the Removed tag to the controller(s) and event.
- Updated documentation

Partial screenshot:

[image]

Resolves #3259
closed 2021-01-08 by TD-er