letscontrolit/ESPEasy #3267: OLED SSD1306 will blank once the CO2 Sensor CCS811 on I2C is enabled

NorbertRoller, 2020-09-23:
I am using a CCS811 + HDC1080 with the development version ESP_Easy_mega_20200608_dev_ESP8266_4M1M.
If I disable both devices, the display will show the fixed data in 4 lines properly. On boot it also shows the ESPeasy text in the 1st line, before it clears the screen properly.
If I enabe HDC1080 everything is still ok. Values are displayed
If I enable CCS811 and reset the behavior changes. The ESPeasy text in the upper line flashes for 0.5 Second and afterwards the screen blanks. The ESP8266 is still working and is even transmitting data to the Homecontroller.

Both I2C Sensor work fine and update the data in the device tab as set.

The hardware used is a WIFIKIT8 from Heltec
https://heltec.org/project/wifi-kit-8/

I have tried with different developer versions back until March, but it is always the same.

tonhuisman, 2020-09-23:
Have you tried pulling the addr pin on the sensor to high, using a pull-up resistor would be the safest, and selecting the alternate address?
Another option would be to install pull-up resistors on the I2C data and clk lines.
Pull-up resistors should be between 4k7 and 10k.

TD-er, 2020-09-23:
If you perform an I2C scan, what addresses show up?
Also some of those boards need an extra pin for the OLED as sort of reset pin.
Do you have a link to the board schematics? (N.B. the technical Parameters on that site do mix up some specs as they seem to have mixed up specs from an ESP32 board and this ESP8266 board)

NorbertRoller, 2020-09-25:
> Have you tried pulling the addr pin on the sensor to high, using a pull-up resistor would be the safest, and selecting the alternate address?
> Another option would be to install pull-up resistors on the I2C data and clk lines.
> Pull-up resistors should be between 4k7 and 10k.

The board and the oled are integrated on this device. Both I2C Pins are pulled-up by default with 10k.
Adressing I2C is no issue, as I can address all 3 devices. I can read each sensor.

NorbertRoller, 2020-09-25:
> This board does need a reset for the OLED via GPIO-16.

But why would it make a difference if the CCS811 Sketch is loaded long before the reset had happened. The display shows the status line "Espeasy". Even so it is for a short time, it signals that the display was properly rebooted and can display data.

To me it looks like something goes wrong, when the CCS811 Sketch initalises.

NorbertRoller, 2020-09-25:
> If you perform an I2C scan, what addresses show up?
> Also some of those boards need an extra pin for the OLED as sort of reset pin.
> Do you have a link to the board schematics? (N.B. the technical Parameters on that site do mix up some specs as they seem to have mixed up specs from an ESP32 board and this ESP8266 board)
> 
> Edit:
> Found [schematics](https://resource.heltec.cn/download/WiFi_Kit_8/WIFI_Kit_8_Schematic_diagram.PDF)
> This board does need a reset for the OLED via GPIO-16.
> There have been some issues with those OLED boards needing a reset signal.

Strange the OLED display, that is connected to 0x60, is not displayed in the I2C Scan. Anyway I do not see any overlap

This is what is displayed from the sensors
0x40 | SI7021HTU21DINA219PCA9685HDC1080
0x5A | MLX90614MPR121CCS811
0x76 | BME280BMP280MS5607MS5611HT16K33

Enabling the Sketch makes the difference. So it can not be an address overlap as the hardware address continues to be available, when the Sketch is disabled.

NorbertRoller, 2020-09-25:
One more information:
When I disconnect the sensor and have the sketches enabled, the display shows the 4 lines. Just without values.
If I connect the sensors while the ESP8266 is running, the display will continue to show the 4 lines for a while and then it turns off.
So it looks to me like the first proper reading of the CO2 sensor, after it had calibrated causes the screen to blank 
or the first data send causes an issue with the OLED sketch. 
The variable is default. Name: eCO2 and has just 2 decimals and no formulars.

TD-er, 2020-09-25:
Can you test one of the more recent test builds I made?
- [ESPEasy_ESP32_mega-20200913-72-g023f38a5.zip](https://www.dropbox.com/s/k183w03846szcrc/ESPEasy_ESP32_mega-20200913-72-g023f38a5.zip?dl=0)
- [ESPEasy_ESP82xx_mega-20200913-72-g023f38a5.zip](https://www.dropbox.com/s/iaea3k4zxe98964/ESPEasy_ESP82xx_mega-20200913-72-g023f38a5.zip?dl=0)

This build has a merged PR in it where all I2C plugins have been changed to allow for multiple instances.
I also fixed a few minor things while working on it, so maybe it also affects your used sensors. (and then some more plugins will be tested before I make a new nightly build)

NorbertRoller, 2020-09-25:
> Can you test one of the more recent test builds I made?
>
> * [ESPEasy_ESP32_mega-20200913-72-g023f38a5.zip](https://www.dropbox.com/s/k183w03846szcrc/ESPEasy_ESP32_mega-20200913-72-g023f38a5.zip?dl=0) [...]

I tried: ESP_Easy_mega_20200920_test_beta_ESP8266_4M1M.bin

with no change.

NorbertRoller, 2020-09-25:
As I wasn't sure what version you really wanted to be tested, I have testet:
normal, custom_beta, custom and dev as well.
Most do not support the sensor and the 2 that does have the same issues.

Pls note that the ZIP has a date code of 20200913 while all the .bin files have 20200920.

TD-er, 2020-09-25:
That ZIP filename is based on the last tag set in the mega branch.
That's set by the build script.

thomastech, 2020-09-27:
I checked the plugin's source code and saw that it reports log debug messages. Not many, but maybe what's there can help with the troubleshooting. So enabled the serial log (debug level), reboot, and review all the log messages that begin with "CCS811 :"

- Thomas

TD-er, 2020-09-27:
Hmm Thomas mentions P023 (OLED), but there's also the Framed OLED plugin for this type of display.
Which one is used? (couldn't find it in the thread here)

AcidSubway, 2021-01-17:
hi all,

i have the same problem, the ccs811 dont work with the new ESP_Easy_mega_20210114_test_beta_ESP8266_4M1M.bin
I have a pull-upo n vcc to add and wak is on GRD.
|1 | ✔ | Gases - CCS811 TVOC/eCO2 [TESTING] | VOC | I2C | ❶ | SDA: GPIO-4 (D2)SCL: GPIO-5 (D1) | TVOC:NaNeCO2:NaN 
| 2 | ✔ | Environment - BMx280 | BME | I2C | ❶ | SDA: GPIO-4 (D2)SCL: GPIO-5 (D1) | Temperatur:23.98Luftfeuchtigkeit:38.60Luftdruck:1021.21
| 3 | ✔ | Display - OLED SSD1306 | Display | I2C |   | SDA: GPIO-4 (D2)SCL: GPIO-5 (D1) |  
| 4 | ✔ | Gases - CO2 Senseair | SenseairCO2 | HW Serial0 swap | ❶ | RX: GPIO-13 (D7)TX: GPIO-15 (D8) | ppm:743

674213: CCS811 : Error: Failed to get ERROR_ID register.
674725: Senseair: co2 = 742.00
687581: BME280: dew point 9.00C
687596: BME280 : Address: 0x76
687597: BME280 : Temperature: 23.94
687597: BME280 : Humidity: 38.72
687597: BME280 : Barometric Pressure: 1021.21
694213: CCS811 : Error: Failed to get ERROR_ID register.
694725: Senseair: co2 = 741.00
702131: WD   : Uptime 12 ConnectFailures 0 FreeMem 17856 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init

plz halp me, whats the prob?

TD-er, 2021-01-17:
Do you have all I2C devices with an unique address?
Can you check with an I2C scan to see if you see all devices you'd expect?

AcidSubway, 2021-01-17:
yes all there,

I2C Addresses in use | Supported devices
-- | --
0x3C |  0x5A | 0x76

the senseair is on

Serial Port: | I2C SerialHW Serial0: GPIO-3 (D9) ← TX / GPIO-1 (D10) → RXHW Serial0 swap: GPIO-13 (D7) ← TX / GPIO-15 (D8) → RXHW Serial1: - ← TX / GPIO-2 (D4) → RXSoftwareSerial
-- | --

AcidSubway, 2021-01-17:
i connect now the add pin with 10K on GRD and select in config

I2C Address: | 0x5A (ADDR pin is LOW)

222368: WD : Uptime 4 ConnectFailures 0 FreeMem 16864 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
226814: CCS811 : tVOC: 0, eCO2: 406
234618: Senseair: co2 = 745.00
238818: BME280: dew point 8.76C
238829: BME280 : Address: 0x76
238829: BME280 : Temperature: 24.08
238830: BME280 : Humidity: 37.77
238830: BME280 : Barometric Pressure: 1021.50
246814: CCS811 : tVOC: 0, eCO2: 400

but the tvoc sai = 0

hm

TD-er, 2021-01-17:
What do you mean with "GRD"?

The pull-up resistors should be between SDA & 3v3 and SCL &3v3.
10k is often enough, but for longer cables you may need to lower the pull-up resistors value to something like 4k7.
Or lower the I2C clock frequency for the task that's experiencing issues.

AcidSubway, 2021-01-18:
Sorry GRD = GND

i have this Modul

https://www.amazon.de/CCS811-Carbon-Monoxide-Quality-Digital/dp/B07194XK25/ref=sr_1_5?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=ccs811&qid=1610915287&sr=8-5

I think that the pullups are already installed.