letscontrolit/ESPEasy #3272: feature request: new plugin to support ESP32 cam

TungstenE2, 2020-09-27:
Hi all,

how about having a plugin to support ESP32 cam?

Just got my ESP32 cam 2 days ago.
In general I really like it, and I think there might be some nice project ideas coming up.

What I currently do not like: 
- the version I purchased has no USB like D1 mini, so power is an issue.
- connecting external antenna is a bit tricky.
- No nightview, as not IR LED and no IRcut.
- built in LED flash does not work with standard Arduino IDE setup.
But there is a project for this:
https://github.com/xenpac/ESP32-CAM-with-LED-control

- standard setup does not provide multi client streaming support.
But there is a project for this:
https://github.com/arkhipenko/esp32-cam ... ulticlient

Would be great to have both projects combined...

For ESPeasy I could think of the following:
- having a generic plugin with some basic options, not all possible I really need, leave most of them in auto mode
- having a video stream for multi clients
- option for free GPIO usage, eg PIR sensor triggering

Here are some details from the forum:
https://www.letscontrolit.com/forum/viewtopic.php?f=5&t=6710&p=47008#p37009

As I am not a developer I can only provide some ideas and hoping somebody with skills will pickup the idea... ;-)
@thomastech would you like to support with your dev skills?

TungstenE2, 2020-09-28:
found a better project for reference:
https://github.com/easytarget/esp32-cam-webserver

thomastech, 2020-09-28:
> Found a better project for reference:

That one looks interesting. I downloaded it today and cannot get it to compile on my Win10 workstation (several missing struct members). Did you get it to compile?

BTW, are you sure there's enough flash space to fit the camera related code into ESPEasy? I only ask since I haven't put ESPEasy on a ESP32, so I don't know much about its surplus memory.

Lastly, the ESP32 Camera code takes up a lot of flash space. I have a feeling that even with a hefty hammer we would run into problems stuffing it into ESPEasy. Especially if it includes facial recognition. 

So maybe it would make more sense to add communication (P2P, MQTT, or HTTP) to the ESP32 Camera so it can interact with an ESPEasy device.

- Thomas

TD-er, 2020-09-28:
There are ESP32 boards with 8 or 16M flash.
Also the ESP32 doesn't have such a limited sketch size limit as the ESP8266 has.
Right now we only hit the boundary of the partition size I defined myself. (on 4M units)
For larger flash sizes we can also use different partition layouts.

TungstenE2, 2020-09-28:
yes, got it compiled, but took me some time to find out, as I am new to Arduino IDE.

steps:

- load zip
- unpack
- load esp32-cam-webserver.ino into Arduino IDE
- IDE gives a warning that folder esp32-cam-webserver is missing and that esp32-cam-webserver.ino should be moved to it
- renamed myconfig.h
- copied all files and folders to esp32-cam-webserver as well
- opened all files in Arduino IDE as tabs
- compiled with AI Thinker ESP32-CAM board selected and Port choosen

If ESPeasy might not be able to handle I can hardly judge. From my point of view face detection has low priority, but video streaming, flash light and still images would be great.

TungstenE2, 2020-09-28:
I think this board has only 4MB

TD-er, 2020-09-28:
What build size do you now have?

thomastech, 2020-09-28:
@TD-er, thanks for the insight. Adding the OV2640 camera module to a generic 8/16MB ESP32 would be painful (too many GPIO wires). However, I looked around and found an ESP32 Camera board with 8GB for $20 shipped. 

FWIW, I think a more appealing solution would use the ESP-32 Cam module ($7 shipped), which currently is a 4MB device. That's the board discussed here.

@TungstenE2, the developer released more new files today. I re-uploaded the revised distribution and updated the ESP32 board library. Now it compiles with zero errors and no warnings.

- Thomas

TungstenE2, 2020-09-28:
> 
> 
> What build size do you now have?

Der Sketch verwendet 2217022 Bytes (70%) des Programmspeicherplatzes. Das Maximum sind 3145728 Bytes.
Globale Variablen verwenden 61032 Bytes (18%) des dynamischen Speichers, 266648 Bytes für lokale Variablen verbleiben. Das Maximum sind 327680 Bytes.
esptool.py v2.6
Serial port COM5
Connecting....
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: c8:2b:96:a1:e3:24
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 4096.0 kbit/s)...
Hash of data verified.
Compressed 17392 bytes to 11186...
Wrote 17392 bytes (11186 compressed) at 0x00001000 in 0.3 seconds (effective 541.4 kbit/s)...
Hash of data verified.
Compressed 2217136 bytes to 1693866...
Wrote 2217136 bytes (1693866 compressed) at 0x00010000 in 39.3 seconds (effective 451.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 119...
Wrote 3072 bytes (119 compressed) at 0x00008000 in 0.0 seconds (effective 1638.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

TD-er, 2020-09-28:
2.2 MB means it is already not possible to perform an OTA update, or we must use some tricks via a small 2nd sketch that can be booted just to have an OTA web server running.

After 17th of October I do have time to look into it, as it looks nice indeed.  (On the 17th the Delsbo Electric event is held)

thomastech, 2020-09-28:
`2.2 MB means it is already not possible to perform an OTA update`

The developer has plans to remove all the facial recognition code. This is intended to free up space for implementing OTA. But I have a feeling that ESPEasy users would like to use facial recognition, so this change might not be ideal for everyone.

@TD-er, good luck at Delsbo Electric. I hope your team does well.

- Thomas

TD-er, 2020-09-28:
I'm not in a team, I'm building the measurement equipment :)
Well the measurement boxes are the same as last year, but now there's going to be a piggyback module being fed via ESPEasy-now ( ;) ) and transmitting it via LoRa to the start/finish for the live event.

thomastech, 2020-09-29:
> I'm not in a team, I'm building the measurement equipment :)

@TD-er, That makes you captain of the measurement team. :)

> What I currently do not like:
>     the version I purchased has no USB like D1 mini, so power is an issue.
>     No nightview, as not IR LED and no IRcut.
> snip .. snip

@TungstenE2, I agree with your list of grievances. Overall, the ESP32 camera board seems to be a novelty item that needs improvement. As a security camera, I'd give it a low score. But it could be useful for some home automation applications.

I ordered a generic IR Cut filter mechanism that accepts common M12 lenses. Hopefully I can hack the OV2640 camera to support day/night modes. 
https://www.aliexpress.com/item/32853565318.html

- Thomas

TungstenE2, 2020-09-30:
my current test setup...

[image]
[image]

next plan is to have a cam in a bird nesting house next spring. The LED flash light is great for this.

But I found the 120 or 160 degree lens much more usefull than the 70 degree.

TungstenE2, 2020-09-30:
btw, I found this information that 5,4v would better for stable wifi and a 220uF capacitor soldered to 5v and GND.

https://nachbelichtet.com/esp32-cam-probleme-beheben/

[image]

[image]

ozett, 2020-09-30:
any progress to spot on intergrating the cam as  a device/plugin in espeasy(32) ?

ozett, 2020-09-30:
also a way to integrate the cam: -> https://github.com/donny681/ESP32_CAMERA_QR

plin2, 2021-02-03:
+1 for the request

8666, 2021-02-08:
Would be a nice addition to a weather station with EspEASY, take cam screenshot with wide lens ( I think a picture can be sent over MQTT). Or just run in parallel like a cam server

darki2002, 2021-02-16:
Another:
+1 for the request

For interested or for help to integrate:
Tasmota has support for the ESPcam:
https://tasmota.github.io/docs/ESP32/

I already have two cams running and it works fine. But I would like ESPEasy more... :)

SebNania, 2021-02-20:
Hi all,

Thanks a lot for making esp-easy for all of us,
I wait for this integration :)
In my case I've buy an ESP32 CAM (AI Thinker) and put in this in : https://github.com/projetsdiy/ESP32-CAM-Projects
It's a really basic program, with all I need => A JPG capture (Domoticz / Download for doing time-lapse) and a stream
We could access to it by theeses two URL :
- http://IP/jpg/image.jpg
- http://192.168.0.19/stream

have a nice day,
Sebastien

SingingDwarf, 2021-06-19:
> @TungstenE2, I agree with your list of grievances. Overall, the ESP32 camera board seems to be a novelty item that needs improvement. As a security camera, I'd give it a low score. But it could be [...]

@thomastech Did you have any joy in getting the IR cut filter working with the OV2640 and ESP32-CAM?

thomastech, 2021-06-21:
> @thomastech Did you have any joy in getting the IR cut filter working with the OV2640 and ESP32-CAM?

Some joy, some heartache. 

I did a quick mock-up of the IR Cut Filter concept using a 3D printed sensor adapter and 2.8mm lens (M12 glass optics). See photos below. Here's what I learned:

1. The stock CMOS imaging sensor's lens has some sealant on it to lock it in place. But I was able to unscrew it without too much trouble. Its IR filter is mounted on the lens and not the sensor, which is a relief.

2. I installed a 2.8mm M12 glass lens in the IR Cut Filter. The CMOS sensor (without stock lens) was mounted on the IR Cut Filter using a tiny 3D printed adapter.

3. The sensor adapter fit like a glove on the back of the IR Cut Filter. But it located the new glass lens too far from the sensor's face, which prevented proper focus. I was able to achieve good focus by removing the adapter and loosely positioning the sensor so it was about 1mm away from the filter's moving shutter. BTW, I didn't take the time to update the 3D printed sensor adapter. I planned to fully remove the imaging sensor's housing (just bare sensor with flex cable) to reduce its height so the completed assembly would achieve correct focus. But that task is still open.

4. The IR Cut Filter is moved in/out by the polarity of a low voltage coil (magnetic operation). It requires 3.3 to 5V at about 170mA. A short voltage pulse is all that is required to change position. For my tests I just swapped alligator clips to change filter positions; GPIO control would simply require a low current H-bridge switch.

Conclusion: Adding an IR Cut Filter is certainly possible. But I currently don't plan on any further work on it. The ESP32-CAM's performance is too low for my intended camera projects. Hopefully my reported findings will inspire someone else to do it.

- Thomas

[image]
[image]
[image]

8666, 2021-06-22:
This camera is more like a toy, not for security at all. Needs a lot of light to get some good frame rate. Higher resolution also results in low framerate. Best option for me was only 800x600

What I found annoying is that some lenses are "rotated" so you should mount the ESP sometimes vertically and sometimes horizontality to get proper orientation

fly74, 2022-06-30:
Long ago but I'm interested to give it a try if there is a espeasy build with support for it.