letscontrolit/ESPEasy #3274: SDS011 dont work correctly

Tartoche, 2020-09-28:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ X] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...ESP_EASY_mega_20200913_normal_ESP8266_4M1M.bin

### Steps already tried...
- [X] Tried a clean install (empty `.bin` files are included in the ZIP)
- [X] Tested previous/other build (mention which one already tested)
- [X] Tested on other node to make sure hardware isn't defective.
- [X] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request

The SDS011 dont send value. I have no value or only 1 per 10 hours, I dont undenstand
I take different node and version off MEGA
I verifie the connection TX / RX ... it's OK
Mode Sleep dont work 
With The Nova PM2.5 Sensor Software it s ok i have values
Can you help me please

Sorry for my english, i am French

TD-er, 2020-09-28:
Have you tested also with an older build?
Just to be sure it isn't something introduced recently and if so, to make sure the connection is OK.

Also to what pins is the sensor connected?

Tartoche, 2020-09-29:
I have tested different build and different node and i have the same problem
I find a solution !
The problem is when i connect the TX on the node : the sensor dont work
If i connect only RX (on D7 for example) and no TX on the node, all is ok
When i connect TX  on D8 i have the problem
If i connect TX on D6 : all is ok
D8 is a wrong connection on NODEMCU or WEMOS D1

Tartoche, 2020-09-29:
In my experience i have only the SDS011 installed on a new system fron scratch

TD-er, 2020-09-29:
See the [documentation on which pins to use](https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp8266)
D8 is GPIO15, which is pulled down via a resistor.

You should always be a bit careful with that specific pin, as it also may cause the ESP not to boot if it is pulled high via an external sensor.

closed 2020-09-29 by TD-er