letscontrolit/ESPEasy #3278: ESP Easy firmware not working on Wemos Lolin32

HosmaxOfficial, 2020-09-29:
Hello. Since some weeks I've been trying to flash ESP Easy on my Wemos Lolin32.
When I started, I got different errors like espcomm_send_command: didn't receive command response or similar. After some time of searchs I tried to change the micro USB cable for a good one. These days it arrived and I tested it. It works with normal code on VS Code (where I tested the board and got some of these errors), but when I try to flash via ESP Easy Flasher I always get the error I described. I also tried with the espressif flashing tool. It downloads and finish but I don't see any result.
I have tried to flash different releases to see if someone works. After some forums, the R20100 works on this board. Now, the problem is the sensor I want to use. I would like to use the SDS011 sensor but in this release I can't see it.
I'm not an expert flashing boards and I tried to check everything before posting the issue.
Could you help me?
[image]

tonhuisman, 2020-09-29:
(This (support-like) question had better be asked on the [forum](https://www.letscontrolit.com/forum/index.php), as this place is intended for software issues.)

The ESP32 needs the bootloader and other stuff to be flashed at specific locations, but the current Flasher GUI doesn't have support for multiple files to be flashed in 1 run. That specific layout is supplied in the ESP_Easy*-factory.bin files, so if you would flash the `ESP_Easy_mega_20200929_test_ESP32_4M316k-factory.bin` (current latest release) it should boot up as expected.
Once you have it running, you can upload a newer version using the Update Firmware feature of the Tools page, this time uploading the .bin file _without_ factory in its name. That also avoids the 'juggling' of the flash/boot buttons to get it into flash mode 😉

TD-er, 2020-09-29:
The ESP32 cannot be flashed using this flasher.

This is also mentioned in the [README.txt in the ZIP file](https://github.com/letscontrolit/ESPEasy/tree/mega/dist).

> For flashing ESP32 you need Espressif's own Flash Download Tools.
> The latest version can be downloaded from: [https://www.espressif.com/en/support/download/other-tools](https://www.espressif.com/en/support/download/other-tools)

HosmaxOfficial, 2020-09-29:
I'm sorry, I'm new in this stuff. Next time I'll use the forum.

> (This (support-like) question had better be asked on the [forum](https://www.letscontrolit.com/forum/index.php), as this place is intended for software issues.)
>
> The ESP32 needs the bootloader and [...]

About this, do you mean that I can flash it with the ESP Easy flasher just using this file? If so, I have just tried it and it doesn't work (same error). 

> The ESP32 cannot be flashed using this flasher.
>
> This is also mentioned in the [README.txt in the ZIP file](https://github.com/letscontrolit/ESPEasy/tree/mega/dist).

I read the file before asking, but I was confused because I don't know which files to add neither which addresses to choose.

TD-er, 2020-09-29:
The "factory" file contains all, so it can be flashed starting at address 0 via the serial port.
The file name for ESP32 without the word "factory" in it, is meant to be flashed OTA (via the web interface)

HosmaxOfficial, 2020-09-29:
> The "factory" file contains all, so it can be flashed starting at address 0 via the serial port.
> The file name for ESP32 without the word "factory" in it, is meant to be flashed OTA (via the web interface)

It worked! Now I can find the SDS011. The values are 0 but the reason for this issue was the flashing process. Thank you a lot.

closed 2020-10-04 by TD-er