letscontrolit/ESPEasy #3281: Anybody tried to get USB Host running on Espeasy?

ozett, 2020-09-30:
i wanted to connect a usb-hid scanner to espeasy.
Anybody here tried to get any usb-device running with espeasy?

like over a usb-host-mini board like here: -> https://www.hackster.io/139994/plug-any-usb-device-on-an-esp8266-e0ca8a

USB Host Mini (clone version), the chip on it (MAX3421E) 
[image]

tonhuisman, 2020-09-30:
This could better be discussed on the [forum](https://www.letscontrolit.com/forum/index.php) and when all is clear, bring it here with a well defined feature request.

ozett, 2020-10-03:
@tonhuisman 
your answer made me search the forum 
(as some years ago in the first days without github, still lots of places to question for help) -- 

and brought me this [tread](https://www.letscontrolit.com/forum/viewtopic.php?f=25&t=6237&p=34741&hilit=hid#p34741) that lets me now take into consideration that "RPI-Easy" as a new way to try...

tonhuisman, 2022-04-18:
I think this is solved by switching to RPiEasy, so not needed here (no follow-up or further replies), I assume?

TD-er, 2022-04-18:
ESP32-S2 and other new ESP32 versions do also have some nice USB features. At least capable of implementing USB slave devices like HID devices.
Not sure if it also supports USB host functionality.
As soon as those functions will be looked into, I will also implement those for ESPEasy.

closed 2022-04-18 by TD-er