letscontrolit/ESPEasy #3290: 3x No sensor value for TaskIndex: 11 varnumber:  in log

urkelbundy, 2020-10-03:
Hi all,

With build ESP_Easy_mega_20200929_normal_ESP8266_4M1M I noticed these lines in the log:
No sensor value for TaskIndex: 11 varnumber: 1
No sensor value for TaskIndex: 11 varnumber: 2
No sensor value for TaskIndex: 11 varnumber: 3

This doesn't happen with build ESP_Easy_mega_20200913_normal_ESP8266_4M1M and many builds I have used before.

It seems to be related with the uptime I send to domoticz. Because the uptime is sent to domoticz every 60 seconds I see these lines appearing in the log every minute.
If I disable this task then there no log entries anymore.

The log shows these messages for TaskIndex 11 while the uptime task is on position 12. Not sure if this is normal behaviour.

[image]

Thanks for looking into this.

TD-er, 2020-10-03:
That's a bug I already solved in the mega branch, but not made a nightly build for yet.

TD-er, 2020-10-04:
Can you try this test build I just made:
- [ESPEasy_ESP82xx_mega-20200929-22-g4c332174.zip](https://www.dropbox.com/s/92maswz93cww11j/ESPEasy_ESP82xx_mega-20200929-22-g4c332174.zip?dl=0)
- [ESPEasy_ESP32_mega-20200929-22-g4c332174.zip](https://www.dropbox.com/s/n1lyowdzksxwwjn/ESPEasy_ESP32_mega-20200929-22-g4c332174.zip?dl=0)

urkelbundy, 2020-10-04:
Trying, but using OTA using ESP_Easy_mega_20201004_normal_ESP8266_4M1M.bin the device became unresponsive
Will power it down and up again to see if it is still functioning.
If not then I will serial flash it.

urkelbundy, 2020-10-04:
After power cycling the device the 20200929 firmware was still present. Tried OTA upgrade again and this time it worked.
The reported issue seems to be solved.

Thanks and keep up the good work

TD-er, 2020-10-04:
I've also noticed lately the ESP8266 frequently fails to perform the OTA update the first time you try to update.
A second attempt does always work.
I have no clue why it fails to OTA.

Regarding this issue, it can now be closed.

closed 2020-10-04 by TD-er