letscontrolit/ESPEasy #3291: 1-pin RX input only (9600b8N1) - What plugin/device/pin to use?

ozett, 2020-10-03:
i have now a new RFID Reader-module for our cats, which sends only TX with 9600baud,8N1.
i am a bit offtrack how to wire this, and what device/pin-settings to use. anybody a suggestion? 

can i ommit the TX/RX dialog without protocoll-errors? Or do i neet something more bit-banging?

 https://www.element14.com/community/thread/74679/l/how-does-tx-only-serial-communication-actually-work?displayFullThread=true
[image]

## the reader is with this sending-only TX like some lidars...

[image]

ozett, 2020-10-03:
maybe set the TX and RX pin to same value works for than receiving only?

https://forum.arduino.cc/index.php?topic=112013.15
[image]

ozett, 2020-10-03:
are there pin-changed-interrupts usable in espeasy for a RX-only serial?

--> Nick Gammon made a arduino lib, maybe something like this is alread usable on espeasy?
-> https://forum.arduino.cc/index.php?topic=112013.30
SoftwareSerial (including the read-only version) is based on pin-change interrupts.

thomastech, 2020-10-04:
> i am a bit offtrack how to wire this, and what device/pin-settings to use. anybody a suggestion?
> can i ommit the TX/RX dialog without protocoll-errors? Or do i neet something more bit-banging?

Interesting questions and comments. But the Github issue tracker is the wrong place for general ESPEasy discussions. Instead, you should be posting here:
https://www.letscontrolit.com/forum/viewforum.php?f=2&sid=7a521548eecf1fd5cb2c2455a453e497

FWIW, I can't think of an existing plugin that would accommodate your serial sensor; A custom plugin will most likely be a required solution. Perhaps the P053 PMSx003 plugin could be used as a reference template for it. Furthermore, if your sensor does not receive data then I suggest assigning -1 to the plugin's TX pin.

- Thomas

TD-er, 2020-10-04:
SDS011 is an example of a plugin which does have the ESP's TX pin optional.

uzi18, 2020-10-04:
Any chance for a link to datasheet to this rfid module?

ozett, 2020-10-05:
@all: thanks for tips and comments, i will go today and give all a  try ...

@uzi18 : i will provide a datasheet here -  if i find any while getting this to work

@thomastech: the forum is a big place and has lots of sections. sometiems difficult to decide where to put as hardware, software or general. but next time i question the forum first, than after this got here. maybe this now goes here as a Feature-Request?

@TD-er : the Gases-Plugins seems to have the needed options.. but i will also try GPS to get the key-id for the reader via serial.
i may allowed to come back if i need more help on getting the animal-RFID to work..? thx up to now for all

[image]

TD-er, 2020-10-05:
Well you can also make a pull request, then we're discussing code, which is more GitHub like :)

ozett, 2020-10-05:
> Well you can also make a pull request, then we're discussing code, which is more GitHub like :)

thx for encouringin me to code something, but i am afraid this is far beyond my skills...

i would only code as a newbee if thats the last exit for me to get anything to work - and would than need a professional to improve all my simplest snippetings...

uzi18, 2020-10-05:
So please provide datasheet or link to module so we will know more about.

TD-er, 2020-10-05:
Remember, we all started as newbee.
And I continously go over the code to make it more uniform, so don't be afraid to submit code.

The worst that can happen is that you may learn how to code :)

ozett, 2020-10-05:
@uzi18 : the reader came from aliexpress. its somekind of a black-box.
 if i find more, i will post here. at the moment this is all i have:

start of reader-idea, working example:  [youtube](https://www.youtube.com/watch?v=y89yJ1Fq-hQ&list=LL9HzKw8KoiO240yW4vKURaA&index=673)
on aliexpress: [animal-rfid-Reader store ](https://www.aliexpress.com/store/group/134-2Khz-Reader/427560_510966273.html?spm=a2g0o.store_home.smartGrouping_836231060.510966273), actual offers: [one](https://www.aliexpress.com/item/32860165792.html?spm=2114.12010612.8148356.30.597c31d81q4Xhk), [two](https://www.aliexpress.com/item/32851899801.html?spm=2114.12010612.8148356.26.597c31d81q4Xhk)

[image]

uzi18, 2020-10-05:
@ozett thx, asked shop for docs.

ozett, 2020-10-05:
did a quick test, module doesnt draw current...and does nothing...
maybe already dead on arrival? 

no signals to see on serial, ...dont know how to check if pinout is correct on this blackbox

with no action from the module. the trip ends here...
[image]

maybe i must start again with the module from the youtube video:
[image]

tonhuisman, 2020-10-05:
> actual offers: [one](https://www.aliexpress.com/item/32860165792.html?spm=2114.12010612.8148356.30.597c31d81q4Xhk), [two](https://www.aliexpress.com/item/32851899801.html?spm=2114.12010612.8148356.26.597c31d81q4Xhk)

> Quiet expensive.

Link 'one' points to a 10-pack of these receivers (scroll down, there's a content list), link 'two' points to a single item, but still, not ali-cheap 🤔