letscontrolit/ESPEasy #3294: Generic HTTP command URL with Controller User & Password?

Wookbert, 2020-10-06:
Noob question/request: Am I correct that as of version ESP_Easy_mega_20200812_normal a Generic http command can not be protected by requiring to submit the Controller User & Password (as defined in the Controller Settings > Credentials)?

What I would expect is something like
http://10.0.1.51/control?cmd=GPIO,5,1&u=maxheadroom&p=3hd7dsdg

I understand that the transmission isn't encrypted, so people with access to the local network can theoretically read user and password when listening to the network traffic.

But demanding user and password to execute the http URL in combination with limiting the acceptance to senders in the local subnet, is still better than nothing.

Am I missing something here?

See also Forum topic ["HTTP control without password???"](https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=6702&p=45150)

TD-er, 2020-10-06:
Not sure if I completely understand what you're asking here.

If you want to send a command, there is some filtering on the source of the command and how much damage it can do.
This PR is intended to extend this filtering: https://github.com/letscontrolit/ESPEasy/pull/3109

Apart from that, there is no support for HTTPS, so even if things are protected via a login, you still can sniff the password and do things you are not allowed to do.

But this is not a new feature since 20200812, as it exists already for a long time.

Is that an answer to your question?  (by the way, this is not really suitable as an issue on Github, more for the forum)

Wookbert, 2020-10-07:
I later on realized that this isn't the place to ask, but didn't think of letscontrolit.com/forum when I did.

As I wrote: I'm totally aware that any transmitted U/P info can be sniffed due to the lack of HTTPS, still it would feel better if a guest (with access to the local network) for instance couldn't switch a relay just by knowing or trying IP addresses. With an U/P in the http-URL the "guest" has at least take the effort to record and search the traffic.

What other access protections are there, besides limiting the controlling IP address (range)? Is there any secure method given that the transmission isn't encrypted?

TD-er, 2020-10-07:
Well security isn't that elaborate right now, but the PR I mentioned is a first step to allow some access control mechanism as I really would like to have (a lot) more control over what can be done and what can be refused.

Wookbert, 2020-10-08:
Excuse my ignorance, but PR stands for? Public request? Preview release?

Also, can you explain in two or three sentences for the average Joe what #2778 / #3109 is about, how it improves security with regards to the said http commands? I don't get it.

tonhuisman, 2020-10-08:
PR is short for Pull Request, a request to pull (merge) a change into the software. It is a git term, where git is the distributed version control system that github is built upon. Github is a web-based wrapper around the git functions, and makes managing those PR's quite easy, also offers this issue management system, and other services.

Wookbert, 2020-10-08:
Pull request, stupid me. Know the term, but rarely are confronted with it. Thanks.

Which leaves us to the second part of my question...

tonhuisman, 2020-10-08:
Ah, well, #2778 is an effort to improve the quite complex code around handling GPIO pins. But this has proven to be quite a big task, nearly impossible to merge into the, ever changing because of other PR's, master branch (that's called `mega`, btw), so TD-er re-did some parts of it in a new PR #3109. But the real motivation for 3109 is improved security, as implied by your original question here. The complete story can best be observed as a game of chess, as it has become a quite complex story.

tonhuisman, 2023-09-02:
This has all been fixed, so issue can be closed.

closed 2023-09-02 by TD-er