letscontrolit/ESPEasy #3297: SGP 30

joe-1, 2020-10-07:
Please bring version lib 1.05 up to date to 2.0.

Thank you.
Joachim Vocke

TD-er, 2020-10-07:
What are the needed improvements?
Quite often a newer version of the library is also a larger library.

thomastech, 2020-10-08:
`Quite often a newer version of the library is also a larger library.`

I compiled the new library with ESPEasy (test_ESP8266_4M1M_VCC). It increased Flash by +368 bytes. No impact to RAM. 

Looking at the 2.0 code, it appears that the only major change was to use Adafruit's busio and i2c libraries (instead of Wire.h) to handle SGP30 communication. 

Doubtful there's any advantage to using 2.0 that justifies the extra Flash usage. But perhaps I missed something important that would encourage upgrading to it.

- Thomas

thomastech, 2020-10-10:
> From lib number 1.21, deep sleep supposedly works.

Seems like a reasonable thing to do. Moving to the 1.2 library (instead of 2.0) would achieve what you want without increasing Flash usage.

The SGP30 plugin would need to be revised too. That is to say, it needs to detect that wake-up from deep sleep has occurred so it uses the new sgp.begin() calling parameter that skips sensor initialization.

I recommend that you create a PR with the updated library and include the code changes to the _P083_SGP30 plugin. Or if you need help then consider bribing @TD-er with a fistful of gold or a case of his favorite beverage.

- Thomas

TD-er, 2020-10-10:
> The SGP30 plugin would need to be revised too. That is to say, it needs to detect that wake-up from deep sleep has occurred so it uses the new sgp.begin() calling parameter that skips sensor initialization.

Wake up from deepsleep you mean, or wakeup of the sensor?

thomastech, 2020-10-10:
From what I saw, the revised library adds a second calling parameter to the sgp.begin(). It can be used to suppress a complete sensor initialization (skips setting IAQ algorithm). So it seems to me that this would be used when the ESP device reboots from a deep sleep to preserve the sensor's existing state.

- Thomas