letscontrolit/ESPEasy #3299: MQTT disconnection

giig1967g, 2020-10-09:
Hi all,
I have 10 units in my home.
This morning at 5:27, all at the same time, 6 out of 10 units disconnected from MQTT broker and are still disconnected (they didn't reconnect until now. %ismqtt%=0). The other 4 have not been disconnected from MQTT.
There hasn't been any wifi disconnection, only MQTT for those 6 units.

It was common for all my units to randomly disconnect and immediately reconnect to mqtt once or twice a day, but it never happened that they remained disconnected.

I'm using openhab + mosquitto.

Any idea on how to troubleshoot the issue.

giig1967g, 2020-10-09:
I discovered something interesting:
the 6 units that have disconnected have all firmware version: 2020-08-12
the 4 units that didn't diconnect have firmware version 2020-06-11

Has anything changed in MQTT handling between these two versions to create this problem?

TD-er, 2020-10-09:
Well not sure what has changed between those builds. Have to look at it.

I have noticed something has changed regarding the WiFi code in the core libs.
Can you also check what core builds both sets have?

TD-er, 2020-10-09:
Hmm I'm afraid it will be quite hard to see what has changed between those 2 builds:
[image]

giig1967g, 2020-10-09:
6 disconnected: ESP82xx Core 683b8e60, NONOS SDK 2.2.2-dev(a58da79), LWIP: 2.1.2 PUYA support

4 working:ESP82xx Core a5432625, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support

TD-er, 2020-10-09:
Can you also give (from the system variables page) what the exact wifi state is and also the mqtt status?

giig1967g, 2020-10-09:
6 non working:
[image]

4 working:
[image]

TD-er, 2020-10-09:
The internal wifi bit-flag representation has changed, so even though one showing 4 and other showing 7, they both mean the same.
Has been no wifi disconnect?

giig1967g, 2020-10-09:
no. no wifi disconnect, as far as I can tell.
I checked in the router's log and there is no disconnection and no DHCP request around that time.

TD-er, 2020-10-09:
You can alo see the connected time on the sysinfo page

giig1967g, 2020-10-09:
right.
All 10 units have the same value (probably a router reboot 24 days ago):

[image]

TD-er, 2020-10-09:
Ah given the uptime I know what is the problem and that has been fixed :)

giig1967g, 2020-10-09:
good to know... what is it?

TD-er, 2020-10-09:
Hint: uptime is probably longer than the connected time?
The connected time passed the 2^31 msec, so it is now "negative" if you compare the time difference.

TD-er, 2020-10-09:
Just like the solve-time needed for this issue, is also negative :)

giig1967g, 2020-10-09:
uptime is 52 days.

giig1967g, 2020-10-09:
But all 10 of them have 52 days of uptime and 24 days of connection. Non only the 6 that have disconnected.

TD-er, 2020-10-09:
The connected time is probably counting backwards now?  (you have the screenshot to prove it)

giig1967g, 2020-10-09:
correct. counting backwards... :) Also on the 4 that are working.
What I don't understand is why in 4 is working and in 6 is not working...
Latest bin version fixes the issue?

TD-er, 2020-10-09:
Yep latest fixes the issue.
What triggered the issue is a seemingly innocent check that was added like this:
```c++
if (timePassedSince(connect_moment) > 200) {...}
```
This was in the function to check if we're connected, which is called from just about anywhere.

giig1967g, 2020-10-09:
ok

closed 2020-10-09 by giig1967g