letscontrolit/ESPEasy #3305: System Variable  %sysbuild_git% is empty in newer builds

Willi-Butz, 2020-10-09:
I'm using the 'System Info' in fhem.
Controller is 'FHEM HTTP'.
In the last builds I can't see the build Version of the Firmware.

I looked into Tools, System Variables and also there the %sysbuild_git% is empty.
The %sysbuild_date% is OK.

In build ESP_Easy_mega_20200929_normal_ESP8266_4M1M i have the problem.
Older build aqre OK.
In build ESP_Easy_mega-20200328_normal_ESP8266_4M1M it is OK 

Thank you for your work!

TD-er, 2020-10-09:
The last few builds were no longer built by Travis.
I made them by hand, so I thing I have to add it to the build script for manual builds.

tonhuisman, 2020-10-10:
> have to add it to the build script for manual builds.

Installing `pygit2` locally (from VSCode shell: `pip install pygit2`) worked for me (installed latest = 1.3.0, requirements.txt lists 1.2.1 but trying to downgrade/install that ends with an error), I now see the current git branch/commit name in my self-built .bin files.

TD-er, 2020-10-10:
this is usually set in the travis environment.
Have not yet updated the build script.

tonhuisman, 2022-04-18:
This has been solved, AFAICS, so issue can be closed.

closed 2022-04-18 by TD-er