letscontrolit/ESPEasy #3308: Device ‘Generic-System Info’ error ‘invalid character in names’

Willi-Butz, 2020-10-10:
Device Name: System-Info
Output Type is ‘Quad’

On ‚submit‘ red error bar on top: Invalid character in names. Do not use ',-+/*=^%!#[]{}()' or space.
Yes name contains '-'.

That also happens if a value Name contains a ‘-‘
But error shows only on value 1
On value 1 it is possible to set a value with ‘-‘, with error
On 2-3 the value is ignored and the default value taken (rssi, freeheap, uptime, load).

Check for invalid character only seems to work on value 1.
To block special character in names is OK. But ‘-‘ would be fine to use.

Tested with the last builds: 20201009, 20200929
The character ‘-‘ works in older builds. Example 
ESP_Easy_mega-20200328_normal_ESP8266_4M1M.bin

Thank you for your work.

tonhuisman, 2020-10-10:
The dash (-) isn't working correctly, as the value interpretation sees it as a subtraction during processing of rules. That's the reason it is no longer accepted. In older releases it didn't work correctly either (when processing rules).
The check not working as expected on value names 2-4 is most likely a bug.

TD-er, 2020-10-10:
> The dash (-) isn't working correctly, as the value interpretation sees it as a subtraction during processing of rules. That's the reason it is no longer accepted. In older releases it didn't work [...]

Maybe the submitter of the previous PR that added this check can also fix this undocumented feature? ;)

tonhuisman, 2020-10-10:
> Check for invalid character only seems to work on value 1.

Well, in fact the extended checks are introduced in release 20200913, and is only enabled in the not memory-challenged builds, like some hardware specific and 'normal' builds.

The anomaly that I found is that it does _not_ check the 4th value name, but it does validate names 1..3 as expected.
I'll fix that for the 4th name too.

tonhuisman, 2022-04-18:
This seems to be fixed, so issue can be closed.

closed 2022-04-18 by TD-er