letscontrolit/ESPEasy #3309: P062_MPR121_KeyPad sensitivity setting

smotek, 2020-10-10:
Requirement to add the ability to set the sensitivity of the touchpads.
The MPR121 chip enables this setting.
See Adafruit Library.

Add the ability to set a master touch.
If the main touch is set, touching it will deactivate the others.
Multiple touch treatment.

Thank you

tonhuisman, 2020-10-10:
As you seem to know what you are talking about, can you create a pull request yourself, containing the changes?
Instructions when using PlatformIO (suggested dev environment): https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html
Instructions when using Arduino IDE: https://espeasy.readthedocs.io/en/latest/Participate/ArduinoIDE.html

smotek, 2020-10-10:
Unfortunately, I do not have such abilities.

I was just looking to see if the chip allowed it.
I found various tips in the forums.

However, I am not able to make an adjustment.

tonhuisman, 2020-10-11:
I've constructed 2 sensitivity settings, is this what you have in mind?

[image]

TD-er, 2020-10-11:
Maybe it is also nice to show the current, max and minimal seen values (in logs or on the config page)
For example the ADC plugin does show the "current" value at load time of the page, which is a great help for the calibration.

smotek, 2020-10-11:
I can test.
But I would need a .bin
Question, does the library allow you to make settings for individual touches or just for all?

Proposal
TD-er is good,
It will help with calibration.

tonhuisman, 2020-10-11:
> Maybe it is also nice to show the current, max and minimal seen values

Well, the library does not supply those values, AFAICS, so I wouldn't know what to report there...
The log already shows the key or scancode at Info log-level, so, setting a value, pressing submit, and touching a button would give an expected feedback when touched soft/hard enough?

I haven't investigated the possible readable values from the controller (and something like that isn't implemented yet in the library), would that be feasible for a 'phase 2' development effort?

> Question, does the library allow you to make settings for individual touches or just for all?

Currently it's applying this to all touch buttons at once (library default feature). Are you implying you would want to set a different sensitivity per button?, as that would take some more time to code (the settings will have to be stored differently).

TD-er, 2020-10-11:
Well if it needs an overhaul of the library, then it is indeed a completely different feature request.
Let's not try to do it all at once :)

smotek, 2020-10-11:
Somewhere I read that they used reading readings for calibration, but probably a different library.

For now, just set the sensitivity for testing.
Calibration is performed experimentally.

The question about the possibility of setting a different sensitivity for the touch, is that if one area is large and the other small, then I expect that another sensitivity will be needed.
For example, one area 2x2cm and 6pcs 1x1cm.

TD-er, 2020-10-11:
For capacitive touch there are several parameters that may affect the read values.
- Distance from the finger
- Size of the object touching
- Dielectric properties of shield between touch pad and finger
- Length of trace from pad to chip
- Presence of other metal objects near the touch pad

And I'm probably missing some.

tonhuisman, 2020-10-11:
I'll keep the common setting, but add 12 settings for each separate touch-area, ignored when the values are 0.

Reading the actual touch values and keep previous values I'll skip for now.

tonhuisman, 2020-10-11:
> I haven't investigated the possible readable values from the controller (and something like that isn't implemented yet in the library),

After reading up on the datasheet of the mpr121, I now recognize the `filteredData(uint8_t t)` function as *the* way to get the touch 'value', so that means it shouldn't be that difficult to add this data to the data structure and UI.
It does involve small modifications to the library, but that's already required for enabling the setting of the tresholds for each separate touch object.

tonhuisman, 2020-10-11:
@smotek what version of ESPEasy are you using? (filename of the .bin file please, that has the most interesting info)

smotek, 2020-10-11:
@tonhuisman currently ESP_Easy_mega_20200929_test_beta_ESP8266_4M1M

smotek, 2020-10-12:
Today I created touchpads and the result is for default values:
- triangle 4x4cm 8cm2 surface works with direct touch.
- surface square 6x6cm 24cm2 does not work.
I'm curious to work after setting the sensitivity.
@tonhuisman  can i please get to the adjustment for testing?

tonhuisman, 2020-10-12:
I'm nearly done with my adjustments, including the calibration data, but I'll need a few more hours for testing. I'll make a PR tomorrow night (CET), if all goes according to plan.

smotek, 2020-10-12:
Thank you for the info.

tonhuisman, 2020-10-13:
I made a PR with the changes, but I don't have the stuff needed to make a complete build zipfile. I am able to provide a single .bin if you want that (assume the same config you mentioned earlier), but I'd have to make some arrangements for that.

smotek, 2020-10-13:
It is enough if in .bin it will be with ESPeasy with plugin 062 for testing.
I use a Wemos D1 board.

tonhuisman, 2020-10-14:
I've made a single build for ESP8266 like you mentioned above, and put it in a zip file:

[ESP_Easy_mega_20201014_test_beta_ESP8266_4M1M-PR#3316.zip](https://github.com/letscontrolit/ESPEasy/files/5380426/ESP_Easy_mega_20201014_test_beta_ESP8266_4M1M-PR.3316.zip)

Please let me know if this runs as expected, as, without the hardware connected and the plugin enabled, it constantly reboots my ESP8266 (and hangs the ESP32 I tested on).

smotek, 2020-10-14:
Test 1
flood settings 12/6
result: 2x touch and restart
log:
265466: ScanCode=0x1
265466: P062 touch current: 371 min: 371 max: 371
265479: EVENT: mpr#SC=1.00
265566: ScanCode=0x0
265603: EVENT: mpr#SC=0.00
267366: ScanCode=0x1
267366: P062 touch current: 371 min: 371 max: 371
267402: EVENT: mpr#SC=1.00
267566: ScanCode=0x0
267602: EVENT: mpr#SC=0.00

Test2:
Change setting 
[image]

Touches react the same way.
The log always has the same value.

Test3:
Set
for all 150/100
then 250/200
the sensitivity did not change.

It always prints the same value in the log.
P062 touch current: 371 min: 371 max: 371
With a small big touch.
Sometimes it restarts even when the HW is connected.
I can't identify when he'll come.
It's not regular.

smotek, 2020-10-14:
I came after it restarts.
When touching the surface 12.
It always happens.

It is interesting that if I make a touch on area 1, the values ​​will be displayed at area 2.

tonhuisman, 2020-10-14:
> When HW is not connected, it will not start ESP.

Would it start with the original build with no hardware connected? If so then I must have broken something...

smotek, 2020-10-15:
It is interesting
both versions behave the same
1. I disconnect the HW, it also does not start.
2. if HW is connected but 3.3V disconnected, it starts
3. if I enter device enable false then it will start.

It is enough for the module to be connected
GND, SCL, SDA and ESP start.

TD-er, 2020-10-15:
Well, SDA and SCL are pulled up to 3v3, so most of the time they will be high.
Maybe the module does have enough internal capacitance to keep running as long as the I2C lines are low.

smotek, 2020-10-15:
It is possible that it can respond to ESPeasy when it is powered via pollup.
If I disconnect the HW while working.
ESPeasy restarts in about 10 seconds, starts and sets Device enabled to false.
You can then connect to ESPeasy via the web.

TD-er, 2020-10-15:
If the task is disabled after reboot, then ESPEasy already crashed at least 10 times.
After 10 incomplete reboots, it starts disabling tasks until it can complete a boot.

tonhuisman, 2020-10-15:
> if I make a touch on area 1, the values ​​will be displayed at area 2.

I fixed that, classic off-by-1 mistake.
Also added the touchobject nr to the log message.

smotek, 2020-10-20:
@tonhuisman Didn't you make a new version?

tonhuisman, 2020-10-20:
I've been milling on that source a bit, and fixed another issue related to the calibration log, plus the fact that the previous commit was a bit cramped in time, so I didn't get to posting a test-build.
This is an updated build, same config as before:
[ESP_Easy_mega_20201020_test_beta_ESP8266_4M1M-PR#3316.zip](https://github.com/letscontrolit/ESPEasy/files/5410722/ESP_Easy_mega_20201020_test_beta_ESP8266_4M1M-PR.3316.zip)

smotek, 2020-10-22:
@tonhuisman
I tested the new version.
Displays calibration values.
Even on the right lines.
I don't understand their range when there is 343.
I have set a default of 12/6, the chip measures e.g. 110 and evaluates it as a touch.
Changing the suction values does not change the sensitivity.
I will try to do the connection with arduino and test how it works there.

tonhuisman, 2020-10-23:
> I tested the new version.

Thank you.

> Displays calibration values.
> Even on the right lines.

Great

> I don't understand their range when there is 343.
> I have set a default of 12/6, the chip measures e.g. 110 and evaluates it as a touch.

All I'm doing here is set calibration from the set values, and present the data that is available from the driver, I don't know how interpret it, nor can I validate the correctness of any data, as I don't have the hardware available for testing, sorry.

> Changing the suction values does not change the sensitivity.
> I will try to do the connection with arduino and test how it works there.

I'd be interested in your findings there.

PR #3316 (2020-11-17): [P062 Touch] Add sensitivity configuration
- Add global sensitivity setting
- Add sensitivity setting per touchobject
- Add calibration data gathering and display

Resolves #3309 

A few screenshots:

[image]

With Calibration = Yes:

[image]

The (marked) Calibration options are *only* available if the task is Enabled.
closed 2020-11-17 by TD-er