letscontrolit/ESPEasy #3312: Logentry loggs to Info not Error Level.

KnatteAnka, 2020-10-12:
When running rule:
"logentry,Test"
it only shows Test for Info and all 3 Debug.

reason is that in: ESPEasy/src/src/Commands/Diagnostic.cpp
**Row:** 161 has it to send to info not Error.
To fix this i see the quick fix by change:
**from:** addLog(LOG_LEVEL_INFO, tolerantParseStringKeepCase(Line, 2));
**To:** addLog(LOG_LEVEL_ERROR, tolerantParseStringKeepCase(Line, 2));

tested with latest github version relased 20201009

TD-er, 2020-10-12:
Well this is not a bug, as it currently only accepts 1 parameter, which is the text to send to the log.
See also the comment line right above what you were looking at.

```c++
  // FIXME TD-er: Add an extra optional parameter to set log level.
  addLog(LOG_LEVEL_INFO, tolerantParseStringKeepCase(Line, 2));
```

KnatteAnka, 2020-10-12:
the problem and bug is that it only shows the log message when your logg level is Info and debug not in Error mode
so if you want to get a logg message you have to get massiv spam of all commands running can use logentry to only show relevant information. Jimmy westberg think as me that its an bug!

meaning logentry is meaningless because it already show the command the act and text
[image]
but in Error Level mode it does not show anything.

Should only show message that of logentry

KnatteAnka, 2020-10-12:
as a long time solution it would be to add support for one more parameter of the Log_Level but until then to only set it to show for all levels is a good start and maybe only need!

TD-er, 2020-10-12:
Ah OK, I didn't get your point about the need for enabling the loglevel to at least LOGLEVEL_INFO.
Yep that's indeed an issue.

KnatteAnka, 2021-01-29:
can you look at my branch and see if you think this typ would work? if then i can do a pull request if needed
https://github.com/KnatteAnka/ESPEasy

in short term what i have done is changed logentry to have 2 variables
and check if second variable is set then use it else continue use LOG__LEVEL_INFO
input need to be in number form but its some other default plugins that use it like gpio and such
(not 100% sure that this will work and not sure what i need to compile it but when i look at GPIO/PWM then it looks like it would work and not be much CPU needed)

KnatteAnka, 2021-05-12:
@TD-er have at last gotten platformio installed successfully for Espeasy. and saw your comment about adding check if a valid option. And apparently my test change of diagnostic.cpp is not working.(Add extra parameter) so in the mean time i built a version that just change it permanently to error level so it is always visible and output is not the same color as other commands in the log.

if you compare the last picture with this one it is much easier to spot my logentry line and see the information.
[image]
My question is if its really reasonable to add multiple lines of code for a Logentry command when do you not want to see a log entry? if not needed dont you comment it out?

so my suggestion as i started is to just change it to output as LOG_LEVEL_ERROR .

i do not see a reason for it to create any problem for anyone

until it is implanted i will use custom build where i change it ;)

TD-er, 2021-05-12:
Well the problem is, there are lots of those "tiny" things that always end up not being so tiny when you add them...
For example today I thought I could finally add "RTC" functionality to ESP32, just as we have it on ESP8266.
Read about how it should be done yesterday and thought it would take just one or two hours to do... In the end it took almost the entire day today.

Log filtering is something a lot of people would like to have, so when I start with that part, I am afraid it may end up taking lots of time, while there are also lots of issues that actually are a bug.

I guess adding a "custom" log level even makes more sense to allow the coloring and/or filtering on a syslog server.

KnatteAnka, 2021-05-12:
Then i will continue to try and get an version that allow 2 parameters to work and get fråm espeasy_log.h how many allowed 0-x modes allowed question is should it show an error if outside range or just set it to default?

uzi18, 2021-05-13:
What if Logentry lines should be always added to log queue, whatever level you have setted?

closed 2021-05-18 by TD-er

uzi18, 2021-05-18:
@tonhuisman @TD-er do you think it is comfortable to change all log entries to an explicit level just because you want also to know espeasy DEBUG messages?
In my opinion, we should have just a command like `LogEntryLevel,level` command just to set such a level for all next entries.

TD-er, 2021-05-18:
@uzi18 Only the logentry command has changed.
And the current default was already `LOG_LEVEL_INFO`
So with this you can set loglevel to specific `LogEntry` commands.
Only real drawback with this -as far as I can tell now- is that it adds the loglevel to the log text if "less strict parsing" is enabled (also stated in the docs)
Adding a new command would also be possible, but setting it for all logentry commands may remove a bit of the flexibility that's now present (added in the PR), where you can set a variable to enable/disable some logentries in the rules while debugging.
This can also be done by switching the logentrylevel, but that's globally while you may only be interested in some specific event logentries while debugging.

One thing I think may be really useful (but may need some more changes) is to add a 'custom' level (maybe multiple) so your log isn't polluted when debugging rules.
It may be slightly more complex since there is already a level 0, 1, 2 and 3 defined.  (maybe negative?)

tonhuisman, 2021-05-18:
Adding a Custom log level (or several) and give these a negative value would offer the benefit of being able to turn off all logging, but still be able to write to the log from rules (and also from code).
But this poses the question that if the user sets logging to level None, he probably meant to have nothing in the log... 🤔 defeating the entire purpose of setting it to None.

OTOH, having commands to set the global log level from rules, and/or set the log level to use for the next LogEntry commands, separately, would be an interesting option, but I'm not sure if the average user would see any benefit in that. Or even use it. So I'm afraid this will only add extra bloat to the base ESPEasy code (that is already growing fast, lately).

TD-er, 2021-05-18:
Yep, it was just me 'thinking out loud' and really needs more thoughts before implementing it.
But regarding this PR I would like to see a reply from @uzi18 what he thinks of my reply and whether I may be missing something he might see as an issue.  (since he very often has shown good insight)

uzi18, 2021-05-18:
First of all, if user used logentry command lines they was used for special proposes, so they always should be shown on all log levels. Because we assume they are here for debugging rules or logging to syslog

If user setted log output level to debug eg. For Weblog all logentry lines outputs should be mixed with standard ESPEasy debug level. 
But one could imagine situation where user want just disable all these logentry lines from output by just issue logentrylevel,0 command even conditionally in special case

In perfect world we should have all these NO_debug defines check changed to 2 checks per plugin and/or subsystem like actual problems with mqtt, one should be able to define at compile time espeasy_debug_mqtt and ESPEasy_debug_mem create for test for user and ask for log to analyze.
This will benefit later because lots debugging is related to hardware and communication. 

Sure my english is not very fluent but maybe you will get my idea ;)
On the other hands for long time have no time for any coding so i'm glad to see here your hard work @tonhuisman and @TD-er !

TD-er, 2021-05-19:
> Sure my english is not very fluent but maybe you will get my idea ;)

I guess the amount of native English speakers among the ESPEasy users is small, so don't worry.
As long as you don't start writing in an eastern-European language, it's fine :) (Polish it was for you, right?)

>In perfect world we should have all these NO_debug defines check changed to 2 checks per plugin and/or subsystem like actual problems with mqtt, one should be able to define at compile time espeasy_debug_mqtt and ESPEasy_debug_mem create for test for user and ask for log to analyze.

I have thought about adding an extra `addPluginLog` macro, which would take out the plugin ID, but since we now moved to plugin data structs, this has become less easy to do.
Such an extra parameter could then be used to turn on filtering for specific plugins, useful for debugging.

The idea of adding a 'custom' log level is also one to allow disabling all rules logs and/or all other logs.
But that's for later.

PR #3639 (2021-05-18): [Commands] Enable LogEntry to log for other level than INFO #3312
Adds an optional 'level' argument at the end of the `LogEntry,<text_to_log>[,<level>]` command, so would look like: `LogEntry,"some logging text",1` will log this at ERROR level (1).
The log entry will include the level at the end if the 'Tolerant last parameter' Advanced setting is enabled.
Checks are in place to only accept normal log levels, ERROR (1), INFO (2), DEBUG (3), DEBUG_MORE (4). Not supported: DEBUG_DEV (9).

TODO:
- [x] Documentation update

Resolves #3312