letscontrolit/ESPEasy #3314: Rebbot due to: Const Interval timer, id: 2

giig1967g, 2020-10-13:
Hi all,
yesterday at 19:21 6 of my production units out of 10 suddendly rebooted with the following error in the sysinfo page:
Const Interval timer, id: 2

firmware: ESP_Easy_mega_20201009_normal_ESP8266_4M1M

in particular 5 units had this error:
```
Boot: | Manual reboot (9)
Reset Reason: | Software/System restart
Last Task: | Const Interval timer, id: 2
SW WD count: | 0

```
only one had a slight different message:

```
Boot: | Manual reboot (1)
Reset Reason: | Exception
Last Task: | Const Interval timer, id: 2
SW WD count: | 0
```
what is **Const Interval timer, id: 2** ?

TD-er, 2020-10-13:
Const interval timer is handled in this function:
```c++
void ESPEasy_Scheduler::setIntervalTimer(IntervalTimer_e id, unsigned long lasttimer) {
```

This is nr 2: `IntervalTimer_e::TIMER_1SEC`

So it was the `PLUGIN_ONCE_A_SECOND` call.
Sadly the plugin causing the crash was not logged.

giig1967g, 2020-10-13:
They rebooted at the same time. Strange that it was a plugin causing it.
In all those units I use P001 and system info. Controller is always OH MQTT mosquitto. I also use Inter-ESPEasy network.

In some I use P009 and P019, MQTT Import, LCD2004.

Is there a way to log the plugin that caused the crash?

TD-er, 2020-10-13:
Well, I may have been a bit too quick with drawing conclusions.
It was the call to this function:
https://github.com/letscontrolit/ESPEasy/blob/3b243f20c0b6f2eead7e25f39f427d225faffea7/src/src/Helpers/PeriodicalActions.cpp#L80-L165

TD-er, 2020-10-13:
I think you should also try my latest fix (not yet merged, will merge it now) regarding a fix for WiFi timers.
https://github.com/letscontrolit/ESPEasy/pull/3313

giig1967g, 2020-10-19:
I just applied latest firmware in all my production units, but today at 16:37, 6 units out of 10 rebooted at the same time.
They are not the same units that rebooted few days ago.
Very odd

giig1967g, 2020-10-19:
is there a way to add some debug information just before the "Const Interval timer, id: 2" message to find the root cause of this issue?
It's quite unstable at present.

giig1967g, 2020-10-19:
Update: at 17:35, 4 units rebooted at the same time

tonhuisman, 2020-10-19:
Is it possible that you have mains power issues (brown-out), that is a possible cause not yet mentioned, or are they battery-powered?

giig1967g, 2020-10-19:
I have one single 12V power supply for all the units that is backed up under a UPS.
Using Wemos as my units, I use in all units a Wemos original DC power shield that receives the 12V and converts to 5DC.
Also, it has never been an issue since latest updates.

TD-er, 2020-10-19:
UPS'es often perform a short test every now and then to test the batteries.
Could this also have been  such a test?

giig1967g, 2020-10-20:
At 19:28 yesterday 5 units rebooted again at the same time with "const interval timer, id: 2"
So resuming:
16:37: 6 units rebooted
17:35: 4 units rebooted
19:28: 5 units rebooted

@TD-er on the same UPS I have connected my Raspberry with OH2. No battery testing or UPS status change at that time in the logs.
Also, if there was a power failure, the reboot would have been due to a power failure, not due to  "const interval timer, id: 2"
Please help me solve it.

TD-er, 2020-10-20:
What I can do if trying to make it a bit more specific to track what the last action of the ESP was before crash.
Were these units running your (pending) PR of the GPIO move to be an internal command?

giig1967g, 2020-10-20:
Hi,
the units were running your latest official release 20201016.
None of them were running my pending PR.

TD-er, 2020-10-20:
Check!
That will limit the search :)

giig1967g, 2020-10-20:
> What I can do if trying to make it a bit more specific to track what the last action of the ESP was before crash.
> Were these units running your (pending) PR of the GPIO move to be an internal command?

Hi Gijs,
I suspect that the fact that the units rebooted all at the same time means that something outside the units is triggering the reboot... 
The power supply was a good hint, but probably the usual suspect is wifi (or mqtt) connectivity. 
So with your more detailed description we might see what the unit was doing just before the crash, but the actual crash is likely due to a timeout somewhere due to an external element (wifi router or mqtt broker, for example).
In your analysis, please take into account to write into the description something related to the wifi or mqtt timeouts.

thanks

tonhuisman, 2020-10-20:
A few, somewhat random, questions:
* On the Tools/Advanced page, do you have any WiFi related settings configured in the 'Special and Experimental Settings' section?
* Do you power the ESP's all on at exactly the same time? (As they have a single power-supply)

giig1967g, 2020-10-21:
>     * On the Tools/Advanced page, do you have any WiFi related settings configured in the 'Special and Experimental Settings' section?
> 
[image]

>     * Do you power the ESP's all on at exactly the same time? (As they have a single power-supply)
Well, I haven't repowered the ESP for a long time. But when I have upgraded the firmware, I did it in sequence, so starting from unit 1 to unit 10 in about 10 minutes.
But these are the units number involved in the latest reboots:
12/10 updated firmware to 20201009
12/10 18:38 -> units rebooted: 1,5,6,7,10
12/10 19:21 -> units rebooted: 1,5,6,10
18/10 6:41 -> units rebooted: 1,5,6,10
18/10 14:41 -> units rebooted: 1,5,6,10
18/10 15:37 -> units rebooted: 1,3,5,6,7,10
19/10 14:41 -> units rebooted: 1,3,5,6,7,10
19/10 14:52 -> updated firmware to 20201016 to all units (so all of them rebooted of course)
19/10 16:37 -> units rebooted: 1,3,5,6,7
19/10 16:37 -> units rebooted: 1,5,6,7
19/10 18:28 -> units rebooted: 1,5,6,10

Units 1,5,6,10 handle mainly push buttons, while units 3,7 handle mainly relais through MCP expanders.

giig1967g, 2020-10-21:
Here a possible hint:
Units 2,4,8 (that do not reboot), have the following setting:
[image]

While units that do reboot (1,3,5,6,7,10) and unit 9 (that do not reboot) have the following setting:
[image]

TD-er, 2020-10-21:
The connection Failure Threshold means it will perform a reboot after N failed connection attempts.
This option is disabled when set to 0.

The reason unit 9 did not reboot, may be caused by the unit just wasn't at the 10th failed attempt when it was successful in connecting.

So I guess that's a perfectly fine explanation to why it rebooted.
However I guess it would be nice to register if it was an intentional reboot and the intended reason.

giig1967g, 2020-10-21:
>  However I guess it would be nice to register if it was an intentional reboot and the intended reason.

sure. 

> So I guess that's a perfectly fine explanation to why it rebooted.

strange thing is that before updating the units to August release, they never rebooted.
Router is the same, LAN is the same, internal network is the same, units and rules are the same

TD-er, 2020-10-21:
I noticed the wifi reconnect routine is now attempting at a shorter interval.
So maybe it is now able to detect a failed attempt more quickly.

I also noticed the CPU load of the units is quite high when there is no WiFi connection, so that's one thing I do have to look into.
It also explains why it may have had 10 failed attempts so quickly where it didn't in older builds.

giig1967g, 2020-10-21:
> I noticed the wifi reconnect routine is now attempting at a shorter interval.

what was the previous and current interval?

TD-er, 2020-10-21:
Well there is not a defined interval, as it reacts on WiFi events.
So I assume one of the fixes I made recently does allow for failed attempts to be detected earlier, without running into timeouts.  (or it is a new bug where I should wait for a timeout...)
But I guess it would not be a bad idea to increase the retry interval if it failed a few times.

tonhuisman, 2023-09-02:
This seems to have been resolved, and can probably be closed.

closed 2023-09-02 by TD-er