letscontrolit/ESPEasy #3315: feature request: possibility to enable / disable specific rules

nitrocharcher, 2020-10-13:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

nitrocharcher, 2020-10-13:
I have several rule sets configured.
For example:
Rule Set 1
Rule Set 2
Rule Set 3

Now I want to trouble shoot something and the Rule Set 2 should be ignored.
At the moment I have to delete the content of the rule. It would be nice if there where a tick box.

ironataerial, 2020-10-13:
I comment out code line by line for my tests. 
Not sure if you can comment out a block though...

tonhuisman, 2020-10-13:
The easiest way of disabling an entire event (`on ... do`//`endon`) if by invalidating the on condition, I often add a X in front of the taskname, assuming it is a task event.

TD-er, 2020-10-13:
> I comment out code line by line for my tests.
> Not sure if you can comment out a block though...

Nope, the block comment is not present in the rules parser
Also not sure if we should have it as it may make the parsing a bit (pun intended) more complex.

Budman1758, 2020-10-15:
https://github.com/letscontrolit/ESPEasy/issues/659