letscontrolit/ESPEasy #3320: RFID ID as decimal in log#EVENT =vs= integer in MQTT ??

ozett, 2020-10-15:
i discovered a difference for the logging of the RFID key-ID on the esp.
it  seems a decimal-number instead of a integer.
is this a bug? i am on firmware 20111-mega
(as far as is rember that has also been the case back in 2016? dont know anymore...)

Sending the key-ID via MQTT seems as integers, also not the same number.
everthing else seems ok. connected is a wiegand reader

[image]

ozett, 2020-10-15:
maybe fixable together with https://github.com/letscontrolit/ESPEasy/issues/3259
?

TD-er, 2020-10-15:
Which RFID plugin are you using?
And which controller?

ozett, 2020-10-18:
controller openhab/mqtt: 

[image]

ozett, 2020-10-18:
Plugin: RFID/Wiegand
(maybe enhance to show the plugin-nr in the gui? at least for debug?

[image]

tonhuisman, 2020-10-18:
> maybe enhance to show the plugin-nr in the gui?

The DEV builds show the plugin P-number when adding a new task.
And as a 'workaround':
Hovering the help button(s) will reveal the plugin number (PluginX for the wiki help, and Pxxx for the Read the docs help) when your browser shows the link it points to, usually at the bottom of your browser window.

ozett, 2020-11-01:
thanks for the "workaround", 
but maybe room for improvement in the "easyness" of espeasy=letscontrolit when its directy shown?

as i over the years seldomly need to debug specifiy failures but in case i have difficulties always to remember alls thes fine details, so - at least for me - it would be easyier to see such things directy on the gui . Seems space and room for this...

TD-er, 2020-11-01:
If you press the buttons for either the wiki or the read the docs, you are immediately sent to the pages of the plugin.
What do you suggest should be added to it?

ozett, 2020-11-02:
simply show the plugin-nr?

how do i know that if have to press the button?
shure, everybody will find it out someday. or after testing.

but plugin-nr visible needs nothing.
(it is at least just a helper ...)

tonhuisman, 2023-09-01:
Doesn't seem to be relevant anymore, so can probably be closed.

closed 2023-09-01 by TD-er