letscontrolit/ESPEasy #3325: Feature Request: Config: Prepend Unit-Nr to hostname

ozett, 2020-10-18:
i like to order my espeasy devices with uni-number in front. 
maybe possible as a new feature?

[image]

[image]

tonhuisman, 2020-10-18:
Doesn't your current solution work, exactly as you want? Your request seems to imply a specific number format as well, how would you specify that?
I like my units ordered by hardware type, so they are named Sonoff-S20-xx, ESP-01-xx, ESP-12-xx, Shelly1-xx, ESP-32-xxx, where xx(x) is both the unit number and the 4th byte of the IP address. (I don't expect the number of units to go over 100, but you never know...)
Every person will have their own preference, but as both the name and unit number need to be set manually, why should that be 'automatic'? IMHO, even the 'Append unit number...' option is obsolete, *if* I want the unit number in the name, I'll put it there myself...

TD-er, 2020-10-18:
As indeed "every person has their own preference", maybe we can also allow for allowing system variables in the name?
e.g. `{unit}-bla`
Next problem will of course be related to the max. length of that string, but that may be something for later ;)
At least the locations where you can set hostname preferences should be improved as some MQTT brokers do require quite specific hostname IDs.

thomastech, 2020-10-24:
> IMHO, even the 'Append unit number...' option is obsolete, if I want the unit number in the name, I'll put it there myself...

I agree with this. The Unit Name field seems to offer sufficient flexibility. I've never understood the value of appending a unit number. 

I suspect I'm missing something. Are there installations where appending or prepending the unit number is necessary (versus including it in the name field)?

- Thomas

TD-er, 2020-10-24:
Well appending a unit number may only need to change a single variable.
The hostname is used in mDNS, and can also be used in the MQTT client etc.
For some brokers it is important to always have an unique name (and also the same every time you reconnect.)

tonhuisman, 2020-10-24:
> For some brokers it is important to always have an unique name

But that is not different for other devices connecting to those brokers anyway, so it a bit unclear (to me) why ESPEasy has the option to append the unitnr to the name. For me it is very normal and natural that each device _has_ a unique name.

TD-er, 2020-10-24:
For deployment of a lot of nodes, you may need to set both the unit nr and the unit name.
By setting the checkbox to append the unit nr to the hostname you can deploy your units using the same settings and only need to change the unit nr.
That's easier to deploy :)

thomastech, 2020-10-24:
> By setting the checkbox to append the unit nr to the hostname you can deploy your units using the same settings and only need to change the unit nr.
Thanks for explaining the value of the appending option. So it appears that prepending would be a good option to add.

- Thomas

TD-er, 2020-10-24:
I think it would be even more useful to allow just about any formatting using system variables.
As it can now almost universally be used all over the software, it makes no sense to not allow it here.
`%unit%_mynode` for example.

thomastech, 2020-10-24:
Universal formatting offers a lot of flexibility. Sounds like a bit more work, but I have a feeling you're ready to add this. My crystal ball sees a PR in your future.

Speaking of PR's, what is the significant difference between a Gijs Noorlander PR and a TD-er PR. In this split personality comparison, is Giji the "admin" guy and TD-er the "technical" dev? Or are you just trying to double your non-paying salary?

- Thomas

tonhuisman, 2020-10-24:
> Speaking of PR's, what is the significant difference between a Gijs Noorlander PR and a TD-er PR?

That looks like a git or Github thingy, in recent release notes I've been mentioned both as Ton Huisman (my real name) and tonhuisman (my account name), though I didn't do anything special to 'achieve' that, AFAICS. Though it _may_ be related to the moment the branch is deleted after it's merged.

TD-er, 2020-10-24:
I guess on one of my computers (or maybe in Github) my user name is different.
And about this "non-paying"... Don't underestimate the Patreon income, which paid for my new development PC I bought a few weeks ago 👍

tonhuisman, 2020-10-24:
> `%unit%_mynode` for example.

That does look quite usable, hadn't thought about the bulk-deploy scenario, obviously, in those cases it should be usable, but then I'd want to suggest to be able to set the 4th digit of the (fixed) IP address to the same value, as that has proven quite valuable/useful to me.

TD-er, 2020-10-24:
`%ip4%` you mean? ;)

tonhuisman, 2020-10-24:
Yup, but how to set that as the unit number, or the other way around....

TD-er, 2020-10-24:
The command `unit,%ip4%` to sync it in rules based on network#connected event ?

ozett, 2020-11-01:
glad that some ideas circled here.

just a general annotation:
using variables brings flexibilty for  the advanced, 
but obscurity and unsureness of the right use for the unexperienced. i guess

i always loved espeasy to be easy, in found it annoying where it was not easy.
maybe one can find a two-fold solution?

maybe variables (and documentation) for the advanced user,
but also stay simple on the gui for beginners? 
means ommit the unit-nr in the gui for simpler layout and use?

maybe keep the gui simple and clean and ommit option not necessary for beginners,
and have advanced option in an advanced section of the gui , or in variables and rules...

TD-er, 2020-11-01:
Just to be clear, I always want to keep it "simple for all, everything possible for experienced"
That being said, it is still a balance between:
- Keeping existing settings compatible
- Trying to fit it all in the binary
- Make it intuitive
- Allow all complex options

So we have to find a compromise somewhere and a compromise is finding a way where everyone is equally unhappy :)

So I guess it should be something like a number of prefix options to chose from, AND allow to use system variables.
Simply 2 checkboxes for "prepend" and "append" is also confusing as it allows for doing both which does not make sense.
This quickly becomes messy, so you need a combobox with some pre-defined formats.
As you never can handle all wishes in a combobox, the hostname should also support system variables (which can also be not used to keep it simple)

Like I said, there is not even a single way how the hostname should be represented, as MQTT brokers may also need specific formats.

ozett, 2020-11-02:
i will think tommorrow if can make a balanced suggestion, 
thanks for yout detailed thought about this

tonhuisman, 2023-09-02:
This has been implemented, but not linked to the PR, issue can be closed.

closed 2023-09-02 by TD-er