letscontrolit/ESPEasy #3327: ESPEasy build limitations (pending move of plugins to lib folder)

TD-er, 2020-10-20:
This is an issue to inform users of pending changes in ESPEasy repository structure.
Also I'm open to suggestions on how we can prevent these changes and still being able to build ESPEasy on all platforms.

As of now, the project has become so large we really need to change things or else we may spend a lot of time to apply work-arounds for the limits we're hitting.

### Problem 1: Crashing PlatformIO + VS code

Recently ESPEasy has passed some kind of limit which caused PlatformIO + VScode to crash frequently (over 10+ times a day when developing)
These crashes were caused by the intellisense + PlatformIO processes needing > 4 GByte to be parsed.
This was more then allowed in the environment used by VS code, which is actually an instance of Chromium using a single web app.

As suggested by the PlatformIO people, this could be mitigated by converting the .ino files of our project into .cpp/.h files as you would then have smaller chunks of code to process.

### Problem 2: Hitting Windows max. command line length

So I started to convert lots and lots of .ino files (e.g. files containing our webserver code), which indeed prevented crashes in my dev environment.
However, while doing this, the next limit appeared...

See: https://github.com/platformio/platform-espressif8266/issues/231

In short, the PlatformIO build process compiles all .cpp files into a .cpp.o file and then tries to link them all.
However, this resulted in a link command exceeding the Windows max. command line limit of 32768 bytes and thus fails.

### Possible work-around: Convert each plugin/controller into a library

While examining the problem I noticed it may be fixed for now by adding some extra statements in excluding libraries during compile, but in the end it would be a matter of time before hitting this limit again.
We can buy some time if the linker command could be executed a few directories closer to the compiled .cpp.o files, but that does need a change in the core PlatformIO build process and is also not a true fix.

The only way (as far as I know) is to move code to libraries as those can be explicitly ignored at compile time.
So what we can do, is create a library for every plugin and controller we now have.

However, this also means the complete project structure may have to be changed to make this work.
And right now I am not 100% sure Arduino IDE users may hit this exact same problem soon.

thomastech, 2020-10-21:
Regarding problem 2, this really sounds like an issue with how the Platformio linker does its thing. But from the dialog on your Platformio PR, it seems that the recommended workaround involves duct tape and bailing wire. And a LOT of work.

I doubt there are many Platformio projects that are currently so big that the linker exceeds the Window's command length. But hopefully Platformio Devs see your issue as a canary in the coal mine. That is to say, maybe the Devs will recognize this as a long term problem and proactively solve it now. 

- Thomas

TD-er, 2020-10-21:
Let's hope so, but even if they do start working on it right now, I think it will hit me in the face a few times before it is truly fixed.

One fix (in PIO) to buy time is to let the longcommand be generated to be executed from the directory shared in all .cpp.o lines of that file.

This could be `.pio/build/custom_ESP8266_4M1M/` or maybe even `.pio/build/custom_ESP8266_4M1M/`src/ 
To have the PIO environment name mentioned in the path in this longcommand is also creating different behavior for different env builds simply due to the difference in length of the name.
That's just silly.

In a typical build, (of ESPEasy) this longcommand file has 420+ lines.
So this can save roughly 12k (or more) in a typical longcommand file.

tonhuisman, 2020-10-21:
Well, as I'm from the 'dark ages', which just means I've done development on MS-DOS 😆 , where linkers used to have the option to specify the files to be linked into an executable to be specified in a file, that could then be passed to the linker using the `@inputfiles.tmp` syntax. AFAIR, they even had support for that per command-line option that was supposed to take a list of files. I'm a bit surprised that modern linkers don't seem to have that feature anymore.

tonhuisman, 2020-10-21:
Regarding the VSCode issues, I've not seen the kind of instability or crashes you describe, but then I'm not juggling around in the ESPEasy code-base all day. I do use VSCode for other dev work, and often have 4 to 6 instances open next to each other that each handle a lot (tenthousands) of files (and a laptop with 32 GB memory, that helps a bit), but not much of the files are sources that is 'applicable' to Intellisense (mostly xml containing pseudo-code).
Can the cashing of VSCode be related to the actual number of files opened in the editor? I'm using the Workspaces feature of VSCode, and switching workspaces, f.e. when switching between branches, could give an improvement in avoiding crashes.

TD-er, 2020-10-21:
Tools like CMake and also how Visual Studio handles projects define exactly what .cpp and .h files will be "seen" by the compiler.
This would also help here to mitigate (part) of this problem.
But right now (unless I'm missing something) PlatformIO only allows to ignore libraries, not .cpp files in the project.

thomastech, 2020-10-21:
> But right now (unless I'm missing something) PlatformIO only allows to ignore libraries, not .cpp files in the project.

Do you mean Platformio's source filter doesn't help in this situation?
https://docs.platformio.org/en/stable/projectconf/section_env_build.html#src-filter

- Thomas

TD-er, 2020-10-21:
Have not yet found/used the src filter.
Looks like it might deserve a second look (as long as it can be set from the Python build setup)

tonhuisman, 2023-09-02:
This has been resolved, so can be closed.

closed 2023-09-02 by TD-er