letscontrolit/ESPEasy #3330: [P092] issues with UVR61-3

pez3, 2020-10-21:
Hello there,

i'm currently trying to get P092 [https://espeasy.readthedocs.io/en/latest/Plugin/P092.html](url) to work with my UVR61-3.

i compiled the plugin myself based on the mega-20201016 release. during my tests i encountered some problems.

### Problem 1: Changing GPIO PIN

 As far as i could debug for now, changing of the GPIO pin is not working correctly. it seems that the interrupt is not attached properly to the new PIN. I encountered two problems here (as far as i can see for now):

- in the case PLUGIN_ONCE_A_SECOND:
  the new interrupt is not attached because IsISRSet is never set to false.
  `      if (!DLbus_Data->IsISRset) {
        // on a CHANGE on the data pin P092_Pin_changed is called
        DLbus_Data->attachDLBusInterrupt();
        addLog(LOG_LEVEL_INFO, F("P092 ISR set"));
      }
` 

- in the case PLUGIN_WEBFORM_SAVE:
  the old interrupt does never get detached because PLUGIN_WEBFORM_SAVE seems not to be called. Here i'm not sure, when PLUGIN_WEBFORM_SAVE is supposed to be called. I tried to set a debug log message in this case but never got the entry in the log. When should this state be called?

### Problem 2: PULLUP resistor
during my debug sessions i saw that the GPIO pin is initialized with a pullup resistor in the case PLUGIN_INIT. this causes the DL bus signal to be distorted (measured with oscilloscope). Why is the pullup resistor activated here?

Hope i can help moving this plugin forward...

Greets peter

TD-er, 2020-10-21:
`PLUGIN_WEBFORM_SAVE` should be called when you press the submit button on the task configuration page via the web interface.

pez3, 2020-10-21:
Hmmmm... I added a debug message just behind the `PLUGIN_WEBFORM_SAVE` case... nothing is printed in the log.

[image]

here the log:
[image]

Then i tested the same in the plugin P003. Here the debug message is printed as expected...
here the expected log when i press the submit button:

[image]

any clue why this could happen? seems for me, that this case is somehow not called. is this possible?

thomastech, 2020-10-22:
> any clue why this could happen? seems for me, that this case is somehow not called. is this possible?

Regarding the missing log messages, unfortunately the weblog is the problem. 
(1) Weblog won't show debug logs unless you specifically choose the debug level. 
(2) Weblog sometimes cannot display all info log messages during PLUGIN_WEBFORM_SAVE's save-to-file activity. I suspect all the log messages that are posted during SaveToFile are filling the buffer and some messages are lost. Just my guess, but could be something else causing this bug.

The Solution? These issues do not occur if you use Serial Log.

- Thomas

pez3, 2020-10-22:
Yep... It was also my guess that there is the posibility of lost log messages..

Nevertheless changing of GPIO Pin is not working correctly here.

thomastech, 2020-10-22:
There was a change made a couple weeks ago that may be related to your problem. See this:
https://github.com/letscontrolit/ESPEasy/pull/3293
https://github.com/letscontrolit/ESPEasy/commit/d6ce926b95d8ec29d91700117a539d12de75c052

The PR's remarks say that now the ISR is enabled **after** Wifi is connected. So be sure WiFi is online when you test the P092 plugin.

- Thomas

pez3, 2020-10-22:
Yep... Wifi is for sure online... Otherwise i would not be able to change GPIO pin in the web interface...

thomastech, 2020-10-22:
I would suggest using an older build (before PR 3293) to confirm the plugin was working. That will provide strong evidence that the recent changes are involved in the issue you are experiencing.

- Thomas

thomastech, 2020-10-22:
BTW, is there any chance the plugin is disabled? The device's "Enabled" setting must have a check mark in it. 

I know, silly question. But without seeing screenshots of your device settings things like this should be asked.

- Thomas

pez3, 2020-10-23:
Hey thx very much for your help. The suggestion with the older build is a good point. I will check this.

Yep device is enabled. Besides another little bug with decoding Uvr61-3 frames (i will file another issue for that). The plugin is working after Power cycling the esp. When changing the GPIO, nothing is received.

Due to the log message `no DL bus connected` i strongly think, ISR setting is not done correctly (as mentioned in the first post).

thomastech, 2020-10-23:
From reviewing the code, the _"## StartReceiving: Error: Nothing received! No DL bus connected!"_ message can occur if data is missing or invalid. For example, if the bus interface/wiring is incorrect. 

The P092 source code mentions that only a pull-up resistor is needed when the UVR61-3 is used. But the online information I found shows that it is a 12V interface; If that is the case, it would need a voltage divider (and disabled internal GPIO pullups) to achieve the required 3.3V logic. 

BTW, I found a very nice optical interface that would be ideal for the DL-BUS: https://github.com/futschikato/UVR2MQTT

But if you can get the older code to work with your existing wiring then that would confirm the hardware stuff is fine. This would confirm the issue involves the recent changes to P092.

- Thomas

pez3, 2020-10-23:
Yes i also read the comment about the pullup several days ago. After measuring the DL Bus signal with an oscilloscpe i emoved the internal pullup and added the voltage divider. As i mentioned before, now i'm receiving correct data. But only after power cycling the esp.

So i am prettysure that wiring is ok...

Can you confirm my conclusion about the missing reset of `IsISRSet`?

thomastech, 2020-10-23:
Ok, I didn't realize that you were reading valid data. That's fantastic. 

But I'm not sure what you mean about IsISRSet is missing. The IsISRSet flag must be set true or you would not be reading data. So if you're reading data then it's not missing.

Do you mean this: After submitting the plugin's configuration, a reboot is required in order to receive data? 
If this is the problem, then post a copy of the first minute of serial log messages after a reboot. Perhaps the P092 messages will tell us something.

- Thomas

pez3, 2020-10-23:
Yep IsISRSet is a flag...

In function attachDLBusInterrupt (src/src/PluginStructs/P092_data_struct.cpp) IsISRSet is set to true when the function is called.

In PLUGIN_ONCE_A_SECOND there is a query if this flag is false. When its false then attachDLBusInterrupt is called to attach the new interrupt. If a new GPIO is chosen, the old Interrupt is detached in PLUGIN_WEBFORM_SAVE but the new one is never attached because IsISRSet is never set back to false. And this is the point why i think changing of GPIO does not work (because the Interrupt is not Set to the New pin)

thomastech, 2020-10-23:
>  If a new GPIO is chosen, the old Interrupt is detached in PLUGIN_WEBFORM_SAVE but the new one is never attached because IsISRSet is never set back to false (at least i did not find the place). 

Good info, confirms why a reboot is needed. Seems like a simple fix.

- Thomas

TD-er, 2020-10-23:
> > If a new GPIO is chosen, the old Interrupt is detached in PLUGIN_WEBFORM_SAVE but the new one is never attached because IsISRSet is never set back to false (at least i did not find the place).
> 
> Good info, confirms why a reboot is needed. Seems like a simple fix.
> 
> * Thomas

You create a branch + PR for it? wink-wink-nodge-nodge :)

thomastech, 2020-10-23:
> You create a branch + PR for it? wink-wink-nodge-nodge :)

Comedy gold. And you'll be here all week. Be sure to tip your waitress.

- Thomas

pez3, 2020-10-23:
I'm not that familiar with Git 🙈🙈🙈 but i try my best ;)

I found another bug with decoding the dataframe for Uvr61-3... Is it ok to fix i here or file another issue?

TD-er, 2020-10-23:
@pez3 See also this PR's discussion to understand my remark :)
https://github.com/letscontrolit/ESPEasy/pull/3333

thomastech, 2020-10-23:
> I found another bug with decoding the dataframe for Uvr61-3... Is it ok to fix i here or file another issue?

This issue ticket would be good for reporting all the P092 bugs you've identified. If you're unable to submit a PR that fixes the bugs then create a summarized list of all the problems (update the first post). This will give the Dev's the info they need to create a fix.

- Thomas

pez3, 2020-10-27:
Since the protocol of UVR61-3 (V8.2) is different to the protocol of UVR61-3 (V8.3) i thought i update the P092 documentation. I saw that the protocol tables are .png files. does it make sense to change them to rst for easier maintainability?

TD-er, 2020-10-27:
If you are willing to put time in updating the charts, please do.
Just keep an eye on the other open PR for this plugin, so you won't make it impossible to merge.

uwekaditz, 2020-11-24:
Please see my PR#3318 for the ISR problem.
I just checked the documentation for the UVR61-3, there is a picture showing the voltage divider.
If you have a link for the documentation of the UVR61-3 v8.3 I'll add the decoding into the [P092] plug in and update the readthedocs pages.

pez3, 2020-11-25:
I asked Technische Alternative for the Protocol definition.They sent me a pdf. What is the best way to hand it over to you?

For my point of view Uvr61-3 works well without resistor.

Greets

uwekaditz, 2020-11-25:
@pez3 Just zip it and add it to a comment on this issue.

pez3, 2020-12-01:
Hi @uwekaditz ...

Technische Alternative did not allow me to post it into a forum or similar... They only allowed me to send it to someone personally. Maybe it's the best way you contact the support of Technische Alternative and ask them for the protocol specification. They send it to me within half an hour or so... so they are really quick.

greets

uwekaditz, 2020-12-01:
Hallo Pez3,

 

ich habe bei TA wegen des UVR61-3 angefragt, aber die wollen eine Seriennummer vom Gerät.

 

Vielleicht kannst Du einfach das PDF auf diese Email senden.

 

Danke,

 

uwekaditz

uwekaditz, 2020-12-03:
The documentation for UVR61-3 (v8.3 or higher) has been added to my latest PR #3318.

TD-er, 2021-10-14:
Close as it is already fixed by a (merged) PR.

closed 2021-10-14 by TD-er