letscontrolit/ESPEasy #3338: Domoticz MQTT Helper sends messages to Domoticz

urkelbundy, 2020-10-25:
Hi all,

I noticed messages in my domoticz log that do not belong there.
When I subscibe to the domoticz/in topic I see the following messages with which domoticz cannot do anything.

domoticz/in {
"log": "GPIO 5 Set to 0",
"plugin": 1,
"pin": 5,
"mode": "output",
"state": 0
}

They seem to be coming from ESPEasy where the Domoticz MQTT Helper plugin is used.
When the Domoticz MQTT Helper plugin is disabled, the messages in de domoticz log are gone.

There is only 1 controller active, Domoticz MQTT.
I see this behaviour using the following software versions:
ESP_Easy_mega_20201004_normal_ESP8266_4M1M
ESP_Easy_mega_20201012_normal_ESP8266_4M1M
ESP_Easy_mega_20201016_normal_ESP8266_4M1M

I do not have older versions running. If necessary I can try any version.

Personally I think the Domoticz MQTT Helper plugin shouldn' t send messages (back) to domoticz. There is no setting in the plugin that enables sending statuses to domoticz.

There are also "Ok" messages sent to Domoticz in the domoticz/in topic that shouldn't be there. At this point I have no clue where these are coming from. Will use tcpdump to try to find out from which IP they are coming.

Can you have a look at it!?

Thanks,
Peter

urkelbundy, 2020-10-25:
The "Ok" messages are also coming from the same Domoticz MQTT Helper plugin on 2 NodeMCU's.
On these NodeMCU's the plugin is configured without a GPIO. The status of the IDX configured in this plugin is used in rules to control a PCFGPIO instead of a normal GPIO. That works fine except for the "Ok" message that is send back to domoticz.

TD-er, 2020-10-25:
Hmm those do indeed look to be quite ESPEasy specific.
There are no Domoticz specific entries in there.
The Domoticz MQTT helper has not been changed for quite a while, so I'm a bit cautious to change it, unless it is perfectly clear it will not affect functionality.  (I have not written the plugin myself)

wdonker, 2020-10-27:
I have the same issue here. ESPEasy version 20200310 was still ok, in 20200929 this bug was already present. So it must be a change within this period. I did not yet test any version in between.

Maybe it is related to this [https://github.com/letscontrolit/ESPEasy/issues/2834]

tonhuisman, 2022-05-07:
I can only assume this is already fixed for some time, so it can be closed.

closed 2022-05-07 by TD-er