letscontrolit/ESPEasy #3339: Problem compiling using the custom.h and the IR plugins

jimmys01, 2020-10-25:
I cannot build the IR plugins using the custom.h file. 
`C:/Users/Jimmys/Documents/GitHub/ESPEasy/src/_P016_IR.ino:22:29: fatal error: IRremoteESP8266.h: No such file or directory`

Building using the eg the minimal_IRext_ESP8266_4M1M works fine

### Steps to reproduce

1.  Use the custom.h file and uncomment any of the IR related plugins
2. build using the env:custom_ESP8266_4M1M

TD-er, 2020-10-25:
Hmm I'm sure I already looked into this yesterday or Friday and made a fix for it...
But right now I cannot find it....

The fix was to add an extra "custom" environment specific for the IR builds.

TD-er, 2020-10-25:
Ah, found it....
I made it in this PR: https://github.com/letscontrolit/ESPEasy/pull/3323

I will cherry pick the commits to the `mega` branch so anyone has them.

TD-er, 2020-10-25:
As soon as Travis agrees with my PR, I will merge it.

jimmys01, 2020-10-26:
Still the same I am afraid.

jimmys01, 2020-10-26:
> 
> 
> Still the same I am afraid.

Wrong, just realized that there is a new custom build just for the IR stuff

PR #3340 (2020-10-25): Fix building using IR plugins & Custom.h (#3339)
Fixes: #3339
closed 2020-10-26 by jimmys01