letscontrolit/ESPEasy #3341: wifi connected but not pingable - wifi instable

riker65, 2020-10-26:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [x] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [x] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

no ping no http response even connected to AP

### Expected behavior

connect to wifi - status connected and pingable -webserver coming up

ping response

### Actual behavior

no ping response
no connection via http

### Steps to reproduce

1. wificonnect
2. restart
3. 

yes
rebooted
restarted
repowered

### System configuration

Hardware:  nodeMCU Amica

ESP Easy version: 

Build:⋄ | 20110 - Mega
-- | --
System Libraries:⋄ | ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ |  
Plugin Count:⋄ | 46 [Normal]
Build Origin: | Travis
Build Time:⋄ | Sep 13 2020 00:22:08
Binary Filename:⋄ | ESP_Easy_mega_20200913_normal_ESP8266_4M1M
Build Platform:⋄ | Linux-4.19.104-microsoft-standard-x86_64-with-glibc2.29
Git HEAD:⋄ | mega_fe6e911

ESP Easy settings/screenshots: 

### Rules or log data

```

Serial LOG: 
07:58:56.972 -> 
INIT : Booting version:  (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
07:58:56.972 -> 73 : Info  : INIT : Free RAM:35072
07:58:56.972 -> 74 : Info  : INIT : Warm boot #1 Last Task: Background Task Last systime: 1074 - Restart Reason: Software/System restart
07:58:57.019 -> 75 : Info  : FS   : Mounting...
07:58:57.019 -> 100 : Info  : FS   : Mount successful, used 75802 bytes of 957314
07:58:57.019 -> 123 : Info  : CRC  : SecuritySettings CRC   ...OK 
07:58:57.112 -> 234 : Info  : INIT : Free RAM:32448
07:58:57.112 -> 235 : Info  : INIT : I2C
07:58:57.112 -> 235 : Info  : INIT : SPI not enabled
07:58:57.206 -> 312 : Info  : INFO : Plugins: 46 [Normal] (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
07:58:57.300 -> 416 : Info  : WIFI : Set WiFi to STA
07:58:57.393 -> 519 : Info  : WIFI : Connecting TR_ESPFB31 attempt #1
07:58:57.440 -> 525 : Info  : Webserver: start
07:58:57.440 -> 526 : Info  : Time set to 1074.000 Time adjusted by 1073475.00 msec. Wander: 298.19 msec/second
07:58:57.440 -> 527 : Info  : Current Time Zone: STD time start: 1970-10-25 03:00:00 offset: 0 min
07:58:57.440 -> 529 : Info  : Local time: 1970-01-01 00:17:54
07:58:58.476 -> 1589 : Info  : WIFI : Connected! AP: TR_ESPFB31 (38:10:D5:89:A4:20) Ch: 6 Duration: 1044 ms
07:58:58.476 -> 1591 : Info  : WIFI : DHCP IP: 192.168.1.143 (ESP-KSW-Easy) GW: 192.168.0.31 SN: 255.255.240.0   duration: 24 ms
07:58:58.570 -> 1674 : Info  : firstLoopConnectionsEstablished
07:58:58.757 -> 1864 : Info  : WD   : Uptime 0 ConnectFailures 0 FreeMem 24528 WiFiStatus WL_CONNECTED
07:59:28.775 -> 31864 : Info  : WD   : Uptime 1 ConnectFailures 0 FreeMem 24008 WiFiStatus WL_CONNECTED

```

logging on router dhcp server:
WLAN-Gerät angemeldet (2,4 GHz), 72 Mbit/s, ESP-KSW-Easy, IP 192.168.1.143, MAC CC:50:E3:4A:51:3F.
26.10.20
07:45:39
WLAN-Gerät hat sich abgemeldet (2,4 GHz), ESP-KSW-Easy, IP 192.168.1.143, MAC CC:50:E3:4A:51:3F.
26.10.20
07:45:22
WLAN-Gerät angemeldet (2,4 GHz), 72 Mbit/s, ESP-KSW-Easy, IP 192.168.1.143, MAC CC:50:E3:4A:51:3F.
26.10.20
07:45:22
WLAN-Gerät hat sich abgemeldet (2,4 GHz), ESP-KSW-Easy, IP 192.168.1.143, MAC CC:50:E3:4A:51:3F.
26.10.20
07:45:10
WLAN-Gerät angemeldet (2,4 GHz), 72 Mbit/s, ESP-KSW-Easy, IP 192.168.1.143, MAC CC:50:E3:4A:51:3F.
26.10.20
07:45:07
WLAN-Gerät hat sich abgemeldet (2,4 GHz), ESP-KSW-Easy, IP 192.168.1.143, MAC CC:50:E3:4A:51:3F.
26.10.20
07:45:07
WLAN-Gerät angemeldet (2,4 GHz), 72 Mbit/s, ESP-KSW-Easy, IP 192.168.1.143, MAC CC:50:E3:4A:51:3F.
26.10.20
07:45:07
WLAN-Gerät hat sich abgemeldet (2,4 GHz), ESP-KSW-Easy, IP 192.168.1.143, MAC CC:50:E3:4A:51:3F.
26.10.20
07:41:54
WLAN-Gerät angemeldet (2,4 GHz), 72 Mbit/s, ESP-KSW-Easy, IP 192.168.1.143, MAC CC:50:E3:4A:51:3F.
26.10.20
07:41:50
WLAN-Anmeldung ist gescheitert (2,4 GHz): Authentifizierungsfehler. MAC-Adresse: CC:50:E3:4A:51:3F.
26.10.20
07:41:47
WLAN-Gerät wurde abgemeldet (2,4 GHz), ESP-KSW-Easy, IP 192.168.1.143, MAC CC:50:E3:4A:51:3F. [2 Meldungen seit 26.10.20 07:41:40]
26.10.20
07:41:33
WLAN-Gerät hat sich abgemeldet (2,4 GHz), ESP-KSW-Easy, IP 192.168.1.143, MAC CC:50:E3:4A:51:3F.
26.10.20
07:41:17
WLAN-Gerät angemeldet (2,4 GHz), 72 Mbit/s, ESP-KSW-Easy, IP 192.168.1.143, MAC CC:50:E3:4A:51:3F.

more serial LOG Issues:
07:44:46.508 -> 224697 : Info  : firstLoopConnectionsEstablished
07:44:54.570 -> 232765 : Info  : WIFI : Disconnected! Reason: '(200) Beacon timeout' Connected for 2 m 53 s
07:44:57.608 -> 235786 : Info  : Reset WiFi.
07:44:57.608 -> 235788 : Info  : WIFI : Set WiFi to OFF
07:44:58.601 -> 236801 : Info  : WIFI  : WiFiConnected() out of sync: DISCONNECTED RSSI: -83 status: STATION_CONNECTING
07:44:58.601 -> 236813 : Info  : AP Mode: Client disconnected: 74:DA:38:00:53:18 Connected devices: 0
07:44:58.721 -> 236918 : Info  : WIFI : Set WiFi to STA
07:44:58.838 -> 237021 : Info  : WIFI : Connecting TR_ESPFB31 attempt #8
07:44:58.944 -> 237125 : Info  : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 3 m 55 s
07:45:01.936 -> 240157 : Info  : WIFI : Connected! AP: TR_ESPFB31 (38:10:D5:89:A4:20) Ch: 6 Duration: 3123 ms
07:45:01.983 -> 240191 : Info  : WIFI : DHCP IP: 192.168.1.143 (ESP-KSW-Easy) GW: 192.168.0.31 SN: 255.255.240.0   duration: 44 ms
07:45:05.846 -> 244039 : Info  : WD   : Uptime 4 ConnectFailures 0 FreeMem 23200 WiFiStatus WL_CONNECTED
07:45:10.519 -> 248739 : Info  : WIFI : Disconnected! Reason: '(200) Beacon timeout' Connected for 8492 ms
07:45:10.652 -> 248756 : Info  : WIFI : Connecting TR_ESPFB31 attempt #9
07:45:13.694 -> 251912 : Info  : WIFI : Connected! AP: TR_ESPFB31 (38:10:D5:89:A4:20) Ch: 6 Duration: 3135 ms

more assoc errors.

07:41:04.608 -> 2730 : Info  : WIFI  : Scan finished, found: 14
07:41:04.608 -> 2731 : Info  : WIFI  : Selected: TR_ESPFB31 38:10:D5:89:A4:20 Ch:6 (-90dBm) WPA2/PSK
07:41:05.872 -> 4056 : Info  : WD   : Uptime 0 ConnectFailures 0 FreeMem 24552 WiFiStatus WL_DISCONNECTED
07:41:07.656 -> 5847 : Info  : WIFI : Connected! AP: TR_ESPFB31 (38:10:D5:89:A4:20) Ch: 6 Duration: 3141 ms
07:41:14.522 -> 12709 : Info  : WIFI : Connecting TR_ESPFB31 attempt #2
07:41:24.601 -> 22820 : Info  : WIFI : Set WiFi to AP+STA
07:41:25.635 -> 23811 : Info  : WIFI : AP Mode ssid will be ESP-KSW-Easy with address 192.168.4.1
07:41:25.635 -> 23815 : Info  : WIFI : Connecting TR_ESPFB31 attempt #3
07:41:26.551 -> 24762 : Info  : WIFI : Disconnected! Reason: '(8) Assoc leave' Connected for 10 s
07:41:27.684 -> 25887 : Info  : WIFI : Disconnected! Reason: '(4) Assoc expire' Connected for 1955 ms
07:41:27.684 -> 25904 : Info  : WIFI : Connecting TR_ESPFB31 attempt #4
07:41:33.915 -> 32129 : Info  : WIFI : Connected! AP: TR_ESPFB31 (38:10:D5:89:A4:20) Ch: 6 Duration: 6212 ms
07:41:35.860 -> 34066 : Info  : WD   : Uptime 1 ConnectFailures 0 FreeMem 21440 WiFiStatus WL_DISCONNECTED
07:41:38.018 -> 36223 : Info  : WIFI : Disconnected! Reason: '(2) Auth expire' Connected for 3998 ms
07:41:38.018 -> 36240 : Info  : WIFI : Connecting TR_ESPFB31 attempt #5
07:41:41.219 -> 39434 : Info  : WIFI : Disconnected! Reason: '(202) Auth fail' Connected for 3085 ms
07:41:41.267 -> 39451 : Info  : WIFI : Connecting TR_ESPFB31 attempt #6
07:41:44.384 -> 42582 : Info  : WIFI : Connected! AP: TR_ESPFB31 (38:10:D5:89:A4:20) Ch: 6 Duration: 3117 ms
07:42:01.275 -> 59465 : Info  : WIFI : Connecting TR_ESPFB31 attempt #7

TD-er, 2020-10-26:
Can you please update to the latest build (or one older)?
I think I already solved this one as there was a bug in the timers used in WiFi.
This could result in such WiFi instability if the first attempt to connect wasn't as smooth as it should be.

riker65, 2020-10-26:
will do

riker65, 2020-10-26:
btw which FW version would you consider as stable concerning best wifi stability?

TD-er, 2020-10-26:
This timer issue was fixed in the last 2 nightly builds.

riker65, 2020-10-26:
Hi
problem still exists.

tried new FM from 1022 
after that flashed with 4MB blank.

Serial port COM12
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: cc:50:e3:4a:51:3f
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Warning: Image file at 0x0 doesn't look like an image file, so not changing any flash settings.
Compressed 4194304 bytes to 4086...
Wrote 4194304 bytes (4086 compressed) at 0x00000000 in 0.4 seconds (effective 89484.0 kbit/s)...

is this correct response flashing 4b bank? all should be erased, right
serial log: 15:39:58.573 -> Flash real size: 4194304
15:39:58.573 -> 
15:39:58.573 -> Flash ide  size: 4194304
15:39:58.573 -> Flash ide speed: 40000000
15:39:58.573 -> Flash ide mode:  DOUT
15:39:58.573 -> Flash Chip configuration ok.
15:39:58.573 -> 
15:39:58.573 -> 
15:39:58.573 -> MAC: CC:50:E3:4A:51:3F

after flashing it again it looks better.

from your experience, what is the average  uptime of the ESP Easy FW.

Thanks T

flashing again with latest firmware

TD-er, 2020-10-26:
> from your experience, what is the average uptime of the ESP Easy FW.

Well most of my units only reboot due to flashing new versions or power outages.  (highest uptime before power outage 2 weeks ago was about 155 days)
Another reason to crash can be when running out of memory.

I'm working on the WiFi part to at least make ESP32 and ESP8266 share more code and one thing I found that was clearly funky...
What happened in the code was that it could start to reconnect even when not all wifi events were processed.
So not sure what issues this could cause, so that's for sure one thing that will be fixed in the next build.

riker65, 2020-10-26:
ok thanks 
will check next build.

Wifi stability is very important. having a lot of disconnects and try to track it down .

TD-er, 2020-10-26:
Just a few hints to try:
- Used a fixed channel on the AP
- Set B/G mode in ESPEasy
- Do not use "Eco mode" in ESPEasy
- Check to send Gratuitous ARP in ESPEasy (that doesn't have to do with WiFi disconnects, but more with units not reachable)

thomastech, 2020-10-26:
> Wifi stability is very important. having a lot of disconnects and try to track it down .

Regarding reliable connections, I noticed the RSSI was -83dBm during the disconnect seen in your log. That RF strength is a bit marginal and could invite random disconnects/reconnects.

- Thomas

iot-sle, 2020-11-13:
I'm faced to the same issue I guess, here are the results of my tests. I'm sure it's not the power supply, see first test.

Hardware: ESP-01S 1Mb RAM
Firmware:
Build:⋄	20111 - Mega
System Libraries:⋄	ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄	
Plugin Count:⋄	46 [Normal]
Build Origin:	Travis
Build Time:⋄	Nov 2 2020 13:12:34
Binary Filename:⋄	ESP_Easy_mega_20201102_normal_ESP8266_1M
Build Platform:⋄	Linux-4.19.104-microsoft-standard-x86_64-with-glibc2.29
Git HEAD:⋄	mega-20201102_ca0be3f 

Wifi:	802.11N (RSSI -71 dB)

WiFi Settings
Force WiFi B/G:	false
Restart WiFi Lost Conn:	true
Force WiFi No Sleep:	true
Connection Failure Threshold:	0

Controller: Domoticz MQTT 
Hardware:
GPIO boot states
Pin mode GPIO-0 (D3) ⚠:	Input pullup
Pin mode GPIO-1 (D10):	Default
Pin mode GPIO-2 (D4) ⚠: Input pullup
Pin mode GPIO-3 (D9):	Default

Devices:
RFID-Wiegand
Sensor
GPIO ← D0 (Green, 5V):	GPIO-0 (D3) ⚠
GPIO ← D1 (White, 5V):	GPIO-2 (D4) ⚠
Wiegand Type:	34 Bits
Present hex as decimal value:	not selected
Data Acquisition
Send to Controller
❶ selected
IDX: 53

Issue01: 
ESP is no longer responding by web/wifi after some time
Description: After some time working correctly esp is no loger responding to http and ping calls.
It is still listed in the device list of the WLAN AP, but there is no MQTT or any reaction by TCP.

To support debugging I did some tests.

Test 1:
Using following code as Rules Set 1:

on Clock#Time=All,**:20 do //will run every hour 20 mins after full hour
  Publish domoticz/in,'{"idx":56,"nvalue":0,"svalue":"%sysname% <font color='red'>rebooted</font> by rule"}'
  reboot
endon

on Clock#Time=All,**:** do //will run every minute
  Publish domoticz/in,'{"idx":56,"nvalue":0,"svalue":"%sysname% alive! %lcltime%"}'
endon

Result: I get a message by MQTT to my domoticz server every minute and every hour the esp is rebooted. 
Remark: It takes 2 minutes until it is reachable by http after a reboot. That seems very long to me.
Corresponding log entries in Domoticz:

2020-11-13 09:23:00	ESP_Easy_RFID_Garage alive! 2020-11-13 09:23:00	Admin
2020-11-13 09:22:00	ESP_Easy_RFID_Garage alive! 2020-11-13 09:22:00	Admin
2020-11-13 09:20:00	ESP_Easy_RFID_Garage rebooted by rule	Admin
2020-11-13 09:19:00	ESP_Easy_RFID_Garage alive! 2020-11-13 09:19:00	Admin
2020-11-13 09:18:00	ESP_Easy_RFID_Garage alive! 2020-11-13 09:18:00	Admin

The connection is alway stable, never stuck. We can eliminate instable power supply as the reason for no response of esp.

Test 2:
Rules Set 1:

on Clock#Time=All,**:** do //will run every minute
  Publish domoticz/in,'{"idx":56,"nvalue":0,"svalue":"%sysname% alive! %lcltime%"}'
endon

Only the message to MQTT server every minute. I guess that will be enough. It's like a "stay alive ping" on wifi/web or what ever.
Startet at 9:44 running until

2020-11-13 12:38:00	ESP_Easy_RFID_Garage alive! 2020-11-13 12:38:00	MQTT
2020-11-13 12:37:00	ESP_Easy_RFID_Garage alive! 2020-11-13 12:37:00	MQTT
2020-11-13 12:36:00	ESP_Easy_RFID_Garage alive! 2020-11-13 12:36:00	MQTT
2020-11-13 12:35:00	ESP_Easy_RFID_Garage alive! 2020-11-13 12:35:00	MQTT
2020-11-13 12:34:00	ESP_Easy_RFID_Garage alive! 2020-11-13 12:34:00	MQTT

But I was wrong after less than 3 hours esp was hanging again. No reaction any longer.

Test 3:
Rules Set 1:

o
on System#Boot do
  Let,1,1
  Let,2,1
endon

on Clock#Time=All,**:** do //will run every minute
  if [Var#2]<[Var#1] do
    Let,2,[Var#2]+1  //skip the call for always 1 minute more
  else
    Publish domoticz/in,'{"idx":56,"nvalue":0,"svalue":"%sysname% alive! %lcltime%"}'
    Let,2,1
    Let,1,[Var#1]+1
  endif
endon

Check whether the count of "Stay alive pings" per period has a significant influence to the hang up time.

Started at 13:48

2020-11-13 15:04:00	ESP_Easy_RFID_Garage alive! 2020-11-13 15:04:00	MQTT
2020-11-13 14:52:00	ESP_Easy_RFID_Garage alive! 2020-11-13 14:52:00	MQTT
2020-11-13 14:41:00	ESP_Easy_RFID_Garage alive! 2020-11-13 14:41:00	MQTT
2020-11-13 14:31:00	ESP_Easy_RFID_Garage alive! 2020-11-13 14:31:00	MQTT
2020-11-13 14:22:00	ESP_Easy_RFID_Garage alive! 2020-11-13 14:22:00	MQTT
2020-11-13 14:14:00	ESP_Easy_RFID_Garage alive! 2020-11-13 14:14:00	MQTT
2020-11-13 14:07:00	ESP_Easy_RFID_Garage alive! 2020-11-13 14:07:00	MQTT
2020-11-13 14:01:00	ESP_Easy_RFID_Garage alive! 2020-11-13 14:01:00	MQTT
2020-11-13 13:56:00	ESP_Easy_RFID_Garage alive! 2020-11-13 13:56:00	MQTT
2020-11-13 13:52:00	ESP_Easy_RFID_Garage alive! 2020-11-13 13:52:00	MQTT
2020-11-13 13:49:00	ESP_Easy_RFID_Garage alive! 2020-11-13 13:49:00	MQTT
2020-11-13 13:46:00	ESP_Easy_RFID_Garage alive! 2020-11-13 13:46:00	MQTT
2020-11-13 13:42:00	ESP_Easy_RFID_Garage alive! 2020-11-13 13:42:00	MQTT
2020-11-13 13:39:00	ESP_Easy_RFID_Garage alive! 2020-11-13 13:39:00	MQTT
2020-11-13 13:37:01	ESP_Easy_RFID_Garage alive! 2020-11-13 13:37:00	MQTT

I assume that the period of MQQT connections has no significant influence to the hang up. 
I can do more test if you have some ideas how to narrow the problem. 
For now I will return to reboot once an hour, but without the "staying alive pinging".
We will see whether that's stable. If not, I will tell you.

TD-er, 2020-11-13:
Please try to format (new) posts a bit to help structure all the information.
This is quite a long text and it is not immediately clear what the intention is of a test.

Things you can try:
- Send a continous ping from another host to the ESP
- Enable "Send Gratuitous ARP packets"

Please have a look at the timing stats page to see if something takes quite a long time to process.
Long blocking code does have an effect on WiFi performance.

ebogaard, 2020-11-15:
I have to report I have the same problem with a certain type of ESP8266: "NodeM" made by RobotDyn: https://robotdyn.com/wifi-nodem-esp8266-cp2102.html
I have two of these and both have the same problem.
I tested this with the newest version (20201102) and a much older version (20200721), with a clean blank image flashed in between to be sure.

As an AP it works: I can conect to it and set the correct wifi password.
Then, after it reboots, I can see the ESP conecting to the AP, getting the correct IP address by DHCP, then a "(200) Beacon timeout".
During this time, the ESP isn't pingable or otherwise reachable.

```
432 : Info  : WIFI : Set WiFi to STA
535 : Info  : WIFI : Connecting XXX attempt #1
540 : Info  : Webserver: start
541 : Info  : Time set to 32.000 Time adjusted by 31460.00 msec. Wander: 8.74 msec/second
542 : Info  : Current Time Zone: STD time start: 1970-10-25 03:00:00 offset: 0 min
544 : Info  : Local time: 1970-01-01 00:00:32
1906 : Info  : WD   : Uptime 0 ConnectFailures 0 FreeMem 24008 WiFiStatus WL_DISCONNECTED
5329 : Info  : WIFI : Connected! AP: XXX (00:1D:AA:9F:XX:XX) Ch: 1 Duration: 4785 ms
5378 : Info  : WIFI : DHCP IP: 192.168.123.83 (ESP-Easy) GW: 192.168.123.1 SN: 255.255.255.0   duration: 55 ms
5422 : Info  : firstLoopConnectionsEstablished
13680 : Info  : WIFI : Disconnected! Reason: '(200) Beacon timeout' Connected for 8256 ms
13709 : Info  : WIFI : Connecting XXX attempt #2
17523 : Info  : WIFI : Connected! AP: XXX(00:1D:AA:9F:XX:XX) Ch: 1 Duration: 3801 ms
19495 : Info  : WIFI : DHCP IP: 192.168.123.83 (ESP-Easy) GW: 192.168.123.1 SN: 255.255.255.0   duration: 1983 ms
28017 : Info  : WIFI : Disconnected! Reason: '(200) Beacon timeout' Connected for 10 s
28034 : Info  : WIFI : Connecting XXX attempt #3
... etc ...
```

I have three other ESP's on the same AP: Wemos D1 mini, Adafruit Huzzah Feather and a custom model with custom firmware.
These Wemos and Adafruit work fine with ESP Easy!

ebogaard, 2020-11-18:
I did some further reasearch and flashed the latest tasmota.bin. Wifi workf flawlessly in AP- and Client-mode with that firmware.

I'm not sure if that helps, or if the Wifi-implementations differ completely.

TD-er, 2020-11-18:
I know Tasmota does do the WiFi completely different from ESPEasy, but I can have a look to see what SDK they use (the "alt-wifi" builds of ESPEasy)

ebogaard, 2020-11-18:
Would it help if I sent you one of my modules (they're useless now anyway)? Maybe that helps you debugging the issue.

TD-er, 2020-11-18:
Not sure if it will be helpful, as I often see those instabilities are caused by a combination of several things like the used access point, other WiFi networks in the neighborhood and used power supplies.

TD-er, 2022-04-01:
Should be fixed by this PR: 
- https://github.com/letscontrolit/ESPEasy/pull/3983

Please let me know if it is indeed fixed now... finally

closed 2022-04-01 by TD-er

ebogaard, 2022-06-30:
A little late, but: yes, this fixes the issues I had with the ESP8266: "NodeM" made by RobotDyn!